SensorsIot/IOTstack #283: Read this if your InfluxDB is broken

Paraphraser, 2021-02-28:
# InfluxDB quick fix

If you have just tried to install IOTstack or upgrade InfluxDB, you will probably be seeing this error:

```
Pulling influxdb (influxdb:latest)...
latest: Pulling from library/influxdb
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries
```

## Solution: pin to version 1.8.4

Here is what you need to do to get going again:

1. Start in IOTstack:

	```
	$ cd ~/IOTstack
	```

2. Open `docker-compose.yml` in your favourite text editor.
3. Look for these lines:

	```
	  influxdb:
	    container_name: influxdb
	    image: "influxdb:latest"
	```

4. Change the third line to look like this:

	```
	    image: "influxdb:1.8.4"
	```

5. Save the file then bring up InfluxDB again:

	```
	$ docker-compose up -d influxdb
	```

## When will I be able to remove the pin?

No idea!

The lack of an ARMv7 version of InfluxDB 2.0 could be a mistake. It also may be that version 2.0 needs 64-bit architecture so we IOTstackers will have to wait until Raspbian and all the containers we rely on are also confirmed as 64-bit compatible or native before we can move to InfluxDB 2.0.

## Pull Requests

Two Pull Requests are in the works to change the IOTstack template for InfluxDB to pin to 1.8.4:

* [PR 280 - master branch aka new menu](https://github.com/SensorsIot/IOTstack/pull/280) (already applied)
* [PR 281 - old-menu branch](https://github.com/SensorsIot/IOTstack/pull/281) (still pending as of 2021-02-28)

A `git pull` will fetch these PRs as they are applied but the changes only affect:

```
~/IOTstack/.templates/influxdb/service.yml
```

Manual action (eg a menu run, or editing as above) is still required to get the pinning into your `docker-compose.yml`.

Zephael86, 2021-03-12:
Would it be not possible to use the compose-override.yml file?
With content like:

services:
  influxdb:
    image: "influxdb:1.8.4"

As I understood the docker-compose.yml is dynamically generated. Overrides (even extensions) can be creaded using the compose-override.yml.

Paraphraser, 2021-03-12:
@Zephael86 - probably. Try it. See what happens.

Speaking strictly personally, I take the view that the menu system is great for **generating** your initial set of containers but not so great at maintaining your IOTstack. I have not run the menu (old or new) "in anger" since before IOTstack was forked from gcgarner. The "old" menu surprised me one-too-many times and I decided that I'd get better outcomes by learning how to do everything by hand.

I also don't see the point of "override" schemes, whether it's the Docker-supplied docker-compose.override.yml or the new-menu-supplied compose-override.yml. I'd rather just edit docker-compose.yml where I can see everything in one place.

Putting those two things together implies that, on **my** system, docker-compose.yml is not generated dynamically. It is fixed and stable, unless and until I choose to edit it with a text editor.

But these are my decisions. What other people do is up to them.

Nevertheless, my personal approach to IOTstack maintenance does mean that, when I spot a problem like this, the only approach I can take is to state the problem (eg Influx broken), define the solution (pin to a version), say what docker-compose.yml needs to look like in order to implement the solution, and test that the solution works before writing about it.

I'm in no position to think about the myriad ways other people might implement a particular solution on their own systems. And I certainly don't have the ability to **test** other methods. Each person has to take responsibility for applying the solution in a way that works for them.

I hope that makes sense.

closed 2021-06-01 by Paraphraser

dalklein, 2021-06-02:
Any further advice?    Leave it using
image: "influxdb:1.8” ?