SensorsIot/IOTstack #289: Remove portainer in favor of portainer-ce

wijnsema, 2021-03-06:
If I'm correct the preferred container for portainer is now portainer-ce (community edition).

Can somebody remove portainer from the list to avoid confusion?

Paraphraser, 2021-03-06:
The original reason for installing CE alongside the original was so that we wouldn't wrong-foot anyone running the original with extensions. There may still be people in that situation. We have no way of knowing. I wouldn't want to go ahead and remove the original, only to have people complain that we've broken their setup.

Maybe it's time we had a "change log" so we can start doing things like adding deprecation notices. Thoughts?

mats-nk, 2021-09-16:
I agree with wijnsema , to have two versions is just confusing. So my 2 cent is for removing Portainer.

PR #567 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3
Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
commit 3239e9b4 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3

Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
closed 2022-06-12 by Slyke