SensorsIot/IOTstack #292: Adminer not responding

coolbikerdad, 2021-03-14:
I am successfully running IOTstack with PiHole, MariaDB, Portainer-CE, Heimdall and more. All work with the exception of Adminer, which I thought I had working a while ago, but even after a stack rebuild, deletion and recreation of container and image, does not respond on 9080:8080. Any ideas welcome.

coolbikerdad, 2021-03-22:
I think the latest version of Adminer exceeds the default memory allocated to PHP in the container. The default is 256M, but if I set the environment variable MEMORY=512M in Portainer and redeploy the container, Adminer works again.

MennoSchakel, 2021-04-09:
For me setting the environment variable MEMORY=512M still doesn't have Adminer responding. Downgrading from Adminer:latests (i.e. 4.8.0) to 4.7.8 (the version running on my live version), does not change anything.
On my current live IOTstack (running old menu) Adminer is running without issues.

wiliezwagerman, 2021-07-26:
I had the same issue on a Raspberry Pi 4, it seems to be an issue for adminer images 4.7.8+
Reverting to 4.7.7 solved the issue for me.
[https://github.com/TimWolla/docker-adminer/issues/96](url)

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke