SensorsIot/IOTstack #293: Backup script only works when run as sudo

Thin-White-Duke, 2021-03-14:
The backup directory belongs to root, therefore backup.sh when run as pi has not the right permissions. For the installation,  I ran menu.sh as pi, not as root, but somehow to install docker in the native menu I was prompted to run as root. grafana, influxdb, nodered, portainer-ce - all directories in volumes belong to root except mosquitto belongs to pi as the user permission.
Can I correct things without a complete reinstall? I also installed Home Assistant (native) and don't want to start right from the beginning.
Thanks!

Paraphraser, 2021-03-15:
I don't know whether this will help as a reference. Fairly obviously, it's just the containers I run so you might need to ferret about in your own system to build up a complete picture.

```
$ cd ~/IOTstack
$ tree -puL 1 .
.
├── [drwxr-xr-x pi      ]  backups
├── [-rw-r--r-- pi      ]  docker-compose.yml
├── [drwxr-xr-x pi      ]  docs
├── [drwxr-xr-x pi      ]  duck
├── [-rw-r--r-- pi      ]  LICENSE
├── [-rwxr-xr-x pi      ]  menu.sh
├── [-rw-r--r-- pi      ]  mkdocs.yml
├── [-rw-r--r-- pi      ]  README.md
├── [drwxr-xr-x pi      ]  scripts
├── [drwxr-xr-x pi      ]  services
└── [drwxr-xr-x root    ]  volumes

$ tree -pudL 2 services
services
├── [drwxr-xr-x pi      ]  classic_mqtt
│   └── [drwxr-xr-x pi      ]  support
├── [drwxr-xr-x pi      ]  gitea
├── [drwxr-xr-x pi      ]  grafana
├── [drwxr-xr-x pi      ]  influxdb
├── [drwxr-xr-x pi      ]  iotstack_mosquitto
│   └── [drwxr-xr-x pi      ]  iotstack_defaults
├── [drwxr-xr-x pi      ]  mosquitto
├── [drwxr-xr-x pi      ]  nodered
├── [drwxr-xr-x pi      ]  pihole
├── [drwxr-xr-x pi      ]  portainer-ce
└── [drwxr-xr-x pi      ]  wireguard

$ tree -pudL 2 volumes
volumes
├── [drwxr-xr-x root    ]  gitea
│   └── [drwxr-xr-x root    ]  data
├── [drwxr-xr-x root    ]  grafana
│   ├── [drwxr-xr-x root    ]  data
│   └── [drwxr-xr-x root    ]  log
├── [drwxr-xr-x root    ]  httpd
│   ├── [drwxr-xr-x root    ]  conf
│   ├── [drwxr-xr-x root    ]  htdocs
│   └── [drwxr-xr-x root    ]  logs
├── [drwxr-xr-x root    ]  influxdb
│   └── [drwxr-xr-x root    ]  data
├── [drwxr-xr-x root    ]  influxdb.test
│   └── [drwxr-xr-x root    ]  data
├── [drwxr-xr-x 1883    ]  mosquitto
│   ├── [drwxr-xr-x 1883    ]  config
│   ├── [drwxr-xr-x 1883    ]  data
│   ├── [drwxr-xr-x 1883    ]  log
│   └── [drwxr-xr-x 1883    ]  pwfile
├── [drwxr-xr-x root    ]  nodered
│   ├── [drwxr-xr-x root    ]  data
│   └── [drwxr-xr-x root    ]  ssh
├── [drwxr-xr-x root    ]  pihole
│   ├── [drwxr-xr-x root    ]  etc-dnsmasq.d
│   └── [drwxrwxr-x 999     ]  etc-pihole
├── [drwxr-xr-x root    ]  portainer-ce
│   └── [drwxr-xr-x root    ]  data
└── [drwxr-xr-x pi      ]  wireguard
    ├── [drwxr-xr-x pi      ]  coredns
    ├── [drwx------ pi      ]  peer_iPad
    ├── [drwx------ pi      ]  peer_iPhone
    ├── [drwx------ pi      ]  peer_laptop
    ├── [drwxr-xr-x pi      ]  server
    └── [drwxr-xr-x pi      ]  templates

$ tree -pudL 2 backups
backups
└── [drwxr-xr-x root    ]  influxdb
    └── [drwxr-xr-x root    ]  db
```

Everything else (and I really do mean **EVERYTHING**, including hidden files and folders like `.templates`) that isn't in `backups`, `services` or `volumes` should be owned by pi:pi.

Maybe also see [Paraphraser/IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup) for an alternative backup/restore solution.

This is how I build RPis:

1. Clean image.
2. All setup steps (`touch /boot/ssh`, WiFi config, boot, full-upgrade, raspi-config steps, system patches I want, install add-on packages I want, etc). One of the most [critical patches](https://github.com/SensorsIot/IOTstack/issues/219#issuecomment-766602850) is:

	```
	$ sudo bash -c '[ $(egrep -c "^allowinterfaces eth0,wlan0" /etc/dhcpcd.conf) -eq 0 ] && echo "allowinterfaces eth0,wlan0" >> /etc/dhcpcd.conf'
	```

3. Then:

	```
	$ git clone -b old-menu https://github.com/SensorsIot/IOTstack.git ~/IOTstack
	$ curl -fsSL https://get.docker.com | sh
	$ sudo usermod -G docker -a $USER
	$ sudo usermod -G bluetooth -a $USER
	$ sudo apt install -y python3-pip python3-dev
	$ sudo pip3 install -U docker-compose
	$ sudo pip3 install -U ruamel.yaml==0.16.12 blessed
	$ sudo reboot
	```

	Note:
	
	* you can omit the `-b old-menu` if you want new menu but I think (a) that just confuses things and, (b) as Slyke is working on a new-new-menu, there's really very little point in bothering with the interim new-menu.

4. If you have a backup, this is when you restore it. A restore should get you your docker-compose.yml and then you can bring up your stack.

<hr>

Another approach is to take down your stack, then:

```
$ cd
$ mv IOTstack IOTstack.old
```

Then do the `git clone` as above which gets you the repo.

> If you want to know how to uninstall everything so that you can do the other commands in the above and get a nice clean slate, see [this gist](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f).

After the reboot, then you can move `docker-compose.yml` from the IOTstack.old folder into the fresh clone. Ditto services. Ditto volumes (which will need sudo). The backups folder probably doesn't matter.

Make sure the permissions are as per the reference above (docker-compose.yml and services owned by pi:pi, volumes and **most** of its contents owned by root:root but pay attention to the exceptions like Mosquitto (1883:1883) and Wireguard (pi:pi).

Then you can "up" the stack and see what happens.

Thin-White-Duke, 2021-03-16:
Thank you, I will compare this with my containers. As a quick fix, I changed the group and owner of the backups directory to pi, then I was able at least to make a backup. I also found, that another user here had exactly the same problem issue #254.