SensorsIot/IOTstack #296: I can get all the containers I want to run but I can't login to NextCloud after changing the password in the docker-compose file

GurenMarkV, 2021-03-21:
I can get all the containers I want to run but I can't login to NextCloud after changing the password in the docker-compose file in the IOTstack folder. 

What else do I need to do?
Here is the compose file with the password for root different than main. (Obviously not the actual passwords)

[docker-compose.txt](https://github.com/SensorsIot/IOTstack/files/6178442/docker-compose.txt)

GurenMarkV, 2021-03-23:
Anyone able to help find what I did wrong?

Paraphraser, 2021-03-23:
Well, I don't **know** but my **guess** is that when NextCloud and its database are installed for the first time (ie the persistent storage area in ./volumes is empty), the password provided via the environment variable is used to seed the database. On the second or subsequent runs it is ignored.

If you think about it logically, the only way changing an environment-variable-based password could work is if the database fires up, some process tries to login with the password from the environment variable, fails, concludes that that means the password must have changed, then uses some root-like privilege to force a change. I can imagine security people just loving that in any production environment because, in the scheme of things, docker-compose.yml files and en-clear passwords are hardly "secure".

One database I do know about is InfluxDB. The admin password that can be set in an environment variable is only used to provide a default when running the CLI. The password is **never** used to "set" an admin password in the first place, even on a first-run initialisation. You have to go into the CLI and define both the admin user and password. If you forget it, you're, err, split out of luck.

My guess is that you will have to consult the user guide for the database you're using for NextCloud, find out how to change the admin password, and do it from something like a CLI, possibly needing to provide both the old and new passwords.

I know this isn't an actual answer but I hope it helps you to find the answer.

Paraphraser, 2021-03-23:
See if [how-to-change-mariadb-user-password](https://linuxconfig.org/how-to-change-mariadb-user-password) helps.

GurenMarkV, 2021-03-27:
I am curious. Shouldn't those instructions on setting up the password be provided on the installation guide?

https://sensorsiot.github.io/IOTstack/Containers/NextCloud/

How were others successful in installing NextCloud?

Paraphraser, 2021-03-27:
@GurenMarkV - probably. A lot of things should be documented. The best way to get material added is by submitting a [Pull Request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d).

My guess is that anyone who has NextCloud running just set their passwords at initial setup and hasn't tried to change them since.

GurenMarkV, 2021-03-28:
My issue is I can't seem to setup the Passwords in intial setup. Every other container seems to be fine but NextCloud/MariaDB won't log in. 

I have tried reinstalling 4 times now. Mix of changing the docker-compose.yml before running the stack and after. 

So I am confused how I need to do my passwords and in what order. If you can help with that.

Syltty, 2021-03-29:
Not sure it's the same issue but I can't install nextcloud either.
"ERROR: Service 'nextcloud' depends on service 'nextcloud_db' which is undefined."

Paraphraser, 2021-03-31:
Assuming you're still having trouble...

I'm having no trouble with NextCloud but I did "roll my own". It might solve your problems too.

* docker-compose.yml

	```
	  nextcloud:
	    container_name: nextcloud
	    hostname: nextcloud.mull.lgosys.com
	    image: nextcloud
	    restart: unless-stopped
	    environment:
	    - MYSQL_HOST=nextcloud_db
	    - MYSQL_PASSWORD=PASSWORDONE
	    - MYSQL_DATABASE=nextcloud
	    - MYSQL_USER=nextcloud
	    ports:
	    - "9321:80"
	    volumes:
	    - ./volumes/nextcloud/html:/var/www/html
	    depends_on:
	    - nextcloud_db
	    links:
	    - nextcloud_db
	#   networks:
	#   - iotstack_nw
	#   - nextcloud_internal
	
	  nextcloud_db:
	    container_name: nextcloud_db
	    image: ghcr.io/linuxserver/mariadb
	#   build: ./.templates/mariadb/.
	    restart: unless-stopped
	    environment:
	    - TZ=Australia/Sydney
	    - PUID=1000
	    - PGID=1000
	    - MYSQL_ROOT_PASSWORD=PASSWORDTWO
	    - MYSQL_PASSWORD=PASSWORDONE
	    - MYSQL_DATABASE=nextcloud
	    - MYSQL_USER=nextcloud
	    ports:
	    - "3306:3306"
	    volumes:
	    - ./volumes/nextcloud/db:/config
	#   networks:
	#   - nextcloud_internal
	```

Notes:

1. Networking:
	
	I run "old menu" so I don't have those "networks" definitions. You **could** enable those if you use "new menu" but it really depends on whether another containerised service needs to talk to nextcloud.

	Think about the basic MING stack. NodeRed needs to talk to both Mosquitto and Influx, and Grafana needs to talk to Influx. It makes sense for them all to share a common internal bridged network.

	Does anything in another container need to talk to NextCloud? Does NextCloud need to talk to anything in another container? If "no" and "no" then networks are irrelevant. If omitted, they will talk to each other across iotstack_default.
	
2. Passwords:

	Make sure the two PASSWORDONE values are the same, and that PASSWORDTWO is different. I steer clear of anything with password substitution or random generation schemes. One day. Maybe. In the meantime, it's a case of, "if I don't understand it, I don't use it."
	
3. Focus on the second and third lines of the `nextcloud_db` definition. The `image:` directive is active and the `build:` directive commented out. If you leave it like that, you'll get the mariadb image "as it comes" from DockerHub. It will work.

	However, there was discussion recently about MariaDB stalling and losing data. The solution was to comment-out a couple of directives in the config file.
	
	You can do the same thing, by hand, in the config file, once it is set up after the first run.
	
	But, if you want to automate it, you can comment-out the `image:` directive and activate the `build:` directive, so long as you also have this file at the path `~/IOTstack/.templates/mariadb/Dockerfile`

	```
	# Download base image
	FROM ghcr.io/linuxserver/mariadb
	
	# apply stability patches recommended in
	#   
	#   https://discord.com/channels/638610460567928832/638610461109256194/825049573520965703
	#   https://stackoverflow.com/questions/61809270/how-to-discover-why-mariadb-crashes
	RUN sed -i.bak \
	  -e "s/^thread_cache_size/# thread_cache_size/" \
	  -e "s/^read_buffer_size/# read_buffer_size/" \
	  /defaults/my.cnf
	
	# EOF
	```
	
	You can follow the links in that code to read about the issue. If you do decide to do this, I'd appreciate feedback. It works on my system but it never hurts to have other people test it. I'm thinking of turning it into a Pull Request. It's in the MariaDB template folder. The idea is that it will apply to all MariaDB instances.

4. If it were me, I'd also erase NextCloud's persistent storage area:

	```
	$ cd ~/IOTstack
	$ sudo rm -rf ./volumes/nextcloud
	$ docker-compose up -d --build
	```

GurenMarkV, 2021-04-08:
Hello again,

It's been a while.

I tried to use the old menu like you suggested. 
Seems to be installing just fine. Except now I am stuck on a new screen. 

I didn't have this screen when using the New Menu. It matches the Documentation but I am getting a new error now. 

[image]

[image]

dparson55, 2021-04-25:
I also got stuck trying to get Nextcloud working with the same db error but was able to solve it with some fiddling.  Maybe some of the steps I did will help.

1. Started the menu with "sudo ./menu.sh"
2. Stopped the containers using the menu > docker commands > stop stack
3. Opened a new terminal and deleted the nextcloud volume "sudo rm -rf volumes/nextcloud"

[image]

4. Went to the menu > build stack > nextcloud > options (hit the right arrow > password options and changed the option to default password (previously it was set to "do nothing")
5. Went back to the menu > build stack menu (hit esc) and hit (enter) to build the stack.  This overwrote the docker-compose.yml
6. Tried to start the stack from the docker commands but it failed saying the container "nextcloud_db" could not be found.  So I tried a couple more times and did something, I don't know what, that caused the nextcloud_db container to be written to the docker-compose.yml.  It looked like this.

[image]

7. Went to the menu > docker commands > start stack
8. Tried browsing to Nextcloud on port 9321 and after a minute or two it came up.  I entered an admin user and password and after a minute or so the browser timed out.  I refreshed the browser a minute later and it came up

[image]

[image]

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke