SensorsIot/IOTstack #297: Adguard support

grubeli, 2021-03-24:
Before anything, amazing project, thanks a lot.

I think AdGuard is pretty good alternative of pi-hole, it's open-source and works well on RasPi.
Could you please consider to adding support?

Regards,
Erica

Paraphraser, 2021-03-25:
Hi Erica,

I thought I'd give this a whirl but I've run into a small documentational snag. I'm hoping you have enough practical experience with Adguard to be able to help me (or, if not, maybe you know who to ask).

On [Dockerhub adguard/adguardhome](https://hub.docker.com/r/adguard/adguardhome), there's this example:

```
docker run --name adguardhome \
    -v /my/own/workdir:/opt/adguardhome/work \
    -v /my/own/confdir:/opt/adguardhome/conf \
    -p 53:53/tcp -p 53:53/udp \
    -p 80:80/tcp -p 3000:3000/tcp \
    -p 67:67/udp -p 68:68/tcp -p 68:68/udp \
    -p 443:443/tcp \
    -p 853:853/tcp \
    -d adguard/adguardhome
```

followed by material including:

> Now you can open the browser and navigate to http://127.0.0.1:3000/ to control your AdGuard Home service.
>
>Ports mappings you may need:
>
> * -p 67:67/udp -p 68:68/tcp -p 68:68/udp - add if you intend to use AdGuard Home as a DHCP server.
> * -p 443:443/tcp - add if you are going to run AdGuard Home as an HTTPS/DNS-over-HTTPS server.
> * -p 853:853/tcp - add if you are going to run AdGuard Home as a DNS-over-TLS server.
> * -p 784:784/udp - add if you are going to run AdGuard Home as a DNS-over-QUIC server.
> * -p 5443:5443/tcp -p 5443:5443/udp - add if you are going to run AdGuard Home as a DNSCrypt server.

If you work through all of that, two ports are mentioned in the `docker run` which are then not defined in the later material. Those are ports 53 and 80.

Port 53 is obvious: normal DNS.

But I have not been able to find any explanation for port 80. If AdGuard's web GUI is at port 3000 then what is 80 for?

I've searched through [github.com AdguardTeam/AdGuardHome](https://github.com/AdguardTeam/AdGuardHome) without stumbling across anything.

Any ideas?

Paraphraser, 2021-03-25:
Further to the above, I've done a test spin-up:

```
$ DPS adguard
NAMES         CREATED         STATUS
adguardhome   3 minutes ago   Up 3 minutes
```

Up and running!

```
$ DNET adguard
NAMES         PORTS
adguardhome   67-68/udp, 443/udp, 443/tcp, 853/tcp, 3001/tcp, 784/udp, 0.0.0.0:53->53/tcp, 0.0.0.0:53->53/udp, 5443/udp, 5443/tcp, 0.0.0.0:8089->80/tcp, 0.0.0.0:3001->3000/tcp
```

Confirmation that I've mapped:

* 3001->3000 (Grafana already uses external port 3000), and
* 8089->80.

Check port 3001->3000 for HTTP patterns:

```
$ curl -I 127.0.0.1:3001
HTTP/1.1 302 Found
Content-Type: text/html; charset=utf-8
Location: /install.html
Date: Thu, 25 Mar 2021 07:06:05 GMT
```

Port is responding!

Check port 8089->80 for HTTP patterns:

```
$ curl -I 127.0.0.1:8089
curl: (56) Recv failure: Connection reset by peer
```

Port not responding.

I'm coming to the view that the `docker run` example is wrong. Thoughts?

Paraphraser, 2021-03-25:
Still, it's defined as an exposed port in the image...

```
$ docker image inspect adguard/adguardhome:latest | jq .[0].Config.ExposedPorts
{
  "3000/tcp": {},
  "3001/tcp": {},
  "443/tcp": {},
  "443/udp": {},
  "53/tcp": {},
  "53/udp": {},
  "5443/tcp": {},
  "5443/udp": {},
  "67/udp": {},
  "68/udp": {},
  "784/udp": {},
  "80/tcp": {},
  "853/tcp": {}
}
```

Paraphraser, 2021-03-25:
Starting point (docker-compose.yml fragment):

```
$ grep -A 28 "^  adguardhome:" ~/IOTstack/docker-compose.yml 
  adguardhome:
    container_name: adguardhome
    image: adguard/adguardhome
    restart: unless-stopped
    environment:
      - TZ=Australia/Sydney
    # enable host mode to activate DHCP server on ports 67/udp & 68/tcp+udp
    # note that you must also disable all other ports if you enable host mode
    # network_mode: host
    ports:
      # regular DNS
      - "53:53/tcp"
      - "53:53/udp"
      # what?
      - "8089:80/tcp"
      # HTTPS/DNS-over-HTTPS
      # - "443:443/tcp"
      # DNS-over-QUIC
      # - "784:784/udp"
      # DNS-over-TLS
      # - "853:853/tcp"
      # administration (http)
      - "3001:3000/tcp"
      # DNSCrypt
      # - "5443:5443/tcp"
      # - "5443:5443/udp"
    volumes:
       - ./volumes/adguardhome/workdir:/opt/adguardhome/work
       - ./volumes/adguardhome/confdir:/opt/adguardhome/conf
```

Spins up OK. I can connect to port 3001 from **another** host (ie not the RPi running adguardhome in a container):
 
[image]

The next screen is, I think, a problem:
 
[image]

I have tried changing "80" (an **internal** port number to both "8089" and "3001" as **external** ports mapping to 80 and 3000, respectively, but I can't see any way of affecting the URLs below that.

127.0.0.1 **might** work if the URL were employed in some user interface running outside of container-space on the same RPi but it is not going to work anywhere else.

172.18.0.5 is the IP address of the container inside container-space so it isn't going to work anywhere (well, not anywhere useful).

Basically, the setup GUI works - until you reach the end of the process, after which nothing works.

I have tried editing:

```
~/IOTstack/volumes/adguardhome/confdir/AdGuardHome.yaml 
```

to set the default:

```
bind_host: 0.0.0.0
```

to the FQDN of the Raspberry Pi running adguardhome but that didn't achieve anything, irrespective of which port number I specified for `bind_port`.

Unless you know of some magic incantation for configuring the thing, I have a lot of trouble believing adguardhome is actually ready for "prime time" when running in a container.

Putting the container in host mode is a possibility but it uses so many ports (which will all become active) that it will cause conflicts with other IOTstack containers. I've already mentioned that 3000 collides with Grafana. Changing Grafana would break so many installations that it would be source of endless complaints.

adguardhome may be great when installed natively in a situation where it doesn't have to cohabit with other IOTstack containers but that's not why we're here, is it?

Final thought. I've only given this a couple of hours so it's entirely possible I missed some critical point which immediately solves all the problems I mention. I'm **not** saying it will **never** work. I'm saying that, right now, I can't see how to make it work.

grubeli, 2021-03-25:
Thanks Phill for your work

grubeli, 2021-03-25:
https://github.com/AdguardTeam/AdGuardHome/discussions/2874

grubeli, 2021-03-26:
From an Adguard maintainer:
"As I see most of the confusion comes from the fact that AdGuard Home prints internal docker addresses in the UI, but this is kinda normal - it simply cannot know the host machine addresses. And with published ports, AGH should be accessible on the outside, i.e. on host_ip:published_port.

I assume that anyone using docker is familiar with this behavior. I guess the only thing that we can is improve the documentation a little bit and explain this."

Paraphraser, 2021-03-30:
@grubeli 

Hi Erica,

I have done some more experimentation and have had success. What I did this time was:

1. Start from a clean slate, as in:

	* AdGuard Home container not running
	* erase the persistent storage area:
	
		```
		$ sudo rm -rf ~/IOTstack/volumes/adguardhome/
		```
		
	* bring up the AdGuard Home container (same docker-compose.yml fragment as before)

2. Connect to `http://new-dev:3001` (which maps to container port 3000).

	> `new-dev` is the name of the Raspberry Pi I am testing on.
	
3. Other than setting a username and password, I accepted all defaults.

Clicking "Open Dashboard" resulted in the browser error message:

```
Safari Can't Connect to the Server
Safari can't open the page "new-dev" because Safari can't connect the server "new-dev".
```

The URL proposed (**by AdGuard Home**) in the Safari URL bar was:
	
```
http://new-dev
```

When I hand-edited that to:

```
http://new-dev:8089
```

I got the login screen and was able to login. Lightbulb moment!

During my initial testing, I focused on setup screen 2/5 because that seemed to be the logical place to affect the URL being proposed by AdGuard Home when the "Open Dashboard" button was clicked. Although it seemed clear that the "Admin Web Interface Port" would only affect the **internal** port, I tinkered with it anyway, getting nowhere.

It simply did not occur to me that there might be an implied 1:1 nexus between the internal and external ports, such that the **SAME** port number **MUST** be used **on both sides**.

I started over with another clean slate, save that this time I changed the "port 80" mapping in docker-compose.yml like this:

```
#     - "8089:80/tcp"
      - "8089:8089/tcp"
```

Then I connected to port 3001:3000 and went through the configuration steps, changing the Admin Web Interface port number to 8089 along the way. Clicking "Open Dashboard" served `http://new-dev:8089` which redirected to `http://new-dev:8089/login.html`, and the login screen appeared. Bingo!

<hr>

> "As I see most of the confusion comes from the fact that AdGuard Home prints internal docker addresses in the UI, but this is kinda normal - it simply cannot know the host machine addresses.

The IP addresses weren't the source of my confusion. I realised neither was going to work as soon as I saw them. That's what I said in the earlier post. The above simply re-states the same thing in different words.

I've now realised *why* I was confused. It was the undocumented and un-obvious 1:1 port-mapping requirement for the AdGuard Home management interface. It's un-Docker-like, which is why the possibility didn't occur to me sooner.

The `docker run` example for AdGuard Home on DockerHub has a 1:1 relationship for every port. I suspect that that is all that has ever been tested. I say that because I can't see how anyone could have tried passing something like `-p 8089:80` without running into the same problems I did.

> And with published ports, AGH should be accessible on the outside, i.e. on host\_ip:published\_port. I assume that anyone using docker is familiar with this behavior. I guess the only thing that we can is improve the documentation a little bit and explain this."

I don't accept that. The average IOTstack user is **not** someone who claims to be "familiar with Docker". The whole point of IOTstack is that users can spin-up a basic MING stack plus other supported containers and have it all "just work". In the IOTstack context, it simply isn't reasonable to dismiss issues by presuming familiarity with anything and that, by inference, anything any user finds confusing is the user's fault for not being sufficiently familiar with [insert chosen topic here]. The IT industry has been doing this blame-and-shame for 75 years and it's high time **we** stopped.
[... 97 lines cut ...]
```

With that kind of arrangement, the IOTstack user could be instructed to simply set the username and password, leaving everything else alone.

Passing "port knowledge" into running containers using environment variables is a reasonably common technique so AdGuard Home wouldn't be setting any precedent:

```
$ cd ~/IOTstack/.templates
$ find . -name "service.yml" -exec grep -H "PORT=" {} \;
./homebridge/service.yml:    - HOMEBRIDGE_CONFIG_UI_PORT=%WUIPort%
./rtl_433/service.yml:    - MQTT_PORT=1883
./openhab/service.yml:    - OPENHAB_HTTP_PORT=4050
./openhab/service.yml:    - OPENHAB_HTTPS_PORT=4051
./wireguard/service.yml:    - SERVERPORT=51820
./zigbee2mqtt_assistant/service.yml:    - VIRTUAL_PORT=8880
./qbittorrent/service.yml:      - WEBUI_PORT=15080
```

We probably could go ahead and integrate AdGuard Home into IOTstack right now.

We could also give AdGuard Home's designers the chance to read this issue, reflect on it, and see whether they find my arguments sufficiently persuasive for them to want to make AdGuard Home slightly more Docker- and IOTstack-friendly.

My 2¢ is to try the latter as the best way of avoiding anyone taking offence at the instruction wording I've suggested above, and complaining that we should have explained the problem and given them a chance to fix it before going into print, as it were.

What do you think?

grubeli, 2021-03-30:
Thanks Phill!
looks amazing

Paraphraser, 2021-04-10:
@grubeli I've gone ahead and put in two Pull Requests:

* master branch - [PR314](https://github.com/SensorsIot/IOTstack/pull/314)
* old-menu branch -  [PR315](https://github.com/SensorsIot/IOTstack/pull/315)

See [Testing Pending Pull Requests](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-) if you want to give it a whirl prior to it being accepted/rejected by Slyke.

The master branch PR also includes some documentation.

closed 2021-11-01 by grubeli