SensorsIot/IOTstack #308: Zigbee2MqttAssistant: Unable to connect to MQTT server mosquitto

batamire, 2021-04-02:
Fresh install, from the logs:

```
Starting Zigbee2MqttAssistant v0.3.157+Branch.master.Sha.ea388c3bbc2d43c4c9c62f27bb6764911ba4acc9...
warn: Microsoft.AspNetCore.DataProtection.Repositories.FileSystemXmlRepository[60]
      Storing keys in a directory '/root/.aspnet/DataProtection-Keys' that may not be persisted outside of the container. Protected data will be unavailable when container is destroyed.
Hosting environment: Production
Content root path: /app
Now listening on: http://[::]:80
Application started. Press Ctrl+C to shut down.
warn: Zigbee2MqttAssistant.Services.MqttConnectionService[0]
      Unable to connect to MQTT server mosquitto.
MQTTnet.Exceptions.MqttCommunicationException: Name or service not known
 ---> System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (00000005, 0xFFFDFFFF): Name or service not known
   at System.Net.Dns.InternalGetHostByName(String hostName)
   at System.Net.Dns.ResolveCallback(Object context)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw(Exception source)
   at System.Net.Dns.HostResolutionEndHelper(IAsyncResult asyncResult)
   at System.Net.Dns.EndGetHostAddresses(IAsyncResult asyncResult)
   at System.Net.Sockets.Socket.DoDnsCallback(IAsyncResult result, MultipleAddressConnectAsyncResult context)
   at System.Net.Sockets.Socket.DnsCallback(IAsyncResult result)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw(Exception source)
   at System.Net.Sockets.Socket.EndConnect(IAsyncResult asyncResult)
   at System.Net.Sockets.Socket.<>c.<ConnectAsync>b__277_0(IAsyncResult iar)
--- End of stack trace from previous location where exception was thrown ---
   at MQTTnet.Implementations.MqttTcpChannel.ConnectAsync(CancellationToken cancellationToken)
   at MQTTnet.Internal.MqttTaskTimeout.WaitAsync(Func`2 action, TimeSpan timeout, CancellationToken cancellationToken)
   at MQTTnet.Adapter.MqttChannelAdapter.ConnectAsync(TimeSpan timeout, CancellationToken cancellationToken)
   --- End of inner exception stack trace ---
   at MQTTnet.Adapter.MqttChannelAdapter.WrapException(Exception exception)
   at MQTTnet.Adapter.MqttChannelAdapter.ConnectAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at MQTTnet.Client.MqttClient.ConnectAsync(IMqttClientOptions options, CancellationToken cancellationToken)
   at MQTTnet.Client.MqttClient.ConnectAsync(IMqttClientOptions options, CancellationToken cancellationToken)
   at MQTTnet.Extensions.ManagedClient.ManagedMqttClient.ReconnectIfRequiredAsync()
```

Paulzie, 2021-04-06:
Not sure if it's related but on a brand new Raspberry Oi OS and IOTStack install, Zigbee2Mqtt cannot connect to Mosquitto MQTT.  I can connect to Mosquitto externally without issue.  Have also tried removing all Docker-compose networking settings just in case.

Had to do some work on mosquitto file permissions as per https://github.com/SensorsIot/IOTstack/issues/307

Paraphraser, 2021-04-07:
Try making the service definition look like this:

```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    environment:
      - TZ=Etc/UTC
    ports:
      - "8080:8080"
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
    devices:
      - "/dev/ttyACM0:/dev/ttyACM0"
    restart: unless-stopped
```

CC2531 adapters should show up on `/dev/ttyACM0` by default but you may wish to follow the instructions at [Running Zigbee2MQTT](https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html) so that you wind up with something like:

```
    devices:
      - "/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E306A3-if00:/dev/ttyACM0"
```

For now, if `/dev/ttyACM0` is "there" in the sense of showing up in an `ls` command, I'd go with that and change it to be "by ID" once it was working.

> Unless you have a different kind of adapter...

Bring up the container:

```
$ docker-compose up -d zigbee2mqtt
```

On a first install, that creates:

```
~/IOTstack/volumes/zigbee2mqtt/data/configuration.yaml
```

Use `sudo` to edit that file:

1. Replace the directive:

	```
	server: 'mqtt://localhost'
	```
	
	with:
	
	```
	server: 'mqtt://mosquitto'
	```
	
2. Add these lines:

	```
	frontend:
	  port: 8080
	```
	
Restart the container:

```
$ docker-compose restart zigbee2mqtt
```

Check your work:

1. Do a `docker logs zigbee2mqtt` to see if anything untoward shows up.
2. If you have the mosquitto clients installed (`sudo apt install -y mosquitto-clients`) then you can also try something like:

	```
	$ mosquitto_sub -v -h "localhost" -t "zigbee2mqtt/#" -F "%I %t %p" &
	```
	
	which will set up a subscriber in the background listening to zigbee2mqtt topics. You'll either get silence (bad) or tons of output (good). You can trigger more MQTT traffic by restarting zigbee2mqtt.
	
	When you get bored:
	
	```
	$ kill %1
	```

3. Use a browser to connect to the RPi on port 8080, and make sure you can see the UI.

Please let me know if this works - I'm contemplating a Pull Request to make the above service definition **and** config changes the default for IOTstack, so that the only thing anyone should have to do is get the device right.

<hr>

It's possible that this would also help with the original post on this issue. I look at:

```
Now listening on: http://[::]:80
Application started. Press Ctrl+C to shut down.
warn: Zigbee2MqttAssistant.Services.MqttConnectionService[0]
      Unable to connect to MQTT server mosquitto.
```

The last line makes me think "localhost in configuration file" while the first line makes me think "IPv6 - now *there's* a whole can of worms."

At the moment, the service definition in `~/IOTstack/.templates/zigbee2mqtt` puts the container in host mode. That isn't needed and I think it might be part of many complaints about zigbee2mqtt.

My **guess** (and it really is no more than a guess; always hard to be sure when service definitions and configuration files aren't attached to the issue) is that the underlying causes of the problem @batamire is reporting might be a combination of host mode, IPv6, [Issue 265](https://github.com/SensorsIot/IOTstack/issues/265), and having perhaps having configured zigbee2mqtt to listen on port 80 instead of 8080.

On my home network, I turn off IPv6 whenever I see it, including on all my Pis. It causes far too much extra network chatter for my liking and it *seriously* tees me off when I'm chasing a problem and I realise IPv6 is implicated in gumming up the works.

My understanding is that Docker starts with IPv6 off and you have to do work to turn it on. I really have no idea what happens if a process like zigbee2mqtt is running in host mode, is binding to IPv6 ports, and is trying to communicate with a process like Mosquitto across a network path which does not transport IPv6. It probably should work. It probably does work. I could be imagining things and tilting at windmills. It just makes me itch.

Taking zigbee2mqtt out of host mode sidesteps the issue entirely. Either the user has done the job of enabling IPv6 for the internal networks or he/she hasn't. Either way, both zigbee2mqtt and mosquitto will have a **consistent** view of the available protocols.

If you want to turn off IPv6 on the RPi, use `sudo` to edit `/etc/sysctl.conf`:

1. Find the line:

	```
	#net.ipv6.conf.all.forwarding=1
	```

2. Insert below that line:

	```
	net.ipv6.conf.all.disable_ipv6=1
	net.ipv6.conf.default.disable_ipv6=1
	net.ipv6.conf.lo.disable_ipv6=1
	```

3. Run:

	```
	$ sudo sysctl -p
	$ sudo reboot
	```
	
	> be aware that if you're connected to the Pi via IPv6 when you do this, the `sysctl` will have immediate effect (ie clobber the connection) so maybe just best to reboot.

Hope something in this helps.

batamire, 2021-04-07:
I did change `docker-compose.yml` per @Paraphraser intructions, adding ports, environment and removing host network mode. Restarted everything.

When i change `server: 'mqtt://localhost' -> 'mqtt://mosquitto'` and restart zigbee2mqtt i get:
```
Zigbee2MQTT:info  2021-04-07 05:14:06: Connecting to MQTT server at mqtt://mosquitto
Zigbee2MQTT:error 2021-04-07 05:14:16: Not connected to MQTT server!
```
`localhost` does work even without `network_mode: host`?

I did disable IP6 per instructions and mosquitto_sub works fine.

Even changing `Z2MA_SETTINGS__MQTTSERVER=mosquitto` to `localhost` in `zigbee2mqtt_assistant` and adding host network mode in docker-compose service does not change anything -> `Unable to connect to MQTT server mosquitto`.

The only service using `network_mode: host` is `home_assistant` and I do need that for device discovery.

@Paraphraser thank you for your time!!! I can live without zigbee2mqtt_assistant for now 😉

Paraphraser, 2021-04-07:
Think of containers as individual computers. Every computer has the concept of "localhost". When you do something like:

```
$ ping -c 1 localhost
```

you are pinging the Raspberry Pi. But when you do:

```
$ docker exec nodered ping -c 1 localhost
```

you are pinging the Node-Red container.

When a container runs in "host mode", whatever ports it sets up are the Raspberry Pi's ports. Any port mappings in docker-compose.yml have no effect (and current versions of docker-compose enforce the rule that, if you put a container in host mode, you **can't** have any port mappings).

When you take a container out of host mode, that's when the port mappings kick in. Given a mapping like "123:345", Docker connects to the Raspberry Pi's port 123 and forwards the traffic to the container on its port 345.

So, let's assume Mosquitto is running in host mode and is exposing port 1883. At the RPi command line, you can get to it as "localhost:1883" and that's how another container (like Zigbee) will think of it too: mqtt://localhost:1883.

Now, let's take Mosquitto out of host mode and assume a port mapping of 1883:1883. At the command line, you still think of it as localhost:1883 but Docker is listening to 1883 and sending it to the Mosquitto container's 1883. The same for Zigbee.

Next, we'll take Zigbee2MQTT out of host mode. Now, mqtt://localhost:1883 means "port 1883 on **this** container". The Zigbee container doesn't have an MQTT service so the request fails. It has to refer to Mosquitto as "mqtt://mosquitto:1883". The "mosquitto" part of that is the container name. Docker sets up something akin to a "hosts" file mapping container names to the dynamically-allocated IP addresses on the internal bridged network.

To complete the picture, let's leave Zigbee2MQTT not in host mode but put Mosquitto back into host mode. Now, the "mosquitto:1883" won't work and neither will "localhost:1883". The only way is to use the IP address or fully qualified domain name of the Raspberry Pi itself, like mqtt:192.168.210.5:1883. That causes the traffic to route out of container-space where Mosquitto is listening to 1883. Its replies get routed back into container space.

To add to this complexity, and assuming both containers are **not** in host mode, we also have "new menu" which introduces a bunch of internal networks (old-menu has a single network called `iotstack_default`). It's possible that this is the missing bit that needs to be added to the end of the Zigbee2MQTT fragment in docker-compose.yml:

```
zigbee2mqtt:
  …
  restart: unless-stopped
  networks:
    - iotstack_nw
```

I run old menu so I keep forgetting about this problem. I run old menu because I **hate** the explosion of networks. I'm sorry if my forgetfulness has wasted your time.

batamire, 2021-04-07:
You have really put me to shame! I am a more than well-seasoned webdev with some devops knowledge and I use docker daily (however only for dev) and I do understand the concept of docker networks, but you are an eye-opener!

Both `zigbee2mqtt` and `zigbee2mqtt_assistant` need `iostack_nw`. `mosquitto` already has it. `host` mode on `zigbee2mqtt` was making `localhost` work initially. It was strange that I had to use full RPi IP to connect HA to MQTT but I do get it now.

I think all services should be on `iotstack_nw` by default, and only `home_assistant` should use `host` - and it does not by default.

And to add more to my confusion, I was restarting single compose services which did not pick up all network changes that a decent down/up cycle did. Not to mention the number of services I selected to start with.

This was a really bumpy and overwhelming beginning with IOTstack and docker based HA! It always is with new tech hehe

@Paraphraser Internet is a better place with You 🙌🏼

closed 2021-04-07 by batamire

Paraphraser, 2021-04-07:
Thanks for the feedback.

Could you please take me one step further and explain why home assistant should use host mode.

We've only just taken it out of host mode [PR300](https://github.com/SensorsIot/IOTstack/pull/300) because there seemed to be no logical reason, other than Amazon Dash, for it to be in host mode. Dash is (supposedly) a discontinued product. Personally, I take the view that if Dash is the sole reason then people who need it should adapt their installations to host mode, leaving everyone else the comparative freedom of running in non-host mode.

Is there something other than Dash that we need to be aware of? Something actually needing multicast support would be a reasonably good reason but I did not find any mention of that when I searched.

What I read about how "Dash support" uses pcap to sniff traffic left me feeling distinctly unclean. With my former network-dude hat on, it troubles me, even on a home network. It's a hack and, I would've thought, a fairly fragile hack at that.

There seem to be numerous examples where containers are in host mode for no discernible reason. I mean, if I was doing doco for a new container and I found it had to be in host mode to work, I'd definitely explain **why**. In the absence of cogent reasons, I'm left thinking that I'm looking at IT archeology. Someone was trying to get something to work, wondered if host mode would help, tried it, it either didn't but they didn't back out the change, or it did but not for that reason and they simply didn't realise. The same thing seems to happen with the privileged flag. My starting position is: if you can't explain it, you don't understand it. From there, if you understand it but don't document it then you're doing the rest of the community a great disservice, the corollary being that the absence of doco is evidence supporting the proposition that host mode ain't needed. My 2c.

batamire, 2021-04-07:
I couldn't get HA to discover any of my local wifi/lan devices without host mode. Is there any other way?

Paraphraser, 2021-04-07:
I honestly don't know. I don't run "home" anything. I sort of came to all this via the roll-your-own Arduino path so it's ESP8266, ESP32, LoRa boards, 3D printed cases, PCBs from JLCPCB, everything eventually speaking MQTT.

The first time I came across "home" was arriving in the middle of what seemed like constant grizzles about things not working, and never being too sure whether HA, hass.io, Home this, that and the other were synonyms or myriad kettles each containing entirely different fish. But the message was clear: here be dragons (or perhaps, to avoid mixing my metaphors, fish-eating dragons).

Then, one day, I decided to give one bit of it a whirl. That's when I came across Network Manager. Holy Hannah! I wound up rebuilding my test machine. There was no way that piece of work was ever going to be let near any of my machines. Not. On. A. Bet. As soon as something says, "you just use home-whatever to glue it together" I run for the hills.

However, putting all that vent and spleen to one side, you've only got three kinds of traffic, right? Unicast, Multicast and Broadcast. If some "device" is unicasting then it's sending its payload to a known end-point (eg ESP32 speaking MQTT over WiFi directed to Mosquitto). If it's unicasting and you **can't** control where it's sending (which I think might be the situation with Dash - save that I don't grok how that works in a switched environment) then you really have no business making it part of your lineup (sorry: more spleen; can't help it). If it's broadcasting then you have to question why you want to hook into such a mechanism? Broadcasts, and their flooding behaviour, are nasty things. Maybe not so much for a small home network but that's where we're training the comms-dudes of the future so we should set good examples. Broadcasts need to be reserved for essentials like ARP and DHCP/BOOTP and so on, and not hijacked as a mechanism of choice by things like IoT devices (spleen!). That leaves multicast. Which is, I reckon, about the only sensible reason.

But, even then, I think I'd argue in terms of an agent registering to receive multicast traffic on behalf of a container, and then forwarding the traffic to the container over established port-pairs. It's cleaner - in comms terms - even though it might also be argued that it is inimical to the idea of containerisation. I have a lot of sympathy with that view so I wind up sitting on the fence, which does my spleen no good at all.

Anyway, the short answer is that if you don't know how HA does its discovery then 🤷‍♂️

<hr>

MEANWHILE, I've just submitted a pair of pull requests:

* [PR309](https://github.com/SensorsIot/IOTstack/pull/309) for master branch (new menu)
* [PR310](https://github.com/SensorsIot/IOTstack/pull/310) for old-menu

I hadn't actually considered the Zigbee2MQTT_Assistant so I added the missing network to that at the same time.

I also expanded the Zigbee2MQTT documentation page.

If you want to give PR309 a try in advance of @Slyke accepting/rejecting, there's some [how to](  https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-) here.