SensorsIot/IOTstack #316: error on docker python

penrose72, 2021-04-13:
I have try several time. With installation from scratch for avoid "noise".

But always have this error:

Step 3/5 : COPY requirements.txt ./
ERROR: Service 'python' failed to build: COPY failed: file not found in build context or excluded by .dockerignore: stat requirements.txt: file does not exist

dinpoplaca, 2021-04-14:
same here. I had to delete the Python config from Docker-compose to work. when I add that back it breaks again.
same thing happened with node red and I removed node red too. I cannot add them back via the menu. as soon as I select them it breaks the docker-Compose file.

hi3ronym0s, 2021-05-16:
I have the exact same problem.

ukkopahis, 2022-06-22:
For the python service to work, you need follow:
https://sensorsiot.github.io/IOTstack/Containers/Python/#addingPackages

ukkopahis, 2022-06-23:
The Python service is for developing your own application.
For the python service to work, you need follow:
https://sensorsiot.github.io/IOTstack/Containers/Python/#addingPackages

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke