SensorsIot/IOTstack #317: Where to find influxdb.env?

KlausHans, 2021-04-15:


KlausHans, 2021-04-15:
I think the documentation is just outdated. 
I simply connected to the container via the Portainer terminal and opened the influxcli here and created a database. Works.

closed 2021-04-15 by KlausHans

Paraphraser, 2021-04-16:
How to find the environment file?

```
$ cd ~/IOTstack
$ mkdir -p ./services/influxdb
$ git checkout old-menu
$ cp -a ./.templates/influxdb/influxdb.env ./services/influxdb/
$ git checkout master
$ cat ./services/influxdb/influxdb.env
```

The result being:

```
# InfluxDB's configuration variables (including how each CONFIGURATION
# variable maps to an ENVIRONMENT variable for use in this file) can
# be found at:
#
#   https://docs.influxdata.com/influxdb/latest/administration/config/#configuration-settings
#
# The URL was correct as at 2020-10-11 but is subject to change.
#
# After you make a change in this file, you bring it into effect by:
#
#   $ cd ~/IOTstack
#   $ docker-compose up -d influxdb

# ---------------- some selected environment variables ----------------

# enable INFLUXDB_HTTP_FLUX_ENABLED to support Flux queries

INFLUXDB_HTTP_FLUX_ENABLED=false

# ---------------------------------------------------------------------

# enable INFLUXDB_HTTP_AUTH_ENABLED to enforce authentication for access
# via HTTP and HTTPS. Node-Red, Grafana and the "influx" command line
# interface all use HTTP/S so enabling this option affects pretty much
# everything. Only enable this option AFTER you have set up credentials
# for your database administrator, like this:
#
#   CREATE USER "dba" WITH PASSWORD 'supremo' WITH ALL PRIVILEGES

INFLUXDB_HTTP_AUTH_ENABLED=false

# if INFLUXDB_HTTP_AUTH_ENABLED=true then INFLUX_USERNAME and
# INFLUX_PASSWORD supply database administrator (DBA) credentials to
# the "influx" command line interface. They serve no other purpose. They
# are NOT general-purpose credentials for use in NodeRed or Grafana etc.
# If INFLUXDB_HTTP_AUTH_ENABLED is enabled but INFLUX_USERNAME &
# INFLUX_PASSWORD do not match the values on the CREATE USER statement
# then you will need to authenticate when you start the "influx" CLI.

INFLUX_USERNAME=dba
INFLUX_PASSWORD=supremo

# ---------------------------------------------------------------------

# enable UDP support

INFLUXDB_UDP_ENABLED=false

# if INFLUXDB_UDP_ENABLED=true then InfluxDB listens for line protocol
# data on UDP port 8086. In the context below, the IP address "0.0.0.0"
# means "this host"; analogous to the way "255.255.255.255" means
# "all hosts"

INFLUXDB_UDP_BIND_ADDRESS=0.0.0.0:8086

# Remember that you will need a matching  "ports" definition in your
# docker-compose.yml. For example:
#
#     - "8086:8086/udp"
#
# Also remember that, just because InfluxDB running inside the container
# is listening on UDP/8086, does NOT mean that the external port MUST be
# the same. It could just as easily be "8089:8086/udp". Keep that in
# mind when you set up the sending process.

# if INFLUXDB_UDP_ENABLED=true then data will be written to the
# database named "udp". You MUST create this database, by hand via
# the "influx CLI", before you start sending data, as in:
#
#   CREATE DATABASE udp

INFLUXDB_UDP_DATABASE=udp

# ---------------------------------------------------------------------
```

"Environment files" are an old-menu construct, which is why you can't find the file when you are running on the master branch (aka "new menu"). New menu generally sticks environment variables inline in the `docker-compose.yml`.

> Strictly speaking, the menu you run has nothing to do with it. Docker-Compose supports both environment files and inline environment variables. You can use either or both. If you use both, the environment files are processed first, then the inline environment variables are applied so, if there's any conflict, the inline variable will take precedence. 

I run "old menu" but I don't use environment files, so my fragment for InfluxDB looks like this:

```
  influxdb:
    container_name: influxdb
    image: "influxdb:1.8.4"
    restart: unless-stopped
    ports:
      - "8086:8086"
    environment:
      - TZ=Australia/Sydney
      - INFLUXDB_HTTP_AUTH_ENABLED=false
      - INFLUXDB_HTTP_FLUX_ENABLED=false
      - INFLUXDB_REPORTING_DISABLED=false
      - INFLUXDB_MONITOR_STORE_ENABLED=FALSE
    volumes:
      - ./volumes/influxdb/data:/var/lib/influxdb
      - ./backups/influxdb/db:/var/lib/influxdb/backup
``` 

Converting that fragment to "new menu" would mean adding these lines at the end:

```
    networks:
      - iotstack_nw
```

If you consult the Influx documentation, you'll find that only two of those environment variables are actually doing anything:

```
      - TZ=Australia/Sydney
      - INFLUXDB_MONITOR_STORE_ENABLED=FALSE
```

See [influxdb issues 19543](https://github.com/influxdata/influxdb/issues/19543) to understand the reason for the second one.

I've taken a look at "flux" once or twice but every time I turn it on, something else seems to break so I keep away from it.

The "reporting" thing is a "phone home" function. In theory it defaults to false. I put it in my docker-compose.yml to make sure it stays that way.

The purpose of "auth enabled" is summarised in the environment file listed above but you might find [this tutorial](https://gist.github.com/Paraphraser/9f3aa21fddd5d4e8141a7ab03ffb4dd7) more helpful. Enabling authentication in Influx is a great way to stop everything from working so it's a good idea to understand what's going to happen before you turn it on. Trust me on this! 😱🤬😭🤬

In general, you always need to create databases before you use them so [this tutorial](https://gist.github.com/Paraphraser/c9db25d131dd4c09848ffb353b69038f) on setting up a basic three-node flow might be helpful too.

KlausHans, 2021-04-16:
Thank you very much for the info's!

Finnitio, 2021-05-06:
Alright guys I was wondering about not finding the env file. If I understand correctly influx is configured via docker-compose.yml:
` influxdb:
    container_name: influxdb
    image: "influxdb:1.8.4"
    restart: unless-stopped
    ports:
    - "8086:8086"
    environment:
    - INFLUXDB_HTTP_FLUX_ENABLED=false
    - INFLUXDB_REPORTING_DISABLED=false
    - INFLUXDB_HTTP_AUTH_ENABLED=false
    - INFLUX_USERNAME=dba
    - INFLUX_PASSWORD=supremo
    - INFLUXDB_UDP_ENABLED=false
    - INFLUXDB_UDP_BIND_ADDRESS=0.0.0.0:8086
    - INFLUXDB_UDP_DATABASE=udp
    volumes:
    - ./volumes/influxdb/data:/var/lib/influxdb
    - ./backups/influxdb/db:/var/lib/influxdb/backup
    networks:
    - iotstack_nw`
So I see user 'dba' and password 'supremo'

openening a shell inside the influx  container and checking I get
'> SHOW DATABASES
name: databases
name
----
_internal
telegraf
> show users
user admin
---- -----
>'

Shouldn't 'dba' turn up here? And checking the 'telegraf.conf' file I see

' # HTTP Basic Auth
   username = "telegraf"
   password = "metricsmetricsmetricsmetrics"'

Which is confusing me completely, so how can I set up users and a database for influxdb without needing to reconfigure it when setting it up again? Or do need to do that in the shell and than just back up or reuse the volume?

Paraphraser, 2021-05-06:
I can't help with the telegraf side of things because I don't run it. Sorry.

<hr>

As to influx, the information you are missing is in the earlier part of this issue and it is also covered in [the gist I wrote on the topic](https://gist.github.com/Paraphraser/9f3aa21fddd5d4e8141a7ab03ffb4dd7) but you might have to study the material for a while and experiment before you have the "ahah" moment.

I'll try explaining it slightly differently.

Study the list of environment variables again:

```
- INFLUXDB_HTTP_FLUX_ENABLED=false
- INFLUXDB_REPORTING_DISABLED=false
- INFLUXDB_HTTP_AUTH_ENABLED=false
- INFLUX_USERNAME=dba
- INFLUX_PASSWORD=supremo
- INFLUXDB_UDP_ENABLED=false
- INFLUXDB_UDP_BIND_ADDRESS=0.0.0.0:8086
- INFLUXDB_UDP_DATABASE=udp
```

Notice that the first three and last three begin with `INFLUXDB_` while the middle two begin with `INFLUX_`. That isn't a mistake and it's a clue.

Also, while the actual ordering of environment variables will never make any difference, there is a reason why the two starting with `INFLUX_` are in the middle. It's because these three go together:

```
- INFLUXDB_HTTP_AUTH_ENABLED=false
- INFLUX_USERNAME=dba
- INFLUX_PASSWORD=supremo
```

When you first install IOTstack, run the menu, choose InfluxDB as one of your containers and "up" the stack, influxdb (the process running inside the influxdb container) sees that the persistent storage area is empty and initialises an empty set of databases.

Those initial databases have no security and no list of users. It's not like some databases where there's a default administrator. InfluxDB starts with absolutely nothing.

Suppose you decide to set up a scheme of usernames and passwords. You do that through the "influx" utility, which you get to in a couple of ways:

1. The direct method:

	```
	$ docker exec -it influxdb influx -precision=rfc3339
	```

2. The indirect method:

	```
	$ docker exec -it influxdb bash
	# influx -precision=rfc3339
	```
	
Again, notice how we are seeing both "influxdb" and "influx".

"influxdb" can mean a lot of different things, all depending on context:

* The image that comes down from DockerHub
* The service definition in `docker-compose.yml`
* The container constructed from the combination of the image and service definition
* The database engine running inside the container

"influx", on the other hand, only means one thing:

* The command line interface utility called "influx" that is available to you if you "go inside the container".

When you run `docker-compose up -d` and the influxdb container is brought up, all the environment variables you set in `docker-compose.yml` are passed inside the container. They are "just there" for the life of the container.

As "influxdb" (the database engine) starts running inside the container, it looks for any environment variables starting with "INFLUXDB_" and applies them.

When you come along later and use one of the methods above to start the "influx" utility, that tool looks for environment variables starting with "INFLUX_".

That's why there's a difference in how those environment variable names begin.
[... 68 lines cut ...]

Anything you do in influxdb is saved in the persistent storage area at:

```
~/IOTstack/volumes/influxdb
```

If you run a backup (either the version that comes with IOTstack, or the one I wrote [Paraphraser/IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup), or one you write yourself) then all that is saved.

I don't understand why you would need to set things up more than once unless you are experimenting, I suppose.

> Edit: And now that I think about is how was the telegraf database created?

Some more clues are in the environment variables:

```
- INFLUXDB_HTTP_AUTH_ENABLED=false
- INFLUXDB_UDP_DATABASE=udp
```

The "HTTP" protocol is how most things talk to the influxdb database engine. When a Node-RED flow writes to an influxdb database, it is doing it over HTTP. When Grafana queries an influxdb database to build a chart, it is doing it over HTTP. When you run the "influx" utility inside the container, all the commands you issue are sent to the influxdb database engine over HTTP.

Telegraf, on the other hand, uses UDP. The first time the database engine receives a UDP datagram containing "influx line protocol", it creates the "udp" database and starts writing to it.

As I said at the start, I don't use Telegraf so I don't understand it but my guess is that the username and password in `telegraf.conf` apply to Telegraf, not to the "udp" database in InfluxDB. But I might be wrong.