SensorsIot/IOTstack #346: Home network 192.168.1.0/24

brunohenriquy, 2021-05-12:
If my home network is **192.168.1.0/24** (for example), then should I change it in the **env.yml** for **iotstack_nw**?

```
networks:
  iotstack_nw: # Exposed by your host.
    # external: true
    name: IOTstack_Net
    driver: bridge
    ipam:
      driver: default
      config:
        - subnet: 192.168.1.0/24
        # - gateway: 192.168.1.1

brunohenriquy, 2021-05-12:
The answer to my question is **No**, I just found it. My Raspberry got inaccessible via ssh, so I had to connect fisically to it and remove this network.

So let me explain my real problem:

My RaspberryPi IP is 192.168.1.9, I used to have an AdGuard installed on it, so on my router I just set the primary DNS to 192.168.1.9, DHCP would just send it to all my devices.

I am trying to install AdGuard on docker so it will be part of my IOTstack. When I add it using menu.sh>>Build Stack, it adds:

``` 
adguardhome:
    container_name: adguardhome
    image: adguard/adguardhome
    restart: unless-stopped
    environment:
    - TZ=Etc/UTC
  # enable host mode to activate DHCP server on ports 67/udp & 68/tcp+udp
  # note that you must also disable all other ports if you enable host mode
  # network_mode: host
    ports:
    # regular DNS
    - "53:53/tcp"
    - "53:53/udp"
    # administration port (http)
    # note: external and internal ports MUST be the same
    #       not active until defined via setup port
    - "8089:8089/tcp"
    # HTTPS/DNS-over-HTTPS
    # - "443:443/tcp"
    # DNS-over-QUIC
    # - "784:784/udp"
    # DNS-over-TLS
    # - "853:853/tcp"
    # setup (http)
    # note: only active until port 8089 becomes active
    - "3001:3000/tcp"
    # DNSCrypt
    # - "5443:5443/tcp"
    # - "5443:5443/udp"
    volumes:
    - ./volumes/adguardhome/workdir:/opt/adguardhome/work
    - ./volumes/adguardhome/confdir:/opt/adguardhome/conf
    networks:
    - iotstack_nw
    - vpn_nw

It adds the networks **iotstack_nw** and **vpn_nw**, when I am going to configure it at HTTP://192.168.1.9:3001 it doesn't show me the option 192.168.1.9:

<img width="1316" alt="Screen Shot 2021-05-12 at 8 07 29 PM" src="https://user-images.githubusercontent.com/8320519/118055069-cd24bf00-b35d-11eb-8d7c-ac4cbd5acb45.png">

How to make it appears? Should I add the iotstack_default network to adguardhome?
Or am I configuring it wrong?

Paraphraser, 2021-05-12:
Read [the instructions](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/AdGuardHome.md#-quick-start-) and do what it says.

The basic problem is that AdGuardHome's designers haven't really thought about containerisation. The software still thinks it is installed natively. It can only "see" the container it is running in. You have to fool it by following the instructions: change the port to 8089 instead of 80 and ignore everything else.

I really do mean **ignore** everything else. Don't try to make sense of it.

Once it is up and running, a DNS query sent to port 53 will be heard by Docker which is listening to the RPi port on behalf of the container (ie 192.168.1.9:53). Docker routes the packet across the internal docker networks to the container on its port 53 where adguardhome is listening. The same happens in reverse when adguardhome replies. You (the user) don't need to care about the IP ranges on the internal docker networks. In fact, the rule of thumb you should adopt is, "any time you start thinking about IP addresses on internal docker networks, you're probably doing something wrong."

The arrangement you describe (where your router offers 192.168.1.9 for DNS resolution in DHCP leases) will continue to work. Definitely do NOT change your router to point to an IP address on an internal docker network!

Paraphraser, 2021-05-12:
In answer to your original question, no, you don't change anything (unless there's a conflict).

Your home network is 192.168.1/24. Under "old menu", docker allocated dynamically from 172.16.0.0/12. Under new menu, Slyke has selected fixed subnets in 10/8 (one of the reasons I don't like new menu and don't run it myself). As long as your home network doesn't collide with either of those (which it doesn't) there is no need to change anything.

If you see references to the "internal bridged network" don't read that as saying that the internal network is "bridged" to the external network and assume that that means they have to be on the same subnet.

What "bridging" means in this context is what happens on the internal network. It is bridged in the layer 2 switching sense. When one container (eg NodeRed) wants to talk to another container (eg Mosquitto) any unicast traffic goes container-to-container. Broadcast frames are still flooded to all containers, just like a physical layer 2 switch would do but, once the virtual MAC address of a container has been learned by the internal network, unicast frames only go to the container that needs them, just like a physical layer 2 switch. Make sense?

brunohenriquy, 2021-05-13:
Damn, thanks @Paraphraser . I've read the instructions before, but for some reason, my brain didn't absorb the port change to 8089 and all the rest hahahaha.

Here is what I was trying to do before you reply:

I changed the docker-compose to network_mode:host:
```
adguardhome:
    container_name: adguardhome
    image: adguard/adguardhome
    restart: unless-stopped
    environment:
    - TZ=Etc/UTC
  # enable host mode to activate DHCP server on ports 67/udp & 68/tcp+udp
  # note that you must also disable all other ports if you enable host mode
    network_mode: host
    # ports:
    # regular DNS
    # - "53:53/tcp"
    # - "53:53/udp"
    # administration port (http)
    # note: external and internal ports MUST be the same
    #       not active until defined via setup port
    # - "8089:8089/tcp"
    # HTTPS/DNS-over-HTTPS
    # - "443:443/tcp"
    # DNS-over-QUIC
    # - "784:784/udp"
    # DNS-over-TLS
    # - "853:853/tcp"
    # setup (http)
    # note: only active until port 8089 becomes active
    # - "3001:3000/tcp"
    # DNSCrypt
    # - "5443:5443/tcp"
    # - "5443:5443/udp"
```

Then it appeared(eth0: 192.168.1.9 wlan0:192.168.1.23):
[image]

But now I am following your instructions:

[image]

And you are right, despite of the **Setup Guide** tells to use this:
[image]

My router has primary DNS configured to 192.168.1.9 and its working:
[image]

Thank you so much!

brunohenriquy, 2021-05-13:
> In answer to your original question, no, you don't change anything (unless there's a conflict).
>
> Your home network is 192.168.1/24. Under "old menu", docker allocated dynamically from 172.16.0.0/12. [...]

Yeah, I got this as well. Thanks for the clarification.
Also, good to know you use the old menu, I'm moving to it xD!

Paraphraser, 2021-05-13:
To be more precise, I actually don't use either "menu". I started with old menu (late 2019) but then set about learning how it all hung together so, now, I just edit docker-compose.yml and get on with my life.

I have NOT adopted the "multiple networks" of new menu but I am using the new menu convention of keeping environment variables in docker-compose.yml rather than in "environment files" in the services directory.

If I had some reason to start from scratch today, and assuming that I know what I know about how it all works, I'd probably run new menu, let it generate my docker-compose.yml, then strip out all the networking crud so that I was back to the single `iotstack_default` internal network. That would be the fastest way to get all the environment variables into inline form. Starting by running old menu involves a lot of cutting and pasting to get the environment variables out of the environment files.

I think it's also true to say that master branch (new menu) gets more consistent maintenance. For example, when I put in the pull requests for AdGuardHome, I did two of them, one for old menu, one for master. There are no guarantees that everyone else will think to do the same. Even so, I only added the doco to master branch because adding it to both seemed like overkill. Slyke is working on a new-new menu and I live in hope we'll get back to a single useful working master branch, and all talk of menu variations will become a distant memory.

But you still won't catch me adopting multiple internal networks if those are part of new-new menu. Gives me the fidgets.

brunohenriquy, 2021-05-13:
Hahahahaha. Gotcha!
Well, It makes sense, you already have your environment all set up, so the menus are not that useful for you.
About networks, I feel your pain xD!!

Thanks again, my friend!! \o/

closed 2021-05-13 by brunohenriquy