SensorsIot/IOTstack #352: Run Backup reinstalls containers?

marceldarvas, 2021-05-22:
I was wondering why some of my apps like Homebridge or Home Assistant act odd.

Yesterday I used the menu to run a backup and it actually downgraded my homebridge version and shutdown my home assistant installed as supervisor

I really like this project but this stability makes me anxious.

@Slyke I've been watching your work, excited to see the new release!

Paraphraser, 2021-05-22:
Perhaps look at this [alternative backup](https://github.com/Paraphraser/IOTstackBackup).

It definitely won't downgrade anything and the backup doesn't stop the stack or take anything down.

The biggest "hole" i'm aware of is that the only database my solution backs up correctly is InfluxDB. If you're running MariaDB or PostgreSQL then the only safe way to capture those is to take down the container before starting the backup. Adding live-backup and restore for those two is on my to-do list.

I followed Graham Garner's lead in explicitly omitting Nextcloud. Aside from the "running database" issue, I assumed Graham's reasoning was that it was likely to grow to be "too big" and that it was up to the user to figure it out.

marceldarvas, 2021-05-23:
I've noticed that there were a couple different versions of images unlinked from containers, I assume that's what caused the the issues.

So I cleared them via the menu.

Going to look more into managing docker all together, too many apps are never simple to manage and backups 😅
But you're making me want to utilize InfluxDB more!

@Paraphraser 
Thank you for your activities as well, for supporting this project!

Paraphraser, 2021-05-23:
At the risk of telling you things you already know, there are three basic types of image/container to think about:

1. Images/containers that are in your `docker-compose.yml` and contain `image:` statements. Portainer, InfluxDB and Grafana are examples of this.
2. Images/containers that are in your `docker-compose.yml` and contain `build:` statements. Node-RED is an example of this and I'm hoping Mosquitto soon will be too (at the moment, Mosquitto is one of the above).
3. Images/containers that are not in your `docker-compose.yml`.

### Type 1

On a first-time install of IOTstack, these images are pulled down from DockerHub, and instantiated to become your running containers. Using WireGuard as the example, if you do a:

```
$ docker images
```

you'll see a pattern like:

```
REPOSITORY                      TAG         IMAGE ID       CREATED          SIZE
ghcr.io/linuxserver/wireguard   latest      229d2ef4682c   6 days ago       291MB
```

The TAG reflects its status. Tags are mostly "latest" but you might see "stable" or an explicit version number too.

You can update Type 1 containers just by doing a:  

```
$ docker-compose pull
```

Absent a "pin" to freeze an image at a particular version, any later image will come down from DockerHub. That new image will gain the tag while the older image will lose its tag and be marked "\<none\>":

```
$ docker images
REPOSITORY                      TAG         IMAGE ID       CREATED          SIZE
ghcr.io/linuxserver/wireguard   latest      637459cc342e   23 minutes ago   291MB
ghcr.io/linuxserver/wireguard   <none>      229d2ef4682c   6 days ago       291MB
```

Nothing about your running stack has changed. You can actually figure that out:

```
$ docker ps
CONTAINER ID   IMAGE         COMMAND  CREATED    STATUS     PORTS                                         NAMES
bb3fca0e153d   229d2ef4682c  "/init" 5 days ago  Up 5 days  0.0.0.0:51720->51820/udp, :::51720->51820/udp wireguard
```

Notice how the IMAGE column which is normally a name has been replaced with the IMAGE ID of the untagged image. When you do:

```
$ docker-compose up -d
```

the new image will be instantiated to become the running container, with the old container sent to the bit-bucket.

```
$ docker ps
CONTAINER ID   IMAGE                          COMMAND  CREATED        STATUS        PORTS                                          NAMES
91c0e4be0dd6   ghcr.io/linuxserver/wireguard  "/init"  7 seconds ago  Up 5 seconds  0.0.0.0:51720->51820/udp, :::51720->51820/udp  wireguard
```

The old (untagged) image can't be cleaned up until the old container has gone away (which makes sense; it'd be like pulling the rug out from under it). But, once the old container has gone, the old (untagged) image can be cleaned up with a:

```
$ docker system prune
WARNING! This will remove:
  - all stopped containers
  - all networks not used by at least one container
  - all dangling images
  - all dangling build cache

[... 87 lines cut ...]
```
$ docker rmi b3cc94b68658
```

Even then you frequently run afoul of messages like:

```
Error response from daemon: conflict: unable to delete b3cc94b68658 (must be forced) - image is being used by stopped container c08168d428d2
```

This means that the container that was instantiated from that image was stopped but not removed.

> With Type 1 and 2 containers, you're generally stopping your stack with a `down` and that implies both a `stop` and a `rm` (remove) of each container. 

A `prune` will get rid of stopped containers but you can also just use the container ID at the end of the error message:

```
$ docker rm -f c08168d428d2
```

and then re-do the `rmi`.

<hr>

Rolling all this together, you do have to run the various `pull`, `build` and `--build up` commands, keep a watch on `docker images`, run `prune` when needed, and occasionally get more heavy-handed with `rmi` and `rm -f`. The menu doesn't really deal with all of the subtleties of this.