SensorsIot/IOTstack #353: Don't see Hass.io (Supervisor)

Zwhouse2021, 2021-05-26:
Hello. I can't see the hassio(supervisor) in my HA. When I try to install hassio in iotstack menu it doesn't seem to be going right. That's what I see
[image]

Paraphraser, 2021-05-26:
Please see [the sudo command](https://sensorsiot.github.io/IOTstack/Getting-Started/#a-word-about-the-sudo-command). In short **stop** using `sudo` for everything. It isn't helping. The error you are getting about "bluetooth" is a **direct result** of misusing `sudo`.

> Please don't just skim-read that link. Please read it carefully so that you understand why chucking `sudo` at everything can do more harm than good.

Based on what I see in your screen shot, I strongly recommend the following:

```
$ cd ~/IOTstack
$ docker-compose down
$ sudo apt -y purge docker-ce docker-ce-cli containerd.io
$ sudo apt -y remove docker-compose
$ sudo pip3 uninstall docker-compose
$ sudo apt update
$ sudo apt upgrade -y
$ curl -fsSL https://get.docker.com | sh
$ sudo usermod -G docker -a $USER
$ sudo usermod -G bluetooth -a $USER
$ sudo apt install -y python3-pip python3-dev
$ sudo pip3 install -U docker-compose
$ sudo pip3 install -U ruamel.yaml==0.16.12 blessed
$ sudo reboot
```

The first two commands take the stack down (if it is running). The next three uninstall docker and docker-compose, the next two make sure your system is up-to-date, the next six install all the scaffolding docker and IOTstack need. The reboot is needed for the usermod commands to take effect. This is what the basic installer does under the hood.

If you **don't** remember installing the [recommended system patch](https://sensorsiot.github.io/IOTstack/Getting-Started/#recommended-system-patch), you should do that too.

If you **do** remember doing things like a `sudo git clone` or anything like that when you first installed IOTstack then you should seriously consider nuking the whole IOTstack folder and cloning again without using sudo. If you've got stuff you want to preserve, try moving the current folder out of the way so it is still around for you to retrieve stuff from, as in:

```
$ cd
$ sudo mv IOTstack IOTstack.old
$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
```

Then:

```
$ cd ~/IOTstack
$ ./menu.sh
```

I am not a Hass.io person so I don't really understand this but, as far as I can tell, all the "features" of supervisor are unnecessary when you run home_assistant as a container. The following screen shot is from [Discord](https://discord.com/channels/638610460567928832/638610461109256194/842967883369152593):

[image]

Certainly, when I've spun up the container, it just seems to work as is.

I am, however, running it like this:

```
  home_assistant:
    container_name: home_assistant
    image: homeassistant/raspberrypi4-homeassistant:stable
    restart: unless-stopped
    network_mode: host
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./volumes/home_assistant:/config
```  

<hr>

Having said that, if you really want a supervised install then see [Home Assistant installation](https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/#installation). Note that you will see many of the same commands as I mentioned above but with some extra commands to install Hass.io's dependencies and network manager in between uninstalling docker and reinstalling it again). I have never done this so I can't help. As soon as I see "network manager" I run for the hills.

Once you have all the prerequisites in place, you will find the supervised install in the menu under "Native Installs".

tablatronix, 2021-05-29:
From what I was reading the supervisor also spins up docker containers for addons, So I am not sure if that means integrations in the future or what or maybe it makes more advanced independent addons i the future with their own storage etc.

Zwhouse2021, 2021-06-08:
> sudo pip3 uninstall docker-compose

Thank you so much for your detailed explanation. I think I just didn't have network manager. I don't think I'm able to understand why supervisor isn't nessesary cuz discort article isn't written on noob language

closed 2021-06-08 by Zwhouse2021