SensorsIot/IOTstack #354: Image Versions

memjr, 2021-05-28:
Hi guys,

It seems the docker-compose file generated always uses the latest versions of whatever images exist at the time it first runs.  This can lead to problems when an environment is rebuilt and the images pulled have newer versions than the ones that were running before.

It would be useful to generate the compose file specifying the version being pulled at the first time, so if the environment needs to be rebuilt it will use the versions being used before.

Thoughts?

-memjr

Paraphraser, 2021-05-28:
My understanding is that if you use the menu to generate docker-compose.yml and then do an "up", the images pulled at that time are, with a few exceptions, the latest and greatest at that time.

Influx is an example of an exception. It's currently pinned to 1.8.4 (should be 1.8 - I should put in a pull request to fix that).

BUT, if you never do another `docker-compose pull` then images not built from local Dockerfiles (eg NodeRed) will never change. Of those that are built from Dockerfiles, the local image will only rebuild if you pass --build to the "up" command, and you have to explicitly call the `docker-compose build` command with additional options (see the NodeRed doco) to force an updated base image to come down from Dockerhub.

In short, **you** are in control of precisely if and when your images update. A "pull" does not have to be a global thing either. You can restrict it to just one container, eg `docker-compose pull telegraf`.

Paraphraser, 2022-06-19:
No further response. Should probably be considered stale.

closed 2022-09-11 by Slyke