SensorsIot/IOTstack #361: ruamel.yaml 503 error

MuhammadYacoub, 2021-06-08:
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ruamel.yaml==0.16.12
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/ruamel-yaml/ (Caused by ResponseError('too many 503 error responses'))

Failed to install ruamel.yaml and Blessed

Paraphraser, 2021-06-08:
I've been reading about Internet outages in the last few hours so that might explain it.

I already have these installed (and I'm not about to remove them just to prove a point) but I did try an update and that behaved normally. My advice is to run the command at the start of the following by hand and, if that works, you should be able to go back to where you were:

```
$ sudo pip3 install -U ruamel.yaml==0.16.12 blessed
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already up-to-date: ruamel.yaml==0.16.12 in /usr/local/lib/python3.7/dist-packages (0.16.12)
Requirement already up-to-date: blessed in /usr/local/lib/python3.7/dist-packages (1.18.0)
Requirement already satisfied, skipping upgrade: ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.9" in /usr/local/lib/python3.7/dist-packages (from ruamel.yaml==0.16.12) (0.2.2)
Requirement already satisfied, skipping upgrade: six>=1.9.0 in /usr/lib/python3/dist-packages (from blessed) (1.12.0)
Requirement already satisfied, skipping upgrade: wcwidth>=0.1.4 in /usr/local/lib/python3.7/dist-packages (from blessed) (0.2.5)
$ 
```

Paraphraser, 2021-06-08:
See [news report](https://www.abc.net.au/news/2021-06-08/websites-smh-nytimes-gov-amazon-affected-internet-outage/100200062).

closed 2021-06-15 by MuhammadYacoub