SensorsIot/IOTstack #375: Not really an issue... Orange Pi Win/Plus

BenMtl, 2021-06-27:
Did not see any other way to leave this comment. I am not sure you will care or not, but I have run the IOTstack script with out issue on one Orange Pi WinPlus so far.  There are no issues at all, it run great! I have a number of these board and a few other types of SBCs I will be trying the script on. If you would like to hear if it works or not let me know. If I hear nothing I will assume not.  

Thanks for your time

Bentley

Paraphraser, 2021-06-28:
Good to know. We could add it as a footnote in "getting started".

BenMtl, 2021-06-28:
That would be cool, being that the Orange pi runs Armbian for its OS I
would think that any board that uses it as a distro should be able to use
the script with out issue.

Paraphraser, 2021-07-13:
@BenMtl words added in [PR 283](https://github.com/SensorsIot/IOTstack/pull/382).

closed 2022-09-11 by Slyke