SensorsIot/IOTstack #376: Postgres passwords not set

berrySkies, 2021-07-01:
Added postgres to my existing stack build. Documentation states password provided in ./volumes/postgres/postgres.env

This file not created on installation ./volumes/postgres empty except for empty "data" directory

Following error message during installation:

    FileNotFoundError: [Errno 2] No such file or directory: './.templates/postgres/passwords.py'

Paraphraser, 2021-07-01:
The bad news is that this is an unfortunate side effect of the migration from old menu to new menu. The doco hasn't kept up. The worse news is that this problem is unlikely to get better during the migration to new-new menu.

The two main differences between old and new menu are:

* old stores environment variables in files while new puts them inline in `docker-compose.yml`; and
* old has a single default internal network while new has a plethora of internal networks.

Now, please study this:

```
$ cd ~/IOTstack/.templates/postgres/

$ git checkout old-menu
Switched to branch 'old-menu'
Your branch is up to date with 'origin/old-menu'.

$ cat postgres.env 
POSTGRES_USER=postuser
POSTGRES_PASSWORD=postpassword
POSTGRES_DB=postdb

$ git checkout master
Switched to branch 'master'
Your branch is up to date with 'origin/master'.

$ cat service.yml 
postgres:
  container_name: postgres
  image: postgres
  restart: unless-stopped
  environment:
    - POSTGRES_USER=postuser
    - POSTGRES_PASSWORD=%randomPassword%
    - POSTGRES_DB=postdb
  ports:
    - "5432:5432"
  volumes:
    - ./volumes/postgres/data:/var/lib/postgresql/data
  networks:
    - iotstack_nw
``` 

You'll see that, with one difference, what was in `postgres.env` in old-menu branch has moved into the service definition in the new menu on master branch. The difference is the right hand side of the password. The theory is that you asked the new menu to generate a password for you.

I hate "opaque" so I'd rather generate my own passwords (eg uuidgen, or a sha256 of some file, or a gpg --gen-random). Armed with the password of my choice, what I would do if I'd already spun-up the container and wanted to start from a clean slate:

```
$ cd ~/IOTstack
$ docker-compose stop postgres
$ docker-compose rm -f postgres
$ sudo rm -rf ./volumes/postgres

... edit docker-compose to set POSTGRES_PASSWORD=

$ docker-compose up -d postgres
```

Hope I've answered your question somewhere in all that.

berrySkies, 2021-07-01:
OK thanks, I was out of touch with all the changes. Will follow your approach.

closed 2021-07-01 by berrySkies