SensorsIot/IOTstack #377: Request: Use/include raw dockerfiles/docker-compose files

wizkidorg, 2021-07-07:
Use raw dockerfiles for all sub-services (e.g. portainer, node-red, influxdb, nginx, or any other service) that now can be platform independent without relying on a "pre-configured" state of the Raspberry Pi (e.g. specific username, and/or folder names).   If this is not possible, make a default dockerfile for base installs, and a dockerfile.template in each folder and have your script customize the template instead.  

This way all dependencies on having ANYTHING on the RPI other then a base OS and possibly docker is needed, e.g. is someone already has a RPI setup running existing containers.

I know this Stack was built for the novices out there to one click install, but most users of RPI who may use this stack already have an environment already setup, and dont want to install additional software into the base OS on the PI. (Also some, like me do not want python on our pi, or really any other language or tools in the base OS either)

Also if possible use docker version 2 config files, this way it can be utilized directly within portainer's WEBUI (or pulled from github) to build the stack. 

This will allow future RPI hardware builds and cross-platform to non-RPI devices alike.  Also will allow users to possibly change the base image used if necessary (e.g. if a maintainer drops support, security updates, etc) for selected services. 

You are basically doing it for the service.yml files (e.g. partial docker-compse.yml files).

Paraphraser, 2021-07-11:
> (Also some, like me do not want python on our pi, or really any other language or tools in the base OS either)

[image]

Paraphraser, 2021-07-11:
The rest of it - I don't understand.

In some cases, Dockerfiles are "necessary evils", such as for Mosquitto (so the container is IOTstack-friendly and will self-repair if its persistent storage area is removed) or Node-RED (because the only way to install some add-on nodes is via a Dockerfile). However, the downside of Dockerfiles **seems** to be that it makes life far more difficult than it should be. Where service definitions in `docker-compose.yml` use `image` directives,  a simple `docker-compose pull` will check for updates against all of those in one hit and download the latest and greatest. But for those service definitions that use `build` directives, it **seems** like the user just has to know when an update has been released (by going to Dockerhub) and then has to `docker-compose build --no-cache --pull CONTAINER` per container.

I can see arguments in favour of a bit more friction because applying updates willy-nilly is a great way to break something. But, on balance, the "just have to know" that **seems** to be implicit in Dockerfiles is something that is likely to cause more problems than it solves.

I've emphasised **seems** because there may be a solution to this I'm unaware of. Any ideas?

Nobody put me in charge of anything so this is just one person's opinion and, at that, only offered after nobody else more knowledgeable chimed in to give their perspective.

I also don't understand the version thing. If memory serves, the first release of gcgarner had version 3.6 and SensorsIot has inherited that. I know my own compose files work if I replace 3.6 with 2 but whether that generalises to all IOTstack users is unknown so, were I the decision-maker, it's not something I'd be leaping to do unless I had some assurance I wasn't going to be giving headaches to a whole bunch of unsuspecting IOTstackers. 😏

wizkidorg, 2021-07-14:
I understand the basis of the scripting, my only issue is it seems is that the stack has some "pre-defined" rules that may not play nice with some installs. e.g. assuming ~ is /home/pi, what user you are running this as, paths to   My argument around python is more about additional software installs into a base os, other than Docker or using scripts in python to build a customize stack install vs using the menu system that rpi-config or any other linux install uses by nature, which from my understanding is just sh scripting (other menu configs may use more/less external scripts but generally just sh). 

My goal is to have a portable IOTstack install which could be used outside of a pi (e.g. OrangePI, or Beagleboard, or even plain Debian), along with a simple yml, xml or similar config file that could be created/saved, etc. and it has a "purpose built" stack for deployment and could be deployed to mutliple pi's without needing the initial scripting to be run again. 

Python just appears to be the go-to language for any script, along with installing more in the base OS than what would be needed to run the containers. 

The issue with version 2 docker files is a portainer issue. In the web stack gui, it only supports version docker-compose files, its a problem with portainer. I have manually made many stacks customized to use it directly, and pull the docker images from either docker hub or an internal gitlab repository and build a stack from it. 

It was more of a feature direction vs an actual request. I will continue to pull the service.yml files and compose my own docker-compose builder from it.  (My usage is probably more customized then many of the users of this stack).  I also use NGINX in front of all the web enabled services as a proxy using letsencrypt to enable SSL. The biggest draw to this repository is the number of services which work on a raspberrypi in docker.  Many other solutions typically only work on real x86 hardware, and there is no alternative image for the install (the base image for many RPI builds has many limitations or changes which require the images to be overhauled before they will work. e.g. Base images from resin to balena which you now have to include an INIT service, etc.)

No worries, this issue can be closed.  Thank you for your time.

closed 2021-07-14 by wizkidorg