SensorsIot/IOTstack #378: NextCloudPi

fmcoulter, 2021-07-08:
Could we have the option of installing NextCloudPi? Collabora Online and CODE dont support the ARM processors on current NextCloud docker in IOTstack.

Paraphraser, 2021-07-08:
Normally, I would not comment on an issue where I didn't understand either the question or the terminology but I do have NextCloud running OK. See [this gist](https://gist.github.com/Paraphraser/4ea9b5ab17d9ba9febc22c27cadb0c54). That said, I'm not actually using NextCloud for anything productive - yet - so I haven't gone much further than getting it working and making sure I can connect to it from client devices. Maybe I'd hit the problems you're talking about if I tried to use it in anger.

My general approach when something doesn't seem to work out of the box (in the sense of starting from the IOTstack template) is to figure out why, run it for a while, write a gist in the hope of getting feedback along the lines of "x doesn't work" or "did you think about y?", and then, after a while, turn it into a pull request.

However, it sounds like you're well in advance of where I am so if you want something added to IOTstack and you are able to spin it up by hand (so you know it works) then maybe consider going the next step and [prepare a pull request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d). I'm sure @Slyke will consider it and either accept/reject on its IOTstackable merits.

Paraphraser, 2022-06-19:
No further response. Should probably be considered stale.

closed 2022-09-11 by Slyke