SensorsIot/IOTstack #390: Apparent Python Issue With ./menu.sh

CFWHISPERER, 2021-08-11:
I am trying to install IOTstack on a Raspberry Pi4 with a USB attached SSD (Samsung SSD T5). When I run ./menu.sh I get the following message.

"Python 3.6.9 or later (Current = 3.7.3), ruamel.yaml 0.16.12 or later      │
 │ (Current = not.not.not), blessed and pip3 are required for IOTstack to     │
 │ function correctly. Install these now?

After selecting Yes, I get the errors shown in the attached file, summarized here as ending with.

Aug 10 00:33:15 raspberrypi systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Aug 10 00:33:15 raspberrypi dphys-swapfile[2489]: want /var/swap=100MByte, checking existing: keeping it
Aug 10 00:33:15 raspberrypi dphys-swapfile[2512]: swapon: /var/swap: swapon failed: Invalid argument
Aug 10 00:33:15 raspberrypi systemd[1]: dphys-swapfile.service: Main process exited, code=exited, status=255/EXCEPTION
Aug 10 00:33:15 raspberrypi systemd[1]: dphys-swapfile.service: Failed with result 'exit-code'.
Aug 10 00:33:15 raspberrypi systemd[1]: Failed to start dphys-swapfile - set up, mount/unmount, and delete a swap file.
dpkg: error processing package dphys-swapfile (--configure):
 installed dphys-swapfile package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 dphys-swapfile
E: Sub-process /usr/bin/dpkg returned an error code (1)
Failed to install Python
[pythonError.txt](https://github.com/SensorsIot/IOTstack/files/6965027/pythonError.txt)

Paraphraser, 2021-08-11:
Short answer: no idea. My guess is missing dependencies.

Longer answer. You might find it helpful to study these resources:

1. Gist: [Installing Docker for IOTstack](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f)
2. Repo: [PiBuilder](https://github.com/Paraphraser/PiBuilder)

You might get out of jail quickly if you just follow:

* [Uninstalling Docker and Docker-Compose](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#-uninstalling-docker-and-docker-compose-); then
* [Scripting IOTstack installations](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#scripting-iotstack-installations).

Longer term, you'll get a quieter life from cloning PiBuilder, customising it to your needs, and using it as your build process.

As you study PiBuilder, you'll realise there are a lot more dependencies than any of the IOTstack installation processes take care of. One example is adding Buster Backports to get an updated `libseccomp2` which is needed if Docker images based on updated Alpine are to have any chance of working.

> By the way, I use Samsung T5 SSDs on my Pi4s too.

CFWHISPERER, 2021-08-11:
Thanks I will look into these and respond with what happens.

PR #560 (2022-09-11): menu: run in python virtualenv 
* manage python requirements using requirements-menu.txt
* more reliable (installs exact versions of all dependencies)
* is completely separate from any system-wide installed python packages,
  won't conflict with existing installed python packages.
* minimizes global system requirements down to python and python-virtualenv

virtualenv requires "--seed pip" or pip will fail with:
`ModuleNotFoundError: No module named 'pip._internal.utils'`
If anyone knows why, I'd be interested to hear.

Fixes  #499, Fixes #390
closed 2022-09-11 by Slyke