SensorsIot/IOTstack #409: Mosquitto: "temporary error" during build

Paraphraser, 2021-09-21:
## Problem

If you are reading this, it is probably because you have just tried to build Mosquitto and it has failed with an error pattern like this:

```
Step 1/9 : FROM eclipse-mosquitto:latest
Step 2/9 : RUN apk update && apk add --no-cache rsync tzdata
fetch https://dl-cdn.alpinelinux.org/alpine/v3.14/main/armhf/APKINDEX.tar.gz
ERROR: https://dl-cdn.alpinelinux.org/alpine/v3.14/main: temporary error (try again later)
```

This seems to be a known issue. It is not an IOTstack problem and neither is it a Mosquitto problem. It's inherited from Alpine Linux. It was first reported in July 2020. See [Docker Alpine issue 98](https://github.com/alpinelinux/docker-alpine/issues/98).

It is not clear why it has suddenly started causing problems for IOTstack Mosquitto container builds. Neither is it clear why it only affects some IOTstack users.

## Solution

The fix is to add these lines to Mosquitto's Dockerfile:

```Dockerfile
# see https://github.com/alpinelinux/docker-alpine/issues/98
RUN sed -i 's/https/http/' /etc/apk/repositories
```

The completed Dockerfile should look like this:

```Dockerfile
# Download base image
FROM eclipse-mosquitto:latest

# see https://github.com/alpinelinux/docker-alpine/issues/98
RUN sed -i 's/https/http/' /etc/apk/repositories

# Add support tools
RUN apk update && apk add --no-cache rsync tzdata

# where IOTstack template files are stored
ENV IOTSTACK_DEFAULTS_DIR="iotstack_defaults"

# copy template files to image
COPY --chown=mosquitto:mosquitto ${IOTSTACK_DEFAULTS_DIR} /${IOTSTACK_DEFAULTS_DIR}

# replace the docker entry-point script
ENV IOTSTACK_ENTRY_POINT="docker-entrypoint.sh"
COPY ${IOTSTACK_ENTRY_POINT} /${IOTSTACK_ENTRY_POINT}
RUN chmod 755 /${IOTSTACK_ENTRY_POINT}
ENV IOTSTACK_ENTRY_POINT=

# IOTstack also declares these paths
VOLUME ["/mosquitto/config", "/mosquitto/pwfile"]

# EOF
```

## Pull Requests

I have filed Pull Requests to implement this fix:

* [PR 406](https://github.com/SensorsIot/IOTstack/pull/406) for "master" branch (aka "new menu").
* [PR 407](https://github.com/SensorsIot/IOTstack/pull/407) for "old-menu" branch.
* [PR 408](https://github.com/SensorsIot/IOTstack/pull/408) for "experimental" branch (aka "new-new menu".

## Interim fix

1. Edit the following file (`sudo` is not required):

	```
	~/IOTstack/.templates/mosquitto/Dockerfile
	```

2. Add the two lines as above.

3. Rebuild Mosquitto:

	```bash
	$ cd ~/IOTstack
	$ docker-compose build --no-cache --pull mosquitto
	$ docker-compose up -d mosquitto
	$ docker system prune -f
	$ docker system prune -f
	```
	
4. Later, when the Pull Requests are applied, you will need to undo your local changes before Git will let you update against GitHub:

	```bash
	$ cd ~/IOTstack
	$ git checkout -- .templates/mosquitto/Dockerfile
	$ git pull
	```

## One more thought…

While you are fixing things, make sure you have implemented both [recommended system patches](https://sensorsiot.github.io/IOTstack/Getting-Started/#recommended-system-patches).

The second patch (libseccomp2) is directly relevant to Mosquitto. See [Issue 401](https://github.com/SensorsIot/IOTstack/issues/401) for more information.

senarvi, 2021-09-25:
Surprisingly, this didn't fix the build for me:

    fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/main/armhf/APKINDEX.tar.gz
    ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.14/main: temporary error (try again later)
    WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.14/main: No such file or directory
    fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/community/armhf/APKINDEX.tar.gz
    2 errors; 20 distinct packages available
    ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.14/community: temporary error (try again later)
    WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.14/community: No such file or directory

I don't get it. I'm able to access the URL with a browser.

Paraphraser, 2021-09-25:
I don't get it either.

```
Building mosquitto
Sending build context to Docker daemon  9.216kB
Step 1/10 : FROM eclipse-mosquitto:latest
latest: Pulling from library/eclipse-mosquitto
Digest: sha256:ce08d3fe69d4170cea2426739af86ac95e683f01dd2c4141da661983a2401364
Status: Image is up to date for eclipse-mosquitto:latest
 ---> 24a85c54a50e
Step 2/10 : RUN sed -i 's/https/http/' /etc/apk/repositories
 ---> Running in 622915d96dfa
Removing intermediate container 622915d96dfa
 ---> c332383971ae
Step 3/10 : RUN apk update && apk add --no-cache rsync tzdata
 ---> Running in d7f0eebee992
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/main/armhf/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/community/armhf/APKINDEX.tar.gz
v3.14.2-52-g922bf2dee7 [http://dl-cdn.alpinelinux.org/alpine/v3.14/main]
v3.14.2-58-gdbf551a5fb [http://dl-cdn.alpinelinux.org/alpine/v3.14/community]
OK: 13102 distinct packages available
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/main/armhf/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/community/armhf/APKINDEX.tar.gz
(1/5) Installing libacl (2.2.53-r0)
(2/5) Installing popt (1.18-r0)
(3/5) Installing zstd-libs (1.4.9-r1)
(4/5) Installing rsync (3.2.3-r4)
(5/5) Installing tzdata (2021a-r0)
Executing busybox-1.33.1-r3.trigger
OK: 14 MiB in 25 packages
Removing intermediate container d7f0eebee992
 ---> b5e2d0baab6c
Step 4/10 : ENV IOTSTACK_DEFAULTS_DIR="iotstack_defaults"
 ---> Running in 738992f2e418
Removing intermediate container 738992f2e418
 ---> e3d1202fdb2c
Step 5/10 : COPY --chown=mosquitto:mosquitto ${IOTSTACK_DEFAULTS_DIR} /${IOTSTACK_DEFAULTS_DIR}
 ---> 14469f815067
Step 6/10 : ENV IOTSTACK_ENTRY_POINT="docker-entrypoint.sh"
 ---> Running in a94f90fbcc9e
Removing intermediate container a94f90fbcc9e
 ---> 9bfd1bac133c
Step 7/10 : COPY ${IOTSTACK_ENTRY_POINT} /${IOTSTACK_ENTRY_POINT}
 ---> 6a19f7be27bf
Step 8/10 : RUN chmod 755 /${IOTSTACK_ENTRY_POINT}
 ---> Running in ad355035cec0
Removing intermediate container ad355035cec0
 ---> fe01c40bca83
Step 9/10 : ENV IOTSTACK_ENTRY_POINT=
 ---> Running in 23e63c8aec3d
Removing intermediate container 23e63c8aec3d
 ---> 6331aff9e195
Step 10/10 : VOLUME ["/mosquitto/config", "/mosquitto/pwfile"]
 ---> Running in 9333dc802fc7
Removing intermediate container 9333dc802fc7
 ---> 172b8de07052
Successfully built 172b8de07052
Successfully tagged iotstack_mosquitto:latest
```

senarvi, 2021-09-30:
I tried it a few times but I'm totally out of ideas. I had to downgrade to eclipse-mosquitto:2.0.11.

Paraphraser, 2021-10-01:
This problem is downright annoying! Annoying for people like you at their wits' end. Annoying for people like me who are trying to figure out why people like you are having trouble that I can't replicate. 🤬🤬🤬

After a bit more Googling, I found these:

* Docker forums [Issue 77838 alpine fails to run apk](https://forums.docker.com/t/docker-build-with-alpine-fails-to-run-apk/77838)
* Geek Bacon [alpine dns problem temporary error](https://geekbacon.com/2020/05/08/docker-alpine-3-5-dns-problem-temporary-error/)

To summarise what I get from both of those:

1. IPv6 *could* be part of the problem. That implies that turning off IPv6 might be the cure.
2. The DNS *could* be part of the problem. It doesn't get into specifics but I take it to mean that the resolvers ISPs provide by default might be suspect (perhaps the way they have been known to intercept DNS requests to feed back advertising). The Geek Bacon URL above suggests sending Docker straight to 8.8.8.8 as the cure.

In effect, I'm already implementing both of those suggestions so either or both **could** explain our different experiences. Here's the background:

1. I routinely clobber IPv6 as soon as I see it. I don't have it running on any of my Pis.
2. I have a local upstream DNS (BIND9 running on a macOS box). Every Pi points to the Mac. The Mac is authoritative for my local domain. Anything the Mac can't answer, is handled like this:

	```
	forwarders {8.8.8.8; 8.8.4.4; 1.1.1.1; };
	```

	That means off-net queries are being directed to those servers in round-robin fashion.
	
I'd be curious to know whether you have IPv6 enabled and how your DNS is set up?

To clobber IPv6:

1. Use `sudo` and your favourite Unix text editor to open:

	```
	/etc/sysctl.conf
	```
	
2. Find the line:

	```
	#net.ipv6.conf.all.forwarding=1
	```
	
3. After that line, insert these lines:

	```
	net.ipv6.conf.all.disable_ipv6=1
	net.ipv6.conf.default.disable_ipv6=1
	net.ipv6.conf.lo.disable_ipv6=1
	```

4. Save the file and reboot. Some guides suggest `sudo sysctl -p` but the problem with that is if your SSH connection was established over IPv6 - it pulls the rug from under your current session (which hangs) and it can be a pain to convince your local host that it needs to try via IPv4. A reboot is much cleaner.

How you convince Docker to use 8.8.8.8 will depend on your arrangements:

1. If you have your own local DNS server, add a forwarders rule like mine.
2. You can usually tell your router where to direct queries. Typically, it's in the DHCP setup. In my case, the router is also my DHCP server so it tells DHCP clients to send their DNS queries to my local DNS server. If I didn't have a local DNS server, I would change the router to tell DHCP clients to use 8.8.8.8 and 8.8.4.4.
3. The suggestion in the two URLs above, augmented with my own recommendations:

	1. Take your stack down.
	2. Use `sudo` and your favourite Unix text editor to create:
	
		```
		/etc/docker/daemon.json
		```

	3. Add the text:

		```
		{
		    "dns":["8.8.8.8"]
		}
		```
		
	4. Save the file and reboot.
	5. Bring the stack up. 

I've added the down/reboot/up because experimentation on my own system suggests Docker can get into a mess if you try to convince it to do any of this dynamically (eg with `systemctl` commands).

The above `daemon.json` patch works on my test Pi in the sense of the containers coming up again without Docker doing any serious dummy-spits. I think it is also working, technically, in that if I go into my Node-RED container (where I have DNS tools installed) and try a `dig` to a local domain name (one that will be resolved by the Mac but is unknown to 8.8.8.8), there's a change in behaviour:

dig       | unpatched Pi | patched Pi
----------|--------------|-------------
@local DNS| succeeds     | succeeds
@8.8.8.8  | fails        | fails
undirected| succeeds     | fails

That pattern implies that the undirected queries are being sent to 8.8.8.8 on the patched Pi, and the `daemon.json` patch is the only possible explanation for the difference.

Running this test from Node-RED also makes the point that the `daemon.json` patch doesn't just affect Mosquitto Dockerfile builds but **all** of Docker. I couldn't find any way of passing a "use 8.8.8.8 for this Mosquitto build". So, even if this does solve your problem, you might need to get into the game of turning it on and off when you want to build Alpine containers with local Dockerfiles. 

And then, of course, we're still left with the other hint in [Alpine issue 98](https://github.com/alpinelinux/docker-alpine/issues/98) that having a proxy between you and the Alpine repos could be part of the problem.

I'm still scratching my head on all of this. I hope this helps you make some progress.

senarvi, 2021-10-03:
Don't worry. I'm happy to use the 2.0.11 for now and hope it gets fixed for the next version. The impression I got from the conversation is that there's a bug in libfetch or some other library that makes HTTPS to fail - maybe IPv6 too. But for the sake of understanding what's going on, I tried what you suggested.

I have DNS set up using DHCP. It creates `resolv.conf` automatically, with the name servers of my ISP. I don't have a private DNS server. I don't understand why using 8.8.8.8 would make any difference, but I tried creating the `/etc/docker/daemon.json`. It didn't help.

I'm using IPv6. I tried disabling it with your instructions. I'm not sure if that really disabled IPv6. This command indicates that it did:

```bash
$ sysctl -a 2>/dev/null | grep disable_ipv6
net.ipv6.conf.all.disable_ipv6 = 1
...
```

But this command seems to indicate that it didn't:

```bash
$ cat /sys/module/ipv6/parameters/disable
0
```

Anyway, it seems that this somehow messes up the network configuration - there's a noticeable lag in my connection to the Pi when I do this. And it didn't work either.

Paraphraser, 2021-10-04:
> Don't worry. I'm happy to use the 2.0.11

However, if you do notice it getting fixed, I'd appreciate it if you'd add a comment to this post, please.

<hr>

>  I don't understand why using 8.8.8.8 would make any difference

In principle, I agree. It shouldn't.

I think the issue with 8.8.8.8 vs ISP-supplied is the theory that ISPs have been known to intercept DNS traffic so they can try to horn-in on Google's act of selling and supplying tailored ads. Some of the reading I've done (eg The Wire and Ars Technica) suggests that those interception systems can get in the way.

A secondary issue can occur from the propagation delays that are inherent in the "distributed database" nature of the DNS. As an example, I showed you my "forwarders" rule but I also have told BIND to send queries for special domains straight to the servers that are authoritative for those domains. Here's an example:

```
zone "hopto.org" in {
    type forward;
    forwarders {194.62.182.53; 45.54.64.53; 204.16.253.53; 194.62.183.53; };
};
```

I'm signed-up with No-IP.com for dynamic DNS. My domain name with them is a sub-domain of `hop-to.org`. The servers in that list will be the first to find out when my ISP changes the IP address of my router's WAN interface so that's the best place to look for authoritative information.

In general, the big, beefy DNS hosts like 8.8.8.8 and friends tend to discover updates more quickly than the "PC under the desk" of a small corner-store ISP. Presumably, most ISPs are somewhere between those extremes.

Nevertheless, I think this only really matters for things that change a lot like dynamically-allocated router WAN interfaces. I doubt that it would be a concern in the situation we're talking about.

Still, it has always seemed to me that the error is coming from the target host rather than being a DNS problem, per se. It was simply other people saying "use 8.8.8.8" and me thinking "I'm already using 8.8.8.8" that made me wonder if it was somehow relevant.

Querying the A record for the service returns four addresses:

```
$ dig +short dl-cdn.alpinelinux.org
dualstack.d.sni.global.fastly.net.
151.101.194.133
151.101.66.133
151.101.2.133
151.101.130.133
```

If I try iterating those IP addresses **directly** with `wget` then the other end complains with "500 Domain Not Found", which I take to mean there's a reverse proxy in the middle matching on some/all of "dl-cdn.alpinelinux.org".

I mostly do these drill-downs into problems other people are having because I always wind up learning a lot. But there comes a point where I say "can of worms", shrug my shoulders and move right on by. I think I've reached that point with this.

<hr>

>  I tried disabling it with your instructions. I'm not sure if that really disabled IPv6

Hmmm...

A test on my running system. First, show that the mechanism I use for disabling IPv6 is in place:

```
$ grep "^net.ipv6" /etc/sysctl.conf
net.ipv6.conf.all.disable_ipv6=1
net.ipv6.conf.default.disable_ipv6=1
net.ipv6.conf.lo.disable_ipv6=1 
```

Second, show that I get the same result as you do on your test:

```
$ cat /sys/module/ipv6/parameters/disable
0
```

Third, ask the Ethernet and WiFi interfaces what they think. Note the presence of `inet` in the second line of output from each interface (IPv4) but the **absence** of any line starting with `inet6`. 

```
$ ifconfig eth0
[... 86 lines cut ...]
The hosts "iot-hub", "sec-dev", "new-dev" and "octopi" are Raspberry Pis with IPv6 disabled:

* "octopi" is a WiFi-connected Raspberry Pi 3B+ running from an SD card.
* The others are Ethernet-connected Raspberry Pi 4s running from USB3 SSDs.

The comms path to the Raspberry Pi 4s is Mac Ethernet to gigabit Ethernet switch to Pi. For the 3B+, there are two extra hops via a WiFi access point.

Running `ssh` with a command argument does the equivalent of connect, login, run the command, logout, disconnect. Each target host name (eg "iot-hub") matches an entry in `~/.ssh/config` which maps like this:

```
host iot-hub
  hostname iot-hub.my.domain.com
```

Thus `ssh iot-hub` will imply a trip through the resolver but I think we can safely assume that all those DNS names were cached in the local resolver and didn't result in DNS queries going to my local DNS server.

All devices (Mac and Pis) are synching with the same NTP server so they should have the same time to within a millisecond or so.

> The "date" command on the Pis does have the ability to show finer resolution than one second (`%N`) but macOS lacks that ability for some reason known only to Apple. Rather than try to find a way around that deficiency, I just went with what I had. 

Making all due allowances for when each command is issued within the span of the "current second", I read the results as saying that the Pi4s are capable of a round-trip response in under ⅓ of a second (the 11:35:00 and 11:35:01 batches in particular) while the 3B+ does seem to take a little bit longer. Whether that's down to WiFi, the slower 3B+ or the SD card is unknown.

I don't call the implied wet-finger estimate of "about ½ of ⅓ of a second" to just do a routine "connect and login" particularly slow but, then, I don't have your experience with IPv6 so maybe it really is considerably more snappy.

Conversely, if disabling IPv6 results in you going from sub-second to significantly longer times then the test results above will show that IPv4 really can hold its own and that you might want to look for another cause.

senarvi, 2021-10-05:
> However, if you do notice it getting fixed, I'd appreciate it if you'd add a comment to this post, please.

Sure, and at least I can repeat the problem, so I if you come up with a solution, I can try it.

> I think the issue with 8.8.8.8 vs ISP-supplied is the theory that ISPs have been known to intercept DNS traffic so they can try to horn-in on Google's act of selling and supplying tailored ads. Some [...]

Thanks. I can now understand the logic behind this.

> Querying the A record for the service returns four addresses:
>
> ```
> $ dig +short dl-cdn.alpinelinux.org
> dualstack.d.sni.global.fastly.net.
> 151.101.194.133
> 151.101.66.133
> 151.101.2.133
> 151.101.130.133 [...]

Indeed sounds like the server is behind a reverse proxy. However, oddly enough I see only one A record, and it's none of those that you listed:

```
$ dig dl-cdn.alpinelinux.org

; <<>> DiG 9.16.1-Ubuntu <<>> dl-cdn.alpinelinux.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 40882
;; flags: qr rd ad; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 0
;; WARNING: recursion requested but not available

;; QUESTION SECTION:
;dl-cdn.alpinelinux.org.                IN      A

;; ANSWER SECTION:
dl-cdn.alpinelinux.org. 0       IN      CNAME   dualstack.d.sni.global.fastly.net.
dualstack.d.sni.global.fastly.net. 0 IN A       151.101.86.133

;; Query time: 0 msec
;; SERVER: 172.30.224.1#53(172.30.224.1)
;; WHEN: Mon Oct 04 20:45:49 EEST 2021
;; MSG SIZE  rcvd: 158
```

> Third, ask the Ethernet and WiFi interfaces what they think. Note the presence of `inet` in the second line of output from each interface (IPv4) but the **absence** of any line starting with `inet6`.

I verified with ifconfig that I'm able to disable IPv6. This sounded like a reasonable explanation, but no help from disabling IPv6.

> I don't understand why IPv4 should be any slower than IPv6.

It's definitely not that IPv4 is so slow. I think I broke something, since I was seeing random delays of up to a few seconds after every key press. Anyway, since disabling IPv6 didn't help anything, I'll just keep it enabled.

Another idea that I had is that I could debug this by running commands in the Dockerfile. There's no dig in the Docker image, so I tried nslookup. I got this error:

    nslookup: clock_gettime(MONOTONIC) failed

If I run `wget http://dl-cdn.alpinelinux.org/alpine/v3.14/community/armhf/APKINDEX.tar.gz`, I get:

    wget: bad address 'dl-cdn.alpinelinux.org'

Actually it seems that I'm not able to access anything with wget from the build script.

The first error message led me to a couple of threads:

https://github.com/haugene/docker-transmission-openvpn/issues/1726
https://github.com/haugene/docker-transmission-openvpn/issues/1564

There also people suggest setting the DNS server manually. I tried several things to no avail, including setting the DNS server in resolv.conf and various settings in docker-compose.yml:

    dns:
    - 8.8.8.8
    cap_add:
    - NET_ADMIN
    sysctls:
    - net.ipv6.conf.all.disable_ipv6=0

Paraphraser, 2021-10-06:
I think I'll get too confused if I try to provide comments and suggestions via both your IOTstack and Alpine posts so I'll stick with IOTstack.

Also, I've been travelling today so half of this has been written on the road and is a bit scatter-brained as a result. Sorry.

<hr>

There are two lines in your `dig` output that trouble me:

```
$ dig dl-cdn.alpinelinux.org
…
;; WARNING: recursion requested but not available
…
;; SERVER: 172.30.224.1#53(172.30.224.1)
```

I'll deal with the `SERVER` line first. I don't know your network setup so I might be wrong about this but seeing 172.30.224.1 makes me itch. It *suggests* (a long way short of *proof*) one of two things:

* Because the .1 address is normally reserved for the router on the internal subnet, the intention might be to direct queries to a DNS server *running in a container*; or
* Somehow .1 has been allocated to a container (like PiHole) running as a DNS server and queries are being directed to it.

Assuming this does indicate queries being directed to a containerised DNS, I'll talk about it.

I don't do that myself. I know opinions vary but I have really deep misgivings about wisdom of a Pi (or any computer) nominating itself for DNS resolution when the DNS is running in a container.

Experience has taught me that that last sentence is often misinterpreted so I'll spell it out. I'm talking about a very specific combination of elements:

* A host.
* The host is running Docker.
* Docker is running a DNS service (like PiHole) in a container.
* The `/etc/resolv.conf` for that host is using its own IP address (or the loopback address, or some other "get me to the container" mechanism) for DNS.

A lot of wacky stuff goes on with the DNS in container-space and people who behave as though the DNS is not a special case seem somewhat over-represented in Discord questions and GitHub issues.

Assuming I've guessed correctly about what 172.30.224.1 represents, I don't know whether we're talking PiHole, AdGuardHome or BIND running in a container. For the moment, I'm going to assume it is PiHole but the same comments apply to any ad-blocker.

I look at it like this. Does the Pi itself **need** ad-blocking services?

Unless you're also trying to use the Pi as a general-purpose computer, I reckon the answer is usually "no". Ads only bother human eyeballs. You typically want ad-blocking on the devices where you routinely fire up browsers. It's those **other** devices that should point to the Pi for their DNS, and then Docker can NAT-forward the traffic to the ad-blocker. That works and works well.

There is also no harm in **other** devices using something like PiHole for general DNS services such as being authoritative for a local domain.

But – if you want the Pi to use **itself** for **any** DNS services (ad-blocking and/or general DNS), then it really shouldn't run in a container. It really should be a native install. That guarantees that (a) the service comes up early (a Docker container arrives pretty late in the scheme of things), (b) there's no NAT nonsense getting to/from container-space, (c) no backdoor routes between containers to worry about, and (d) no Docker hocus-pocus with 127.0.0.11.

Just one person's opinion, of course.

\</soapbox\>

Moving on.

The `WARNING` line message looks like trouble. There's an [explanation](http://www.zytrax.com/books/dns/ch2/index.html#recursive) that might help you understand the context. In general, the first DNS server your query reaches should always support recursion. Servers further away need not support recursion but might. I've always taken the view that if the reason a server isn't supporting recursive queries isn't obvious then it's probably a mistake.

I can't explain why you are getting that warning. I can't explain what it actually means in your environment. I can't say for certain if it is the root cause of your problem. All I can say is that that message, and the fact that you only get one A record back, makes me itch.

For the sake of comparison, though, here's the result of me running the same query, three times, on my Raspberry Pi which is running PiHole in a Docker container. The first query:

* is **directed** to the IP address of the Pi. That forces it to go to PiHole by virtue of Docker listening to port 53. PiHole will forward it to 8.8.8.8 and friends because that's what I've told PiHole to do with anything it doesn't block and isn't in my local domain.
* is **directed** to the IP address of the router on the internal bridged network. That also forces it to go to PiHole which will forward it as above.
* is **undirected**. That causes it to follow `/etc/resolv.conf` meaning that the query will be answered by my local upstream DNS (BIND running on macOS). BIND will forward it to 8.8.8.8 and friends. Again, that's what I've told BIND to do with any queries it can't answer.

```
$ dig @$(hostname -I | awk '{print $1;}') dl-cdn.alpinelinux.org

; <<>> DiG 9.11.5-P4-5.1+deb10u5-Raspbian <<>> @192.168.203.60 dl-cdn.alpinelinux.org
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 52741
;; flags: qr rd ra; QUERY: 1, ANSWER: 5, AUTHORITY: 0, ADDITIONAL: 1

[... 182 lines cut ...]
      - WEBPASSWORD=wouldntuliketoknow
      - INTERFACE=eth0
      - REV_SERVER=true
      - REV_SERVER_DOMAIN=my.domain.com
      - REV_SERVER_TARGET=192.168.132.65
      - REV_SERVER_CIDR=192.168.132.0/24
    ports:
      - "53:53/tcp"
      - "53:53/udp"
[... cut ...]
      - NET_ADMIN
```
The "REV\_SERVER" vars mean "if it's a name-to-address query where name is in my.domain.com **or** it's an address-to-name query in the 192.168.132.0/24 subnet, forward the query to 192.168.132.65".

I have to admit that I've never really thought about the `dns:` directives. The PiHole GUI is set to round-robin 8.8.8.8, 8.8.4.4 and 1.1.1.1. I watched traffic for a while so I know it's doing that.

In context, that 127.0.0.1 should mean "this container". But why it's there ... dunno.

AkameSenshi, 2021-11-01:
Tried everything, then had a light bulb moment and switched from my router wifi to phone hotspot (using cellular data) and worked like a charm.

poseidogithub, 2021-12-28:
> Tried everything, then had a light bulb moment and switched from my router wifi to phone hotspot (using cellular data) and worked like a charm.

Well, unfortunately this didnt work for me neither. same error. (Ihave a pi-hole on another raspberry) 

as I'm quite dumb, I'll try to a clean istall, maybe having mosquitto installed on the rraspbian without docker before it's causing issues...

cstoker2, 2022-01-16:
I had same issue, updated libseccomp2 per the "Getting started" instructions without any luck. After _also_ updating docker to v20+ per https://blog.samcater.com/fix-workaround-rpi4-docker-libseccomp2-docker-20/ it installs fine.

closed 2022-03-03 by Paraphraser