SensorsIot/IOTstack #410: TLS support for all services

marcelstoer, 2021-09-23:
I was wondering...would IOTstack somehow be able to facilitate TLS connections to defined services?

Two examples:
- (The) Home Assistant (dashboard) has limited functionality if not accessed over HTTPS
- ESPHome in Docker can't support flashing binaries to nodes in the browser if not accessed over HTTPS\*

Wouldn't it be cool if I could tell IOTstack to do TLS termination even if that would mean using self-signed certificates? Spin up a reverse proxy that does the necessary forwarding/mapping?

\* it uses the browser Serial API which browsers only allow for HTTPS sites

Noschvie, 2021-09-24:
#160

ukkopahis, 2022-06-18:
Home assistant documentation now includes how to setup SWAG: https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/#httpsWithSSLcert

Should be adoptable to also work for ESPHome.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke