SensorsIot/IOTstack #422: Configuration to route Wireguard clients through Pihole

Nurgak, 2021-10-05:
I set up a stack with Pihole and Wireguard. Pihole works great while I'm at home. The VPN fires up as soon as I'm out of my home network, but it seems Pihole is not there to block the ad traffic, at least I do not see the Pihole client list.

Can anybody confirm that Wireguard traffic is not routed through Pihole by default (why would it, some people do not want Pihole...).

My main question is: what should I change, I guess in `docker-compose.yml`, to have my Wireguard traffic routed through Pihole? The [documentation does go into the details](https://sensorsiot.github.io/IOTstack/Containers/WireGuard/#optional-configuration-dns-resolution-for-peers), but it's not clear enough for me.

I have enjoyed this setting before, natively without the IOTStack, and I absolutely loved it. IOTStack makes it so convenient to keep everything up to date without breaking, so I prefer switching to that. Before, I just had to change the DNS address in the Wireguard client to that of my RPi, now this doesn't work any more.

Paraphraser, 2021-10-05:
Use the PEERDNS environment variable. It defaults to "auto" but if you set it to the IP address of the host running the PiHole container, it should work. At least, that's how I do it and it seems to work. The only caveat I'll put on that is that the host running my WireGuard container and the host running my DNS service are not the same physical device. I've never tried doing everything on the same device but I can't see any reason why it should be any different.

Be aware, though, that the PEERDNS variable makes its way into the peer configurations so, when you change it and "up" the stack, the new container will regenerate the peer configurations and you'll have to set up new tunnels with the new QR codes on each device.

Nurgak, 2021-10-06:
I was not successful. I tried different values for the `PEERDNS`: the device local IP, `127.0.0.1` and `pihole`, but none worked (DNS addresses were not resolved). Only `10.13.13.1` worked (default value when `PEERDNS` is set to `auto`). I restarted the stack every time and replaced the tunnel on my device using the newly generated QR code. I could confirm, that the DNS address changed on the client.

> Be aware, though, that the PEERDNS variable makes its way into the peer configurations so, when you change it and "up" the stack, the new container will regenerate the peer configurations and you'll have to set up new tunnels with the new QR codes on each device.

Does the `PEERDNS` really have other effects than defining the client DNS address? I can change that in the client to anything I want...

I guess the *right* way to do it is to set the DNS address on the router to the device running Pihole. That creates a sort of a loop and catches everything. Unfortunately, my potato of a modem/router provided by my ISP does not allow that (luckily I can still do port forwarding).