SensorsIot/IOTstack #424: Node-red on Orange Pi Lite & /dev/ttyAMA0

MaxPetukhov, 2021-10-10:
I tried IOTstack on Orange Pi Lite, everything seems to work. But when installing Node-red, I have an error:
ERROR: for nodered  Cannot start service nodered: error gathering device information while adding custom device "/dev/ttyAMA0": no such file or directory

Since I don 't have "/dev/ttyAMA0"

buseng, 2021-11-09:
Linux MINT 
*************
For practice I installed IOTstack on my system: 
*  Host: Dell-i5 Kernel: 5.4.0-89-generic x86_64 bits: 64 compiler: gcc v: 7.5.0 
*  Desktop: Cinnamon 4.4.8 wm: muffin dm: LightDM 
*  Distro: Linux Mint 19.3 Tricia base: Ubuntu 18.04 bionic 

It used to work with the old repository. After upgrade to SensorsIot/IOTstack/ I encountered the same issue with nodered concerning "/dev/ttyAMA0": no such file or directory.

Modifying the /IOTstack/.templates/nodered/service.yml  file in the same way as mentioned above for the nodered section of the docker-compose.yml  does keep node-red functional after modifications of the IOTstack inside menu.sh.

Menu.sh modyfies the docker-compose.yml using the templates.

Paraphraser, 2022-06-19:
Different platform.Should probably be considered stale.

closed 2022-09-11 by Slyke