SensorsIot/IOTstack #439: Backup script issues

Roberto6969, 2021-11-06:
Is backup script still in order with IOTstack project? After "all default" and everything is OK, backup script doesn't do anything useful, neither backup option from ./menu.
Just asking, before further investigation or modification of backup script. Backup script is always important more than any other.

Current Directory: /home/pi/IOTstack/scripts
Backup Type: 3
Backup File: ./backups/backup/backup_2021-11-06_1937.tar.gz
Rolling File: ./backups/rolling/backup_6.tar.gz

Executing prebackup scripts
bash: ./scripts/backup_restore/pre_backup_complete.sh: No such file or directory
tar: ./services: Cannot stat: No such file or directory
tar: ./volumes: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors

Paraphraser, 2021-11-07:
Maybe take a look at [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup) and see if it meets your needs.

Roberto6969, 2021-11-07:
I have seen your project. It looks attractive, but it is more to do to install it comparing to native backup option in IOTstack. My opinion is, Backup option in IOTstack project have to be fixed or removed. In last case, there is no doubt I will implement your solution ;-) But worth mention: can you implement your backup solution into IOTstack?

Roberto6969, 2021-11-08:
Actually, it's all about working directory. Adding path to /scripts solves all.

bburkett52, 2021-11-27:
I am a very casual user here and have been trying to set up a new clean install. A prior version used a "docker_backup.sh" script and the newest uses "backup.sh". I was able to have my backups sent to Google Drive in the old setup, but can't seem to get that working again. Is the rclone upload to Google Drive no longer functioning? Try as I might, I have not been able to figure it out.

I apologize if I have missing something in the documentation. 

~Bob

stonez56, 2021-12-15:
@Roberto6969  
Please tell me where to add the path to script to make the restore script work?

Thanks!
stonez56

Roberto6969, 2021-12-15:
1. check echo $PATH - in my case is /home/pi/IOTstack/scripts: beside others
2.  first line of backup.sh is cd /home/pi/IOTstack

This should work, although I'm not sure if first one is necessary. Please, let me know if this work.