SensorsIot/IOTstack #442: Hass.io Installation Fails

ShortPutt, 2021-11-14:
Ran Native installs Hass.io installation fails on a new Raspian 10.11 (buster) installation on a Raspberry Pi 4 Model B.
I am using the 10.11 buster version of Raspian because the xrdp does not work on bullseye.
Although the script says it was successful in installing Home Assistant, it actually fails to install properly as evidenced by running the following after exiting the menu:
	sudo su
	root@raspberrypi:/home/pi/IOTstack# systemctl status hassio-supervisor.service
	Unit hassio-supervisor.service could not be found.
	root@raspberrypi:/home/pi/IOTstack# systemctl status hassio-apparmor.service
	Unit hassio-apparmor.service could not be found.

Additionally docker.service also fails to run - and here is the first clue
/etc/docker/daemon.json contains "404 Not Found"
deleting /etc/docker/daemon.json and rebooting gets docker.service running again, but home assistant still fails w/ the above

So I traced through the code and found the following:
	IOTstack/menu.sh calls IOTstack/scripts/menu_main.py
	IOTstack/scripts/menu_main.py calls snative_installs.py
	IOTstack/scripts/native_installs.py calls IOTstack/.native/hassio_supervisor.sh
	IOTstack/.native/hassio_supervisor.sh contains:
		curl -sL "https://raw.githubusercontent.com/Kanga-Who/home-assistant/master/supervised-installer.sh" | sudo bash -s -- -m $hassio_machine
	https://raw.githubusercontent.com/Kanga-Who/home-assistant/master/supervised-installer.sh" contains:
		URL_RAW_BASE="https://raw.githubusercontent.com/home-assistant/supervised-installer/master/files"
		which resolves to 404 Not Found

Here is the offending code:

URL_RAW_BASE="https://raw.githubusercontent.com/home-assistant/supervised-installer/master/files"
URL_VERSION="https://version.home-assistant.io/stable.json"
URL_BIN_APPARMOR="${URL_RAW_BASE}/hassio-apparmor"
URL_BIN_HASSIO="${URL_RAW_BASE}/hassio-supervisor"
URL_DOCKER_DAEMON="${URL_RAW_BASE}/docker_daemon.json"
...
# Detect wrong docker logger config
if [ ! -f "$FILE_DOCKER_CONF" ]; then
  # Write default configuration
  info "Creating default docker daemon configuration $FILE_DOCKER_CONF"
  curl -sL ${URL_DOCKER_DAEMON} > "${FILE_DOCKER_CONF}"
...
URL_DOCKER_DAEMON resolves to "https://raw.githubusercontent.com/home-assistant/supervised-installer/master/files/docker_daemon.json" which does not exist

Please understand I am new to github, so please excuse me if I did not report this issue properly.  I am more than willing to learn. If there is a better way to report the issue, or if additional information is needed, please let me know.

Paraphraser, 2021-11-14:
No, how you've reported the problem is fine.

I've also run into issues with Hass.io which I have traced to the script behind the "native installs" no longer working.

I had borrowed the IOTstack script for [PiBuilder](https://github.com/Paraphraser/PiBuilder). I was doing a test build with Bullseye which wouldn't work. I repeated the test with Buster and that wouldn't work either. I was really confused because it wasn't all that long ago that it "worked" (and I still have the running system to prove it).

But it turns out the IOTstack "native installs" script was borrowed from somewhere else and that, in turn, was borrowed from yet another place. Meanwhile, over at Home Assistant, the whole approach to installation seems to have changed to be based on `.deb` packages.

> I don't understand the timing of any of this. I just report my findings.

I'm still in the process of testing a revamped solution for PiBuilder but, thus far, it works for both Buster and Bullseye, at least in the sense that HA installs and the containers come up OK.

> I can't comment on `xrdp`. I had to Google that to come up with "open source MS Remote Desktop Protocol". I'm still none-the-wiser. And, to be brutally honest, I can't see that I'm ever going to need it which, in turn, means I'm never going to care about it. Sorry.

If you're in a hurry, you could start with this:

```bash
#!/usr/bin/env bash

# should not run as root
[ "$EUID" -eq 0 ] && echo "This script should NOT be run using sudo" && exit -1

# the name of this script is
SCRIPT=$(basename "$0")

if [ "$#" -gt 0 ]; then
    echo "Usage: $SCRIPT"
    exit -1
fi

# has the user asked for home assistant?
if [ "$HOME_ASSISTANT_SUPERVISED_INSTALL" = "true" ] ; then

   # yes! use a default release if not otherwise provided
   HOME_ASSISTANT_AGENT_RELEASE="${HOME_ASSISTANT_AGENT_RELEASE:-"1.2.2"}"

   # check how the hardware describes itself
   case "$(grep "^Model[ :]*" /proc/cpuinfo | cut -c 10-23)" in

     "Raspberry Pi 3" )
       HINT="raspberrypi3"
       ;;

     "Raspberry Pi 4" )
       HINT="raspberrypi4"
       ;;

     *)
       echo "Home Assistant Supervised Install is enabled but this hardware does not"
       echo "identify as either Raspberry Pi 3 or 4. Either this configuration is not"
       echo "supported by Home Assistant or has not been tested for PiBuilder. If you"
       echo "wish to try anyway, modify this script to include your hardware in this"
       echo "case statement."
       exit 1
       ;;

   esac

   # the agent is ALWAYS armv7 - aarch64 does not work
   #    dpkg: error processing archive /tmp/Imx1h-DOWNLOADS/os-agent_1.2.2_linux_aarch64.deb (--install):
   #          package architecture (arm64) does not match system (armhf)

   AGENT="os-agent_${HOME_ASSISTANT_AGENT_RELEASE}_linux_armv7.deb"

   echo -e "\n\n\n========================================================================\n"
   echo "Hint: during Home Assistant installation, please choose:"
   echo "           \"$HINT\""
   echo "      at the \"Select machine type\" prompt"
   echo -e "\n========================================================================\n"

[... 88 lines cut ...]
echo "Setting groups required for docker and bluetooth"
sudo usermod -G docker -a $USER
sudo usermod -G bluetooth -a $USER

echo "Installing docker-compose and IOTstack dependencies"
sudo pip3 install -U docker-compose
sudo pip3 install -U ruamel.yaml==0.16.12 blessed

# reboot (applies usermods and any network manager change)
echo "$SCRIPT complete - rebooting..."
sudo reboot
```

That installs Hass.io, Docker and Docker-Compose plus everything needed for IOTstack to work.

Assuming you gave the script a name like `install-the-lot.sh`, the calling sequence would be:

```
$ export HOME_ASSISTANT_SUPERVISED_INSTALL=true
$ ./install-the-lot.sh
```

The script displays a "hint" recommending choosing either "raspberrypi3" or "raspberrypi3" when the HA installer asks. I've found that the "-64" variants don't work. It also seems to be problematic to be running the 64-bit kernel while running this script. My take-home is that switching to the 64-bit kernel should be the last step, not the first.

Hope this helps. If you try the script, I'd appreciate feedback, whether it works or fails. Do please note that there is an underlying assumption (made by Home Assistant) that Docker is **not** installed. Trying to run this script on a system where Docker is installed results in a mess.

ShortPutt, 2021-11-14:
Thanks Paraphraser. I used your script and it worked perfectly. The only change I made to the script was to remove the reboot so I could review the output before it rebooted. Everything is running as it should. 

I appreciate the help!

KlausHans, 2021-12-12:
Hello, unfortunately it does not work for me on a fresh Raspberry Pi Os install on a Raspberry Pi 3. 
It installs, but Home Assistant is not supervised. HA supervisor has problems with apparmor, the networkmanager (in my case in the conf was only one row "404 not found") and the operating system. Apparmor and network manager was solvable with the help of the provided HA documentation. But the error message regarding the operating system is still there and i can't get rid of it, it seems. Any help is appreciated.
https://www.home-assistant.io/more-info/unsupported/os

KlausHans, 2021-12-12:
I've read a little further, no progress. It seems Home Assistant Supervised on Raspberry OS is simply dead. What a shame.

ShortPutt, 2021-12-12:
KlausHans - its not dead. I've had Home Assistant supervised running on an RPi 4 for a couple weeks now. I used the script posted by paraphraser. 

Here's the process I used.
Use Raspberry Pi Imager to install the OS. I used 64 bit Bullseye. Also, use ctrl shift X to enable ssh
ssh to RPi
apt update
apt upgrade
Install hassio w/ paraphraser script
Install docker-compose - I did this but I think it would have been installed by IOTstack anyway.
Install IOTstack and the containers you want.

btw, the image of the error is in German, so I'm not able to read it.

KlausHans, 2021-12-12:
I am confused. I checked an other, old installation with working supervisor, the supervisor check there spits out even more errors/warnings. But why am i missing the supervisor in the side menu in the new installation?
[image]
(sorry for german screenshots)

ShortPutt, 2021-12-12:
That's a different problem. I'm still running  2021.11.5 and Supervisor side menu is there.
Based on the issue u r experiencing, I'm not going to upgrade.
I suggest you open a new issue as this is completely different than what I initially reported.

Paraphraser, 2021-12-12:
Perhaps take a look at [PiBuilder](https://github.com/Paraphraser/PiBuilder).

If you start with a clean Raspbian image (Buster or Bullseye; I recommend the former if you depend on camera support), PiBuilder will:

* install docker
* install docker-compose
* install Supervised Home Assistant as an option
* clone the IOTstack repo.

The docker-compose you get from PiBuilder is the modern "plugin" version (currently v2.2.2). If you let the menu install docker-compose, you get an obsolete version which also somehow futzes with docker's version number, and then the menu has conniptions. Then, when you try to fix that, uninstalling docker-compose takes docker with it. A real mess.

Similarly, the Supervised Home Assistant you get from PiBuilder is the current recommended method. Everything is installed in the proper order and you get a set of Home Assistant containers running independently of IOTstack.

After PiBuilder has completed, you can either:

1. Restore a backup (I **know** a backup taken by [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup) works because I do this all the time); or
2. Run the IOTstack menu and choose your containers.

The only part I **don't** know is how to save/restore Home Assistant data. I don't actually use Home Assistant so I don't know where it stores its data. You'll have to figure that bit out yourself. Sorry.

Paraphraser, 2021-12-13:
In case there was any doubt, I started with a clean Bullseye image, ran PiBuilder on it and restored an IOTstack backup for good measure.

25 minutes go to whoa.

Docker thinks Supervised Home Assistant is running alongside the IOTstack containers:

[image]

Home Assistant thinks it is running and presents its GUI.

[image]

Not being a user of Home Assistant, I'm not quite sure what is meant by the "side menu". I'm assuming it means the panel on the left in this shot:

[image]

I can't find anything that matches the pattern mentioned earlier (integrations -> supervisor -> configuration). I only see "configuration":

[image]

Within "Configuration", the "Info" tab has a list of "integrations":

[image]

If that's not what you mean, please give me instructions as to what I should click on and capture and I'll see what I can do.

Complete build log on [PasteBin](https://pastebin.com/itatx3xu).

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke