SensorsIot/IOTstack #448: This is more discussion than an issue.

NossieUK, 2021-11-21:
Ijust ruined my pi by trying to install xxxx 

I run pi and unbound. I realise I should not run this in a container so I do not....

I'm also trying to install a number of things from https://github.com/Brandawg93/Pi-Hole-Monitoring

I'd also like to tear into running home assistant supervised - does anyone have the experience of using this script and supervised and how did you get on?

NossieUK, 2021-11-22:
This appears to be my own issue and not reviewing the content.....

Please mark this as closed as installing the supervisor is part of the script.

Paraphraser, 2021-11-22:
Perhaps read [Issue 442](https://github.com/SensorsIot/IOTstack/issues/442#issuecomment-968293272). It explains the background (why the "native install" supplied with IOTstack doesn't work) and contains a script for installing Docker and Docker-Compose, plus the option of installing Supervised Home Assistant.

I recommend starting from a clean Raspbian. If you don't want to do that and already have Docker installed, you might need to run some or all of the following commands first:

```
$ sudo apt-get purge -y docker-ce docker-ce-cli docker-ce-rootless-extras
$ sudo rm -rf /var/lib/docker /etc/docker
$ sudo groupdel docker
$ sudo rm -rf /var/run/docker.sock
$ sudo ifconfig docker0 down
$ sudo brctl delbr docker0
$ sudo reboot
```

Then follow Issue 442. Do please remember that the Supervised Home Assistant installation includes Network Manager, which turns on random MAC addresses for WiFi which, in turn, means that you really need to be connected to Ethernet for this installation.

<hr>

I don't think there's a huge problem running a service like unbound in a container. I'm running

* wmandra/bind9:9.16
* kosdk/dhcpd
* lscr.io/linuxserver/syslog-ng
* garethflowers/svn-server

in containers and it works just fine - provided they all run in Host Mode. By "just fine" I mean it's the authoritative DNS and DHCP server for my network.

You didn't say which image you were using for unbound. There seem to be a lot on DockerHub. The one I picked at random (klutchell/unbound) was not running in Host Mode but **did** use the "privileged" flag and I reckon that's a bit unsafe.

NossieUK, 2021-11-22:
"Do please remember that the Supervised Home Assistant installation includes Network Manager, which turns on random MAC addresses for WiFi which, in turn, means that you really need to be connected to Ethernet for this installation."

This in itself is very useful - thank you for the information.

NossieUK, 2021-11-22:
I do actually have 1 Q - from the link I posted, are you aware of those scripts?  can I grab them from somewhere else that does not involve running their own  stack? (the grafana pihole and  pi graphs)

Paraphraser, 2021-11-22:
Can't help with the second part. Except for testing the install procedures, I don't run HA (either supervised or standalone). Sorry.

On the first part, why not run it "as is"? Think about IOTstack. It's just a folder with a structure that all containers adhere to. The PiHole monitor is the same. Just a folder with a structure. There's no reason why you can't run Supervised HA (a set of containers), IOTstack (a set of containers) and PiHole Monitor (a set of containers) on the same machine at the same time.

Now, granted, there may be a few things to sort out. For example, the Grafana service definitions in the respective `docker-compose.yml` files both use external port 3000. If you want to run Grafana in both environments (and there is absolutely no reason why you should not do that) then one of them will have to use a different external port. Once you sort those things out, everything will happily coexist.

Paraphraser, 2021-11-22:
Does ["Help! I'm locked out of HA"](https://www.home-assistant.io/docs/locked_out) help?

NossieUK, 2021-11-22:
using the second part of your suggestion I realized that the previous login may still work (even tho I wiped the SD card???) and this has got me back into HA but without the supervisor ......

When I tried to run a stack on a pizeo it failed with critical errors that I could not resolve.  I then tried to run a selected stack using that link from above on a pizero2 and it ran, while pihole and unbound ran natively.

However that ran out of memory pretty damn quick.  So I gave up with my pizero2 and pretty inkyphat setup and dropped back to this pi4 ( I apparently collect pi's but I was hoping not to use this due to power)  So I installed pihole, unbound and homebridge natively, - and telegraph, grafana, influx and I then realized the benefits of home assistant and installed that using the python virtual environment ... and after my approx. 30 ish hours of working it all out  - I realized I needed the benefits of the supervisor (ssh, etc)  - I wiped everything and fell back to this IOTstack and marvelously got myself back into a better position in 20 mins compared to what I had been fighting with for the last week or so...  that then made me post what I did until I realized that the option is in the script (installing the supervisor) and then for me to realize I was never asked to set a username even though what I had set before worked fine.

Very long story short - I installed the supervisor using this script.  HA has accepted my previous login even though I wiped the SDcard and so far I still do not have supervisor access...

I could do my usual and install portainer, neofetch, fish and my other customisation  - but I'm just trying to not tie myself in knots getting this working!    - I guess I'm just trying to get the pihole monitor template and the one for the pi working and to kick the supervisor into gear.   ... I need to go lie down now.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke