SensorsIot/IOTstack #450: Dashmachine not starting?

NossieUK, 2021-11-27:
Just wondering if anyone else is having the same issue?
the stack starts with no obvious errors but going to the address 5000 fails and then checking the logs

dashmachine exited with code 1
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error
dashmachine       | standard_init_linux.go:219: exec user process caused: exec format error

Any ideas?   I have almost everything  else from the stack running fine

Paraphraser, 2021-11-28:
The service definition has:

```
image: rmountjoy/dashmachine:latest
```

Heading to [DockerHub](https://hub.docker.com/r/rmountjoy/dashmachine/tags) shows three images. The OS/ARCH for all of them is "linux/amd64" which won't run on the Raspberry Pi. That's what you're seeing - an attempt to launch something compiled for amd64 on arm.

There's a related rmountjoy/dashmachine0.7 but that's also AMD64.

There is no history so I have no idea how long ago there was an image for the RPi. If there was, we could just pin to it.

Maybe try this alternate image:

```
image: volcan96/dashmachine-docker-rpi:latest
```

If it works, tell me and I'll put in a Pull Request for you (unless you want to do it yourself).

The image was last updated 2 years ago. You might be better off opening an issue at [github.com/rmountjoy92](https://github.com/rmountjoy92/DashMachine) to see if the maintainer would be prepared to build a variant for armv7.

The last possibility is to change IOTstack to construct from scratch using a local Dockerfile.

NossieUK, 2021-11-28:
Thanks - so I updated the config with the one you gave, and it pulled 0.5 no problem - I'm running on rpi4 64 so I should have expected to run into this - however, having reviewed this and that it has not been updated in 2 years I fear it's a dead project.

On reddit about a month ago, the maintainer has resurfaced and appears to have started a new repo, however I don't have another 2 years to wait for an up to date 0.7 that works on arm64,  I had been using Heimdall while trying to get this one working, but I'm just going to find something else even if I must install it manually.

https://old.reddit.com/r/DashMachine/

I wouldn't trust myself to submit a PR as I'm just dipping my toes into containers and things - but for the sake of keeping things working, you may wish to update it..

Paraphraser, 2022-06-19:
Current. Needs PR to change image name.

derskythe, 2023-02-01:
> Maybe try this alternate image:
> 
> ```
> image: volcan96/dashmachine-docker-rpi:latest
> ```

For security reasons, I wouldn't recommend it. The docker file contains some copying of some suspicious files ([Line 16](https://hub.docker.com/layers/volcan96/dashmachine-docker-rpi/latest/images/sha256-d6f8fcf1a81992218d9ddc31ed453a2d627dcd46aeb6d5e2544f8b741b6ba825?context=explore)) and the image itself was created with a large amount of unnecessary data