SensorsIot/IOTstack #455: updated Influxdb container

jabss, 2021-12-06:
Hello,

I've updated all containers and influxdb is now version 2.1.1.
Although it may be not working for some people, I'm using bullseye 64bit on a RPI4 and the database goes up properly (it gets in "green running" in portainer CE) and I'm able to access its web GUI.

However, its way of working seems to be different from the previous release (18.x). The commands from the earlier version don't seem to work.

Are there any plans to update the documentation in order to show how to configure it?   Or would it be recommended to remain in 1.8.x? In that case, any plan to update menu so it fetches 1.8 only?

Thanks,

Paraphraser, 2021-12-06:
See the current [InfluxDB service definition](https://github.com/SensorsIot/IOTstack/blob/master/.templates/influxdb/service.yml).

To put that in words, if you start from a totally clean clone of IOTstack, run the menu (old or new), and choose InfluxDB, what you get is version 1.8 of InfluxDB. In terms of what comes down from DockerHub, the "1.8" in `image: "influxdb:1.8"` implies "1.8.x" so, right now, that's [1.8.10](https://hub.docker.com/_/influxdb?tab=tags).

IOTstack has always been a bit laissez-faire when it comes to forcing updates on you. Assuming you run the menu from time to time it will tell you when you are out-of-sync with GitHub and accepting the recommendation will do a `git pull` but that still only changes the templates. Under old menu, you'd have to do a "pull full service from template" to get a change applied. Under new menu, I think you have to turn a service off and on again. The reason the menu is a bit reticent is because of the risk of overwriting any changes you might have made.

> My personal view is that the menu is an excellent way of getting started. After that, you're better advised to do your own `git pull`, monitor any changes in the templates for the services you use, consider whether you want to adopt any service definition updates, case by case, and edit your compose file by hand. I reckon knowing the answer to "what changed?" is 90% of the battle in figuring out why something is suddenly misbehaving. But that's just me.

As to your second question, I haven't been successful in changing to "latest". That's with Bullseye and the 64-bit kernel but I don't think I've tried it with the modern docker-compose. My experience to date is that there is no aarch64. I'll give it another try and see if I get any further.

Paraphraser, 2021-12-06:
I still can't get anywhere with this.

Identify the machine and OS:

``` 
$ uname -a
Linux sec-dev 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
$ grep "PRETTY_NAME" /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
```

Start with a clean clone.

```
$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
Cloning into 'IOTstack'...
remote: Enumerating objects: 7449, done.
remote: Counting objects: 100% (2188/2188), done.
remote: Compressing objects: 100% (1059/1059), done.
remote: Total 7449 (delta 1283), reused 1835 (delta 1039), pack-reused 5261
Receiving objects: 100% (7449/7449), 7.41 MiB | 11.15 MiB/s, done.
Resolving deltas: 100% (4435/4435), done.
``` 

Edit the influx template to use "latest" instead of "1.8"

```
$ cd IOTstack/.templates/influxdb/
$ grep "image" service.yml 
  image: "influxdb:1.8"
$ sed -i.bak 's/1.8/latest/' service.yml 
$ grep "image" service.yml 
  image: "influxdb:latest"
```

Run the menu and choose influx.

```
$ cd ../..
$ ./menu.sh 
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python Version: 'Python 3.9.2'. Python is up to date.
ruamel.yaml Version: 'ruamel.yaml 0.16.12'. ruamel.yaml is up to date.
Blessed Version: 'blessed 1.19.0'. Blessed is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.11 >= 18.2.0. Docker is good to go.
Project dependencies up to date

...

Exiting IOTstack menu.
Build completed:  ./docker-compose.yml                                                                                                                   You can start the stack from the Docker Commands menu, or from the CLI with:  docker-compose up -d
```

Show the result of running the menu.

```
$ cat docker-compose.yml 
version: '3.6'
services:
  influxdb:
    container_name: influxdb
    image: "influxdb:latest"
    restart: unless-stopped
    ports:
    - "8086:8086"
    environment:
    - INFLUXDB_HTTP_FLUX_ENABLED=false
    - INFLUXDB_REPORTING_DISABLED=false
    - INFLUXDB_HTTP_AUTH_ENABLED=false
    - INFLUX_USERNAME=dba
    - INFLUX_PASSWORD=supremo
    - INFLUXDB_UDP_ENABLED=false
    - INFLUXDB_UDP_BIND_ADDRESS=0.0.0.0:8086
    - INFLUXDB_UDP_DATABASE=udp
    volumes:
    - ./volumes/influxdb/data:/var/lib/influxdb
    - ./backups/influxdb/db:/var/lib/influxdb/backup
    networks:
    - iotstack_nw

networks:
  iotstack_nw: # Exposed by your host.
    # external: true
    name: IOTstack_Net
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 10.77.60.0/24
        # - gateway: 10.77.60.1

  iotstack_nw_internal: # For interservice communication. No access to outside
    name: IOTstack_Net_Internal
    driver: bridge
    internal: true
    ipam:
      driver: default
      config:
      - subnet: 10.77.76.0/24
        # - gateway: 10.77.76.1
  vpn_nw: # Network specifically for VPN
    name: IOTstack_VPN
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 10.77.88.0/24
        # - gateway: 192.18.200.1

  nextcloud_internal: # Network for NextCloud service
    name: IOTstack_NextCloud
    driver: bridge
    internal: true

  # default:
  #   external: true
  #   name: iotstack_nw

  # hosts_nw:
  #   driver: hosts
``` 

Bring up the stack.

``` 
$ docker-compose up -d
[+] Running 0/1
 ⠇ influxdb Pulling                                                                                                                                                                3.7s
no matching manifest for linux/arm/v7 in the manifest list entries
$
```

That's what I get every time. The uname shows aarch64 but Docker keeps pulling armv7. I've tried forcing the architecture on a manual pull but the container just goes into a restart loop:

```
$ docker pull --platform "linux/arm64/v8" influxdb:latest
latest: Pulling from library/influxdb
39e4f823356a: Pull complete 
df66cf961d40: Pull complete 
cf6c547e43b8: Pull complete 
fb4e453752af: Pull complete 
d0f4c33d2485: Pull complete 
8240edd1b1df: Pull complete 
3d470fb04618: Pull complete 
f374b9f02973: Pull complete 
19627f99f85f: Pull complete 
792639683dfd: Pull complete 
Digest: sha256:6597ad58ab996d75e4f000e6295d2a2db8543644e329f009d0e3b7ceaaed3940
Status: Downloaded newer image for influxdb:latest
docker.io/library/influxdb:latest
$ docker-compose up -d
[+] Running 2/2
 ⠿ Network IOTstack_Net  Created                                                                                                                                                   0.4s
 ⠿ Container influxdb    Started                                                                                                                                                   1.1s
$ DPS
NAMES      CREATED         STATUS
influxdb   7 seconds ago   Restarting (159) 1 second ago
$ 
```

For the record, I install:

* docker using the so-called "convenience" script. That updates via apt. On this machine it's at 20.10.11.
* docker-compose in [plugin form](https://github.com/docker/compose/releases). On this machine it's at 2.1.1 but I doubt that updating to 2.2.2 would change anything.

What are you doing differently that gets you a running InfluxDB "latest"?

jabss, 2021-12-07:
Hello,

I think I know what may have happened, although it doesn't answer all the questions.
I did a fresh install of bullseye 64bit and copied the IOTstack folder (which I had backed up from buster 64bit) into it. The IOTstack version I had was somewhere from Jan 2021 and I suspect that by then, the influx parameter for image was "latest", as I don't remember to ever having changed that manually.
After restoring it, I did all the updates I could (the git update as well as "updating all containers" option in the menu), and this may explain why I ended up with influxdb2 because I was always re-using my backed-up docker-compose.yml file (I have my configurations there) and it is likely that this old version still had the "latest" instead of "1.8".
I can't explain, however, is how I got the correct architecture, but I do remember browsing around docker hub, although I wouldn't know how to take it into my RPI.

I have now manually edited to "1.8" and I got back the old version, with the old data. At this moment, I don't think I need influxdb2 as the 1.8 seems to be enough for my needs (and it is working as I expect).

So bottom line is (and please correct me if I'm wrong), IOTStack default configuration specifies influxdb1.8 and that is working with the documentation available (actually nothing written, but Andres video that shows how to add users and create databases within).

In this case, I believe this issue doesn't make sense to be open as the problematic situation derived from my older docker-compose.yml file, and not default and updated configuration from IOTStack project.

I'll be, however, willing to help trying to sort this Influxdb2 architecture issue as I believe I still have the image in my deployment.

PR #567 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3
Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
commit 3239e9b4 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3

Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
closed 2022-06-12 by Slyke