SensorsIot/IOTstack #459: Grafana working on computer, but not mobile

Nurgak, 2021-12-14:
I have a wonderful setup with a working Grafana instance, but for some reason I cannot load it on my iPhone. It does not matter what browser I use (Firefox, Safari/DuckDuckGo), the page will always show the same error:

[image]

I have had this issue from the very beginning. What could be the reason for it? The Grafana logs don't seem to show anything relevant to this.

Paraphraser, 2021-12-14:
This happened to me a few weeks ago - the same messages - but I must admit I did not go to the trouble of checking different browsers on different platforms. I just noticed it on my iPad and immediately reverted to the previous image.

The problem was traced to a bug (or so I assume) in whatever process gets images built and propagated to DockerhHub:

* [issue 42362](https://github.com/grafana/grafana/issues/42632)

It was fixed fairly quickly, then, but it looks to me like it's just happened again - and has been fixed equally quickly:

* [issue 43004](https://github.com/grafana/grafana/issues/43004)

I've just done a "pull", v8.3.2 has downloaded and seems to be OK so my guess is that if you just do another pull/up the problem will go away.

Nurgak, 2021-12-15:
@Paraphraser Thanks, recreating the container, while pulling the latest image (in Portainer), fixed the issue as you said.

I am able to view my dashboards on my phone now.

closed 2021-12-15 by Nurgak