SensorsIot/IOTstack #464: The Postgres Template Throws an Error

hughk, 2021-12-15:
Attempting to create a Postgres container on the stack will throw an error on a missing file passwords.py. I have copied the version from mariaDb changing the variable names to postgresDb. This goes further but throws an error on missing generateRandomString during the build. This in turn was solved by adding generateRandomString to _from deps.common.functions_ import in build.py. The stack build now appears to work. If this worksd and is stable, I'll raise a PR or someone else can do the changes as they are minor.

Paraphraser, 2021-12-15:
I just tried this to see if I could replicate your experience. I'm afraid I can't. For me, it just works "out of the box".

This is what I did:

1. Clean clone of IOTstack.
2. Run the menu.
3. Choose Build Stack.
4. Choose PostGreSQL as the only container. My exact sequence was:

	- Use the down-arrow key to put the cursor on top of PostGreSQL
	- hit the space bar to select it
	- press <kbd>return</kbd> to signal my choices were complete (which goes back to the main menu).

5. Hit <kbd>esc</kbd> to exit the menu.
6. List the resulting docker-compose.yml.
7. UP the stack (an alias for `docker-compose up -d`).
8. DPS (an alias for `docker ps`) to make sure the container was running and not restarting.
9. Examine the log.

Things to note:

1. `POSTGRES_PASSWORD=IOtSt4ckpostgresDbPw` so the Python script has replaced the `%randomPassword%` placeholder.
2. The "database system is ready to accept connections" message at the end of the log.

Looks OK to me. I think there must be some other explanation for what is happening on your system.

```
pi@sec-dev:~$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
Cloning into 'IOTstack'...
remote: Enumerating objects: 7449, done.
remote: Counting objects: 100% (2188/2188), done.
remote: Compressing objects: 100% (1063/1063), done.
remote: Total 7449 (delta 1280), reused 1835 (delta 1035), pack-reused 5261
Receiving objects: 100% (7449/7449), 7.38 MiB | 10.18 MiB/s, done.
Resolving deltas: 100% (4432/4432), done.
pi@sec-dev:~$ cd IOTstack
pi@sec-dev:~/IOTstack$ ./menu.sh 

[ ... choose PostgreSQL ... ]

pi@sec-dev:~/IOTstack$ cat docker-compose.yml 
version: '3.6'
services:
  postgres:
    container_name: postgres
    image: postgres
    restart: unless-stopped
    environment:
    - POSTGRES_USER=postuser
    - POSTGRES_PASSWORD=IOtSt4ckpostgresDbPw
    - POSTGRES_DB=postdb
    ports:
    - "5432:5432"
    volumes:
    - ./volumes/postgres/data:/var/lib/postgresql/data
    networks:
    - iotstack_nw
networks:
  iotstack_nw: # Exposed by your host.
    # external: true
    name: IOTstack_Net
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 10.77.60.0/24
        # - gateway: 10.77.60.1

  iotstack_nw_internal: # For interservice communication. No access to outside
    name: IOTstack_Net_Internal
[... 83 lines cut ...]
2021-12-15 11:06:35.364 UTC [49] LOG:  database system is ready to accept connections
 done
server started
CREATE DATABASE

/usr/local/bin/docker-entrypoint.sh: ignoring /docker-entrypoint-initdb.d/*

2021-12-15 11:06:35.902 UTC [49] LOG:  received fast shutdown request
waiting for server to shut down....2021-12-15 11:06:35.904 UTC [49] LOG:  aborting any active transactions
2021-12-15 11:06:35.909 UTC [49] LOG:  background worker "logical replication launcher" (PID 56) exited with exit code 1
2021-12-15 11:06:35.910 UTC [51] LOG:  shutting down
2021-12-15 11:06:35.942 UTC [49] LOG:  database system is shut down
 done
server stopped

PostgreSQL init process complete; ready for start up.

2021-12-15 11:06:36.075 UTC [1] LOG:  starting PostgreSQL 14.1 (Debian 14.1-1.pgdg110+1) on arm-unknown-linux-gnueabihf, compiled by gcc (Debian 10.2.1-6) 10.2.1 20210110, 32-bit
2021-12-15 11:06:36.076 UTC [1] LOG:  listening on IPv4 address "0.0.0.0", port 5432
2021-12-15 11:06:36.076 UTC [1] LOG:  listening on IPv6 address "::", port 5432
2021-12-15 11:06:36.081 UTC [1] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
2021-12-15 11:06:36.091 UTC [63] LOG:  database system was shut down at 2021-12-15 11:06:35 UTC
2021-12-15 11:06:36.108 UTC [1] LOG:  database system is ready to accept connections
pi@sec-dev:~/IOTstack$ 
```

Paraphraser, 2021-12-15:
The other thing it might be worth thinking about is how environment variables are used. For passwords, they tend to be "one time only". So, let's say you had:

```
POSTGRES_PASSWORD=%randomPassword%
```

at some point. If Postgres got going far enough to initialise everything, that will be the password. If you come along and change the right hand side later, it doesn't cause a change in the database. The database will continue to believe `%randomPassword%` is the root password. All sorts of things behave this way, not just databases.

It might be that all you need to do is:

1. Stop Postgres from running.
2. `sudo rm -rf ~/IOTstack/volumes/postgres`
3. Start Postgres again.

hughk, 2021-12-15:
First, thanks for getting back to me so quickly. The issue I believe is that I attempted to use the options alongside the Postgres entry on the new menu to set the password rather than the default  (I have an allergy to default passwords). On a new installation with a fresh git clone as you suggest, the problem repeats when I get to step 4 in the list that you made in your first answer.

`Traceback (most recent call last):
  File "/home/pi/IOTstack/./scripts/menu_main.py", line 456, in <module>
    runSelection(currentMenuItemIndex)
  File "/home/pi/IOTstack/./scripts/menu_main.py", line 411, in runSelection
    mainMenuList[selection][1]()
  File "/home/pi/IOTstack/./scripts/menu_main.py", line 110, in buildStack
    exec(code, execGlobals, execLocals)
  File "./scripts/buildstack_menu.py", line 635, in <module>
    main()
  File "./scripts/buildstack_menu.py", line 604, in main
    executeServiceOptions()
  File "./scripts/buildstack_menu.py", line 519, in executeServiceOptions
    exec(code, execGlobals, execLocals)
  File "./.templates//postgres/build.py", line 397, in <module>
    main()
  File "./.templates//postgres/build.py", line 387, in main
    eval(toRun)()
  File "./.templates//postgres/build.py", line 252, in runOptionsMenu
    menuEntryPoint()
  File "./.templates//postgres/build.py", line 360, in menuEntryPoint
    runSelection(currentMenuItemIndex)
  File "./.templates//postgres/build.py", line 317, in runSelection
    postgresDbBuildOptions[selection][1]()
  File "./.templates//postgres/build.py", line 218, in setPasswordOptions
    with open(passwordOptionsMenuFilePath, "rb") as pythonDynamicImportFile:
`
If I retry using the default or an environment variable it works perfectly.

PR #580 (2022-09-11): postgres: remove broken menu options and fix docs
Add compose variables and indentation as explained in #505

Fixes #383, Fixes #517, Fixes #464
closed 2022-09-11 by Slyke