SensorsIot/IOTstack #470: Add support for zwavejs2mqtt?

JetForMe, 2021-12-24:
Hi there. Could you add the following as an optional container to include in the docker compose? Thanks! https://zwave-js.github.io/zwavejs2mqtt/#/getting-started/docker

ukkopahis, 2022-06-22:
Could you provide your configuration and use-case as a starting point?

JetForMe, 2022-06-22:
My use-case is I want to be able to talk to my Z-wave devices via MQTT (https://zwave-js.github.io/zwavejs2mqtt/#/README).

I don't yet have a configuration. I came across this back in December when I was first installing IOTStack, and haven't gotten any farther.

ukkopahis, 2022-06-23:
What's your docker knowledge level? Would you be able to try out the example from their homepage (https://zwave-js.github.io/zwavejs2mqtt/#/getting-started/docker?id=run-as-a-service)?

Or in a more "IOTstack":ish style:
``` yaml
version: "3.7"
services:
  zwavejs2mqtt:
    container_name: zwavejs2mqtt
    image: zwavejs/zwavejs2mqtt:latest
    restart: always
    tty: true
    stop_signal: SIGINT
    environment:
      - SESSION_SECRET=mysupersecretkey
      - ZWAVEJS_EXTERNAL_CONFIG=/usr/src/app/store/.config-db
      # Uncomment if you want log times and dates to match your timezone instead of UTC
      # Available at https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
      #- TZ=America/New_York
    devices:
      # Do not use /dev/ttyUSBX serial devices, as those mappings can change over time.
      # Instead, use the /dev/serial/by-id/X serial device for your Z-Wave stick.
      - '/dev/serial/by-id/insert_stick_reference_here:/dev/zwave'
    volumes:
      - ./volumes/zwavejs2mqtt:/usr/src/app/store
    ports:
      - "8091:8091" # port for web interface
      - "3000:3000" # port for Z-Wave JS websocket server
```