SensorsIot/IOTstack #472: ERROR: Service 'prometheus' depends on service 'cadvisor' which is undefined.

Melistraza, 2022-01-06:
Trying to add *prometheus* for *grafana*.
Added service in *Build Stack* menu and get docker-compose.yml

```
version: '3.6'
services:
  portainer-ce:
    container_name: portainer-ce
    image: portainer/portainer-ce
    restart: unless-stopped
    ports:
    - "8000:8000"
    - "9000:9000"
    volumes:
    - /var/run/docker.sock:/var/run/docker.sock
    - ./volumes/portainer-ce/data:/data
  prometheus:
    container_name: prometheus
    build: ./.templates/prometheus/.
    restart: unless-stopped
    user: "0"
    ports:
    - "9090:9090"
    environment:
    - IOTSTACK_UID=1000
    - IOTSTACK_GID=1000
    volumes:
    - ./volumes/prometheus/data:/prometheus
    command:
    - '--config.file=/prometheus/config/config.yml'
    #    defaults are:
    #    - --config.file=/etc/prometheus/prometheus.yml
    #    - --storage.tsdb.path=/prometheus
    #    - --web.console.libraries=/usr/share/prometheus/console_libraries
    #    - --web.console.templates=/usr/share/prometheus/consoles
    depends_on:
    - cadvisor
    - nodeexporter
    networks:
    - iotstack_nw

  grafana:
    container_name: grafana
    image: grafana/grafana
    restart: unless-stopped
    user: "0"
    ports:
    - "3000:3000"
    environment:
    - GF_PATHS_DATA=/var/lib/grafana
    - GF_PATHS_LOGS=/var/log/grafana
    volumes:
    - ./volumes/grafana/data:/var/lib/grafana
    - ./volumes/grafana/log:/var/log/grafana
    networks:
    - iotstack_nw

networks:
  iotstack_nw: # Exposed by your host.
    # external: true
    name: IOTstack_Net
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 10.77.60.0/24
        # - gateway: 10.77.60.1

  iotstack_nw_internal: # For interservice communication. No access to outside
    name: IOTstack_Net_Internal
    driver: bridge
    internal: true
    ipam:
      driver: default
      config:
      - subnet: 10.77.76.0/24
        # - gateway: 10.77.76.1
  vpn_nw: # Network specifically for VPN
    name: IOTstack_VPN
    driver: bridge
    ipam:
      driver: default
      config:
      - subnet: 10.77.88.0/24
        # - gateway: 192.18.200.1

  nextcloud_internal: # Network for NextCloud service
    name: IOTstack_NextCloud
    driver: bridge
    internal: true

  # default:
  #   external: true
  #   name: iotstack_nw

  # hosts_nw:
  #   driver: hosts
```

Running *Docker Commands* > *Restart stack* and get

```
Restarting Stack...
Stop Stack:
docker-compose down
ERROR: Service 'prometheus' depends on service 'cadvisor' which is undefined.

Start Stack:
docker-compose up -d --remove-orphans
ERROR: Service 'prometheus' depends on service 'cadvisor' which is undefined.

Stack Restarted
Process terminated. Press [Enter] to show menu and continue.
```

according to this config lines
```
depends_on:
    - cadvisor
    - nodeexporter
```
this two services also should be added to docker-compose.yml
but they missed.

Using latest version of IOTstack:
```
git fetch
git pull
Already up to date.
```

Found article:
https://sensorsiot.github.io/IOTstack/Containers/Prometheus/#dependencies

Paraphraser, 2022-01-06:
I am not sure what is going on. The three services are in the template but they aren't getting into the compose file.

As a workaround, you can do this:

```
$ cd ~/IOTstack
$ sed -e "s/^/  /" ./.templates/prometheus/service.yml >replace.yml
```

and then use the content of `replace.yml` to replace the Prometheus service definition in your `docker-compose.yml`. After that, a `docker-compose up -d` should work.

Compared with how they appear in your `docker-compose.yml`, template `service.yml` files are left-shifted by two spaces. That `sed` command is just adding two spaces to the start of each line.

Melistraza, 2022-01-07:
This is exactly what I have done. But that weird because previously I added *portainer* and it's works perfectly.
After that I decided to add metrics and faced with this issue.
If it's only my *local* issue we can close this ticket.

Paraphraser, 2022-01-07:
It is **not** your local issue. I repeated your process (clean slate, fresh clone of IOTstack, run the menu, choose Prometheus) and saw the same behaviour. It’s an IOTstack problem.

Melistraza, 2022-01-07:
Glad I helped quickly find the problem!

Melistraza, 2022-01-13:
I suppose it's same behaviour for all related containers.
```
Restarting Stack...
Stop Stack:
docker-compose down
ERROR: Service 'telegraf' depends on service 'mosquitto' which is undefined.

Start Stack:
docker-compose up -d --remove-orphans
ERROR: Service 'telegraf' depends on service 'mosquitto' which is undefined.

Stack Restarted
Process terminated. Press [Enter] to show menu and continue.
```

Paraphraser, 2022-01-13:
Mosquitto is a different issue. You have to select that in the menu so it gets added.

The Prometheus service definition contains cadvisor and nodeexporter. If you run old menu then all three containers get added to your docker-compose.yml (the intention). New menu (which is what you get by default) strips cadvisor and nodeexporter. No idea why.

If you want, you can switch back to old menu:

```
$ cd ~/IOTstack
$ git checkout old-menu
```

That avoids and need to cut and paste stuff into your compose file. But you still need to select Mosquitto as well as Prometheus.

Melistraza, 2022-01-13:
@Paraphraser ok got it thanks!
Right now I trying to figure out how to add mosquito. Because I have it already in HomeAssistant as integration with different devices.

Paraphraser, 2022-01-21:
@Melistraza 

This has been dealt with by [PR476](https://github.com/SensorsIot/IOTstack/pull/476).

Prometheus was always OK under old-menu where just selecting Prometheus brought CAdvisor and NodeExporter along for the ride. New menu was deleting CAdvisor and NodeExporter even though the base service definition included all three containers.

The new-menu "fix" is to split the three service definitions. If you run the menu and select all three (they sort together in the menu) then it works.

Please give it a try. If the fix has solved your problem then you can probably close this issue.

DutchShadow, 2022-02-04:
@Paraphraser heyo, the current version of the service file for prometheus depends on cadvisor and nodeexporter.... 
After a bit of digging with selecting all three I found I need to rename the depends: to the respective new "prometheus-cadvisor" :)

Paraphraser, 2022-02-04:
@DutchShadow Ah! I see what you mean. I didn't spot any problem with the existing structure when I first tested it. I also didn't immediately spot anything when I re-tested a few moments ago starting from a clean clone.

But then I tried things I should've tried before like terminating prometheus. That's when docker-compose finally grizzled that the dependent services didn't exist.

Gotcha!

I'll submit some PRs.

PR #487 (2022-02-23): 20220206 prometheus - master branch - PR 1 of 2
Uses `prometheus-` prefix consistently for subordinate containers (service name, container name, dependency name) to address concern raised in comment to [Issue 472](https://github.com/SensorsIot/IOTstack/issues/472#issuecomment-1030104238).

Documentation updated.

Fixes #472