SensorsIot/IOTstack #473: Can't update Node-RED to 2.1

DIYtechie, 2022-01-07:
tried updating the project, pulling latest image, rebuilding the stack, but I'm stuck on 1.0.6

Can I just use portainer to change the environment variables? And what should they be?

Paraphraser, 2022-01-07:
Did you follow the instructions in the [IOTstack Node-RED documentation](https://sensorsiot.github.io/IOTstack/Containers/Node-RED/#upgrading-node-red)?

It sounds like you're just doing a simple `pull`. That won't work when a Dockerfile is involved. It's explained in that doco.

If that still doesn't work you can try brute force. Stop the nodered container (or your entire stack), then run `docker images` and, for each image that mentions nodered, do a `docker rmi IMAGEID` where the IMAGEID comes from the `images` output. Once all the images have been removed, bring up the container again. Then docker has no choice but to pull down the most recent base image and run the Dockerfile on top of that to produce a new local image.

Paraphraser, 2022-01-07:
If none of the above helps, run

```
$ cat ~/IOTstack/services/nodered/Dockerfile
```

and paste the result here (a line of triple back-ticks - ``` - then paste the text, then another line of triple back-ticks)

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke