SensorsIot/IOTstack #474: error while loading shared libraries

huseyinkogo, 2022-01-13:
Getting this error fresh install of Raspberry Pi 4
```
git clone https://github.com/SensorsIot/IOTstack.git IOTstack
Cloning into 'IOTstack'...
/usr/lib/git-core/git-remote-https: error while loading shared libraries: libhogweed.so.4: cannot open shared object file: No such file or directory
```
Tried reinstalling git curl
Tried rebooting

Paraphraser, 2022-01-13:
Not a lot to go on...

Just did an apt update/upgrade. Nothing in it about git. Then tried a clean clone which worked so, whatever is causing your problem isn't affecting me.

[image]

Have you tried pinging GitHub to see if it is reachable?

A lot of recent IOTstack problems turn out to be a combo of Buster and libseccomp2. Can't really see why that would affect git but if you're running Buster maybe see [recommended system patch](https://sensorsiot.github.io/IOTstack/Getting-Started/#patch-2-update-libseccomp2).

Alternatively, try rebuilding with [PiBuilder](https://github.com/Paraphraser/PiBuilder).

huseyinkogo, 2022-01-13:
@Paraphraser thanks for the reply. Yeah i will need to re install a clean copy. maybe try the Pibuilder  :)

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke