SensorsIot/IOTstack #479: building mosquitto fails

cmedianu, 2022-01-19:
Any ideas why building the mosquitto container fails?

Thanks!

pi@rasp4:~/work/IOTstack $ docker-compose up -d --remove-orphans
WARNING: Some networks were defined but are not used by any service: nextcloud_internal, vpn_nw, iotstack_nw_internal
Building mosquitto
Step 1/13 : FROM eclipse-mosquitto:latest
 ---> b24dbb9c5469
Step 2/13 : RUN sed -i 's/https/http/' /etc/apk/repositories
 ---> Using cache
 ---> fcefb76df897
Step 3/13 : RUN apk update && apk add --no-cache rsync tzdata
 ---> Running in be1d5167682f
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/main/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.14/main: temporary error (try again later)
WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.14/main: No such file or directory
fetch http://dl-cdn.alpinelinux.org/alpine/v3.14/community/armhf/APKINDEX.tar.gz
ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.14/community: temporary error (try again later)
WARNING: Ignoring http://dl-cdn.alpinelinux.org/alpine/v3.14/community: No such file or directory
2 errors; 20 distinct packages available
ERROR: Service 'mosquitto' failed to build: The command '/bin/sh -c apk update && apk add --no-cache rsync tzdata' returned a non-zero code: 2

**The network is fine:**

 curl -v -s http://dl-cdn.alpinelinux.org/alpine/edge/main/x86_64/APKINDEX.tar.gz > /dev/null
*   Trying 151.101.54.133...
* TCP_NODELAY set
* Expire in 149987 ms for 3 (transfer 0x7338b0)
* Expire in 200 ms for 4 (transfer 0x7338b0)
* Connected to dl-cdn.alpinelinux.org (151.101.54.133) port 80 (#0)
> GET /alpine/edge/main/x86_64/APKINDEX.tar.gz HTTP/1.1
> Host: dl-cdn.alpinelinux.org
> User-Agent: curl/7.64.0
> Accept: */*
>
< HTTP/1.1 200 OK
< Server: nginx
< Content-Type: application/octet-stream
< Last-Modified: Tue, 18 Jan 2022 22:31:48 GMT
< ETag: "61e73fd4-9f49a"
< Strict-Transport-Security: max-age=31536000
< X-Frame-Options: DENY
< X-Content-Type-Options: nosniff
< Via: 1.1 varnish, 1.1 varnish
< Content-Length: 652442
< Accept-Ranges: bytes
< Date: Wed, 19 Jan 2022 06:40:56 GMT
< Age: 8
< Connection: keep-alive
< X-Served-By: cache-lga21983-LGA, cache-sea4462-SEA
< X-Cache: HIT, HIT
< X-Cache-Hits: 1, 1
< X-Timer: S1642574457.614065,VS0,VE1
<
{ [2888 bytes data]
* Connection #0 to host dl-cdn.alpinelinux.org left intact

cmedianu, 2022-01-19:
This worked:
RUN echo http://69.65.16.171/alpinelinux/v3.14/community/ > /etc/apk/repositories
RUN echo http://69.65.16.171/alpinelinux/v3.14/main/ >> /etc/apk/repositories

apparently the DNS is not working, and not any mirror works, only mirrors not using HTTP1.1 virtual hosts, so you may have to try to resolve a few alpine linux mirrors to addresses...

closed 2022-01-19 by cmedianu

ukkopahis, 2022-01-20:
Did you apply libseccomp2 patch?
https://sensorsiot.github.io/IOTstack/Getting-Started/#patch-2-update-libseccomp2

This is already reported in https://github.com/SensorsIot/IOTstack/issues/409

DSew20, 2022-07-17:
Hello - I have run into the same problem:

`ERROR: http://dl-cdn.alpinelinux.org/alpine/v3.14/main: temporary error (try again later)`

I have tried the libseccomp2 patch, rebooted.  

I also followed this, and I have the lines added.  But, still get the error.

https://github.com/SensorsIot/IOTstack/issues/409

I am running PiHole - could this be messing with DNS?  Or, any other ideas re a fix for Mosquitto?

Paraphraser, 2022-07-17:
[image]

The screenshot shows a successful build a few moments ago so, given what you said, it's more likely that your problems will be explained by something peculiar to your setup than something inherently wrong with the basic build process.

Although I do run PiHole, I do **not** let the Pi itself use PiHole for DNS. Opinions vary on this but I think there are too many chicken-and-egg situations. It's only *other* hosts that I direct to PiHole for DNS and, even then, only those hosts that actually need ad-blocking services.

My guess is that either PiHole or your ISP's resolvers will be the root cause, the latter being the likely culprits (but only you will know whether you ultimately rely on your ISP's resolvers - eg if PiHole relays to your router, and your router just accepts whatever DNS servers are handed out by your ISP when your router connects).

As to solutions, I have three suggestions. First, I seem to remember people reporting getting around such problems by creating/editing (sudo) `/etc/docker/daemon.json` with content:

```
{
    "dns": ["8.8.8.8"]
}
```

followed by a `sudo systemctl restart docker`. I'm not sure of the exact scope of this so I'd be cautious about leaving it in place if it solved the problem with the build. That file doesn't exist by default so don't be surprised if it isn't there.

The second is to edit `/etc/resolvconf.conf` to add:

```
name_servers=8.8.8.8
```

followed by `sudo resolvconf -u`. This sends the whole Pi to 8.8.8.8 and bypasses PiHole. That may not be useful long term if you are doing more with PiHole than ad-blocking (eg you are using it to manage a local domain).

Suggestion 3 is to get into your router and tell it to relay DNS queries to 8.8.8.8 and 8.8.4.4. That will bypass your ISP's DNS and **is** a viable long-term option.

Hope this helps.