SensorsIot/IOTstack #484: Airthings wave radon/voc/... integration - Bluetooth device in a docker container

Tzaphkiel, 2022-01-23:
I have fought to get this working and thought it might help someone using the platform and this stack.

I have used the following HA integration:
* https://github.com/custom-components/sensor.airthings_wave

You have to make sure you define the sensor with a `mac` address otherwise (for me) it did not work. You can find it using the `find_wave.py` script here:
* https://www.airthings.com/resources/raspberry-pi

```yml
- platform: airthings_wave
  scan_interval: 300
  mac: e4:xx:xx:xx:xx:xx:
```

To make bluetooth work in the container, I had to modify the `docker-compose.yml` file and add the following:
```yml
  home_assistant:
    volumes:
      # ...
      - /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket
    devices:
      - "/dev/ttyAMA0:/dev/ttyAMA0"
      - "/dev/vcio:/dev/vcio"
      - "/dev/gpiomem:/dev/gpiomem"
    privileged: true
```

I then checked that `bluetooth` was enabled on the host by editing `/etc/bluetooth/main.conf`
```conf
[Policy]
AutoEnable=true
```

I then checked after a reboot that BT was up:
```bash
hciconfig
...
UP
...
```

ukkopahis, 2022-01-23:
This is something you could add at the end of the Home-Assistant documentation and submit as a proposed change:

https://github.com/SensorsIot/IOTstack/edit/master/docs/Containers/Home-Assistant.md

Tzaphkiel, 2022-01-23:
Tbh, I think this should be included in the default setup.
Probably here: https://github.com/SensorsIot/IOTstack/blob/master/.templates/home_assistant/service.yml

ukkopahis, 2022-01-24:
Yes, having service.yml correct out-of-the box is preferred. Still, the mac address suggestion would go to docs?

Tzaphkiel, 2022-01-24:
I should think that the mac address part actually belongs to the
integration’s doc.

Btw, the services.yml part is corroborated by the experiments and very
lengthy details of someone in another issue in this repo; search about
bluetooth in the issues.

ukkopahis, 2022-01-25:
Could you create this into a pull-request? Otherwise it most likely won't be added.

PR #485 (2022-02-23): Enable BT & BLE in Home Assistant container
Fixes #484 by:
* updating `service.yml` for Home Assistant container
* updating associated documentation
commit b57ec021 (2022-02-23): Allow using BT and BLE devices from HA

This closes #484 by allowing the powering on of bluetooth inside the Home Assistant container and using the `hci` devices otherwise not accessible.
This was initiated to be able to use the Airthings Wave Plus device that uses BLE from an integration interacting with `bluetoothd` and the `/dev/hci` directly.
closed 2022-02-23 by Slyke