SensorsIot/IOTstack #486: IOTstack won't build containers

larduino, 2022-02-03:
I just tried adding tasmoadmin container . My IOTstack is running on Bullseye on Pi3B+ and has been working fine for long time running mosquitto, portainer, telegraf, influxdb, and grafana as well as a couple compose-overide containers for my solar system controllers. But when I went menu and selected tasmoadmin , hit enter, then went to Build Stack - it does not build the new container. If I go look in the docker-compose.yml the info for the tasmoadmin is in there . 
  tasmoadmin:
    container_name: tasmoadmin
    image: raymondmm/tasmoadmin
    restart: unless-stopped
    ports:
    - "8088:80"
    volumes:
    - ./volumes/tasmoadmin/data:/data
Is there a way to force this to build ? 
This may be related - I was helping a friend the other night use IOTStack and it was doing the same thing - not building containers. 
Then last night I tried IOTStack but was using on older raspberry pi model B I think - and same thing - containers would not build. I tried running just ./menu.sh and also sudo ./menu.sh  . I tried two different sd cards last night with fresh updated Bullseye and the IOTstack installs and builds didn't work properly - one did sort of when I used sudo ./menu.sh . I chalked that up to it being an old raspberry pi that can't run IOTstack.  
But is there a solution for the newer P3B+ model that I am using now ? 
Thanks

Paraphraser, 2022-02-03:
Short answer: works for me.

This is on a 3B+ running Buster:

[image]

and this on a 4B running Bullseye 64-bit:

[image]

<hr>

But let's unpack this a bit because it might turn out to be a matter of terminology or expectations.

First off, please don't try to run the menu with `sudo` because that is only ever going to compound your problems. See [a word about the sudo command](https://sensorsiot.github.io/IOTstack/Getting-Started/#a-word-about-the-sudo-command). It's **possible** that running the menu with `sudo` has broken something but I honestly don't know how to diagnose that.

Next, the word "build" can mean different things. The IOTstack menu uses "build" to mean "construct `~/IOTstack/docker-compose.yml` from menu choices". Docker uses "build" to mean "run a Dockerfile on top of a base image to build a local image".

If a service is built from a Dockerfile, it needs a "build:" directive in its service definition:

```
$ find ~/IOTstack/.templates -name "service.yml" -exec grep -H "build:" {} \;
/home/pi/IOTstack/.templates/telegraf/service.yml:  build: ./.templates/telegraf/.
/home/pi/IOTstack/.templates/python/service.yml:  build: ./services/python/.
/home/pi/IOTstack/.templates/blynk_server/service.yml:  build:
/home/pi/IOTstack/.templates/prometheus/service.yml:  build: ./.templates/prometheus/.
/home/pi/IOTstack/.templates/nodered/service.yml:  build: ./services/nodered/.
/home/pi/IOTstack/.templates/nextcloud/service.yml:  build: ./.templates/mariadb/.
/home/pi/IOTstack/.templates/mariadb/service.yml:  build: ./.templates/mariadb/.
/home/pi/IOTstack/.templates/mosquitto/service.yml:  build: ./.templates/mosquitto/.
/home/pi/IOTstack/.templates/zigbee2mqtt/service.yml:  build: ./.templates/zigbee2mqtt/.
/home/pi/IOTstack/.templates/rtl_433/service.yml:  build: ./services/rtl_433/.
```

You won't see tasmoadmin in that list so that means it doesn't actually get "built" as a local image and that, in turn, means there's no way to "force" it to "build" in the *Dockerfile* sense. So that leaves us with the *menu* sense.

Here's what *should* happen. I'm doing this on a 3B+ running Buster:

1. Prove nothing is running:

	```
	$ docker ps
	CONTAINER ID   IMAGE     COMMAND   CREATED   STATUS    PORTS     NAMES
	```

2. A clean clone of IOTstack:

	```
	$ git clone https://github.com/SensorsIot/IOTstack.git
	Cloning into 'IOTstack'...
	remote: Enumerating objects: 7681, done.
	remote: Counting objects: 100% (1794/1794), done.
	remote: Compressing objects: 100% (346/346), done.
	remote: Total 7681 (delta 1478), reused 1510 (delta 1375), pack-reused 5887
	Receiving objects: 100% (7681/7681), 7.30 MiB | 7.62 MiB/s, done.
	Resolving deltas: 100% (4721/4721), done.
	```

3. Run the menu:

	* Cursor on "Build Stack", press <kbd>enter</kbd>
	* Cursor on "tasmoadmin", press <kbd>space</kbd> to select
	* Press <kbd>enter</kbd> to begin build (ie "construct docker-compose.yml")
	* Cursor on "Exit", press <kbd>enter</kbd>

	```
	$ cd IOTstack
	$ ./menu.sh 
	Checking for project update
	From https://github.com/SensorsIot/IOTstack
[... 135 lines cut ...]

One thing I did notice is this:

```
$ docker images raymondmm/tasmoadmin
REPOSITORY             TAG       IMAGE ID       CREATED         SIZE
raymondmm/tasmoadmin   latest    664c6c070847   16 months ago   40.3MB
```

Given the usual fast pace of change for all things IT, not being updated in 16 months seems like it might be a cause for concern.

I did a [DockerHub search](https://hub.docker.com/search?q=tasmoadmin&type=image). Nine hits. The one updated most recently is [lettore/tasmoadmin](https://hub.docker.com/r/lettore/tasmoadmin/tags) with 2.6k downloads. Not quite in the same league as raymondmm/tasmoadmin with 50M downloads but enough to suggest it's probably worth a look.

However, no matter what combination of "lettore/tasmoadmin" and tag I tried, I always wound up with something created 16 months ago and, in the case of the 3B+, with identical Image IDs (SHA256 digests). I have no idea what that means.

<hr>

Anyway, to summarise:

1. "build" in the Dockerfile sense doesn't apply to tasmoadmin.
2. "build" in the menu sense generates a service definition pulling raymondmm/tasmoadmin. I've shown it working on both a 3B+ running Buster 32-bit and a 4B running Bullseye 64-bit.
3. the available raymondmm images are 16 months old.
4. the most likely candidate for something more up-to-date is lettore/tasmoadmin but it also seems stuck 16 months ago.

I hope something in all of the above helps you to zero-in on the problem.

larduino, 2022-02-04:
I tried checking some of the things you mentioned. 
Now all my docker containers are down and won't come up again even after a reboot. 
I tried update all containers but this is all I get 
                                                                             ╚════════════════════════════════════════════════════════════════════════════════╝
Update All Containers:
docker-compose down
WARNING: Some networks were defined but are not used by any service: nextcloud
ERROR: Service "classic_mqtt" uses an undefined network "iotstack_nw"

docker-compose pull
WARNING: Some networks were defined but are not used by any service: nextcloud
ERROR: Service "classic_mqtt" uses an undefined network "iotstack_nw"

docker-compose build
WARNING: Some networks were defined but are not used by any service: nextcloud
ERROR: Service "classic_mqtt" uses an undefined network "iotstack_nw"

docker-compose up -d
WARNING: Some networks were defined but are not used by any service: nextcloud
ERROR: Service "classic_mqtt" uses an undefined network "iotstack_nw"

Process terminated. Press [Enter] to show menu and continue.

larduino, 2022-02-04:
These are the versions I have 
`pi@PiBullseye:~/IOTstack $ docker -v
Docker version 20.10.5+dfsg1, build 55c4c88
pi@PiBullseye:~/IOTstack $ docker-compose version
docker-compose version 1.25.0, build unknown
docker-py version: 4.1.0
CPython version: 3.9.2
OpenSSL version: OpenSSL 1.1.1k  25 Mar 2021
pi@PiBullseye:~/IOTstack $'

I am wondering now if there is some files I deleted - when i did a Pi update / upgrade it said there were some files not needed and auto remove would clean them. I forget what they were - like lib something ?

larduino, 2022-02-04:
I tried reinstalling - also it says on startup that menu might crash because of dependencies and my menu does act all weird. 

`Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python Version: 'Python 3.9.2'. Python is up to date.
ruamel.yaml Version: 'ruamel.yaml 0.16.12'. ruamel.yaml is up to date.
Blessed Version: 'blessed 1.19.0'. Blessed is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
**Project dependencies not up to date. Menu may crash.
To be prompted to update again, run command:
  rm .docker_notinstalled || rm .docker_outofdate || rm .project_outofdate**

Existing installation detected.
[image]
`

Paraphraser, 2022-02-04:
I think you might be caught in the middle of the networking transition.

Perhaps read [significant change to networking](https://github.com/SensorsIot/IOTstack#significant-change-to-networking).

It doesn't really matter whether you're running old or new menu because the networking is now the same for both but you mentioned "override" so I'm going to assume you're running new menu. Make sure everything is fully up-to-date:

```
$ cd ~/IOTstack
$ git checkout old-menu
$ git pull
$ git checkout master
$ git pull
```

As explained in the link above, the **only** service definitions that should have **any** `networks:` directives are `nextcloud` and `nextcloud_db`. Consider that to be "rule 1". You should inspect:

```
~/IOTstack/docker-compose.yml
~/IOTstack/services/docker-compose.save.yml
```

and make sure that "rule 1" applies.

You should also inspect your override file and make sure "rule 1" applies there too.

Next, look at:

```
~/IOTstack/.templates/env.yml
```

That **defines** the networks (the "rule 1" stuff tells containers which networks to use). Everything you see in that "env.yml" should be considered to be "rule 2".

Repeat the check on all the files (compose, save, override) and, wherever you see network definitions, replace them with the contents of `env.yml`. Don't get creative. Don't mix and match. **Replace**!!!

With all that in place, the menu should "play nice".

But, if it were me, I'd probably just list the tasmota service definition:

```
$ cat ~/IOTstack/.templates/tasmoadmin/service.yml 
tasmoadmin:
  container_name: tasmoadmin
  image: raymondmm/tasmoadmin
  restart: unless-stopped
  ports:
    - "8088:80"
  volumes:
    - ./volumes/tasmoadmin/data:/data
```

and observe that it is left-shifted by two spaces. I'd fix that with:

```
$ sed -e "s/^/  /" ~/IOTstack/.templates/tasmoadmin/service.yml >~/tas.yml
```

and then use copy-and-paste techniques to get the contents of `~/tas.yml` into my compose file. But that's just me.

<hr>

I see another message has come in since I started to write the above. I'm being called for lunch. I'll take a look at that when I come back.

Paraphraser, 2022-02-04:
At this point, you might want to consider a system rebuild using [PiBuilder](https://github.com/Paraphraser/PiBuilder). If you do that, make sure you use an up-to-date [image from Raspberry Pi](https://www.raspberrypi.com/software/operating-systems/). I always start from "Raspberry Pi OS with desktop". I am running a mixture of "full 32-bit", "32-bit user + 64-bit kernel" and "full 64-bit".

> As far as I'm concerned, all those Bullseye combos work **except** for OctoPrint/Docker which seems to have dependencies on Buster (for camera support) and doesn't really like 64-bit - but that might have something to do with it running on my 3B+.

But, if you don't want to do that, you could also consider nuking what's there and starting over.

If any part of your stack is running, take it down. Then:

```
$ sudo apt -y purge docker-ce docker-ce-cli containerd.io
$ sudo apt -y remove docker-compose
$ sudo pip3 uninstall -y docker-compose
$ curl -fsSL https://get.docker.com | sudo sh
$ sudo usermod -G docker -a $USER
$ sudo usermod -G bluetooth -a $USER
$ sudo reboot
```

Depending on how recently/long ago you did a system update, you might also want to consider:

```
$ sudo apt update
$ sudo apt upgrade -y
```

Next, move your existing IOTstack out of the way:

```
$ cd
$ mv IOTstack IOTstack.old
```

> If that `mv` doesn't work then it's OK to force it with `sudo` but understand that that is actually evidence of a much bigger problem.

Get a clean copy of IOTstack

```
$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
```

Merge the persistent storage areas (this does need `sudo`):

```
$ sudo mv IOTstack.old/volumes IOTstack/
```

Everything left behind in the old copy should be owned by "pi" (well, there's one exception - `backups/influxdb` - but it doesn't really matter) so let's make sure that assumption holds:

```
$ sudo chown -R pi:pi IOTstack.old
```

Move other bits and pieces:

```
$ mv IOTstack.old/docker-compose.yml IOTstack/
$ mv IOTstack.old/services IOTstack/
```

If there's anything else in the old folder you want to bring across, you can use the same approach.

> I don't use "override" files so I don't know whether they are loose in `~/IOTstack` or in the `services` directory. You'll have to figure that out.

OK. Now we're at the point where `docker` is installed, the groups are set correctly, we have a clean clone of IOTstack, and all the critical bits have been brought across. That leaves `docker-compose`:

```
$ cd
$ wget https://raw.githubusercontent.com/Paraphraser/PiBuilder/master/boot/scripts/helpers/upgrade_docker_compose.sh
$ chmod +x upgrade_docker_compose.sh
$ ./upgrade_docker_compose.sh
```

That will pull down and install version 2.2.3 of docker-compose. It finishes off by recommending that you add two lines to your `.profile` (or perhaps your `.bashrc` if that's how you login):

```
export COMPOSE_DOCKER_CLI_BUILD=1
export DOCKER_BUILDKIT=1
```

Then logout and login so those take effect.

Now you should be ready to roll.

<hr>

Odds and ends:

* Assuming the above gets you a bit further, at some later time you will want to remember to see if there's anything left in `~/IOTstack.old` that needs to be preserved, and then clobber that folder.
* The `upgrade_docker_compose.sh` script is part of [PiBuilder](https://github.com/Paraphraser/PiBuilder). It currently defaults to v2.2.3. There is, as yet, no way of writing a script which senses when a newer version has become available. You need to figure that out for yourself from the [releases](https://github.com/docker/compose/releases) page, after which you can invoke the script with the newer version number. For example, when v2.2.4 is released:

	```
	./upgrade_docker_compose.sh v2.2.4
	```

larduino, 2022-02-04:
thanks - I will check into the change to networking  and read the link you have there . 
I changed some of the networks in existing files on my backup pi and it seemed to work. 
Now I am going to change everything to nextcloud in my yaml files. 
I will let you know how it goes
thanks

Paraphraser, 2022-02-04:
> Now I am going to change everything to nextcloud in my yaml files.

Errr - if that means what I think it means (that you intend to use `nextcloud` everywhere), it probably won't work.

When we're talking about networks, `default` becomes the `iotstack_default` network, and `nextcloud` becomes the `iotstack_nextcloud` network. These networks are created when the stack comes up.

The `nextcloud` (`iotstack_nextcloud`) network is private to the `nextcloud` and `nextcloud_db` containers. It has no ability to communicate with the outside world.

The service definition for the `nextcloud` container should attach to `default` (`iotstack_default`) and `nextcloud` ('iotstack_nextcloud`).

The service definition for the `nextcloud_db` container should only attach to `nextcloud` ('iotstack_nextcloud`).

No other container (repeat **no** other container) should attach to any network.

The **absence** of a `network:` clause in any service definition means "attach to `default`" (which is the `iotstack_default` network).

The service definition for the `nextcloud` container needs an explicit connection to `default` (`iotstack_default`) because it **also** needs an explicit connection to `nextcloud` ('iotstack_nextcloud`).

The service definition for the `nextcloud_db` container doesn't need an explicit connection to `default` (`iotstack_default`) because it **only** needs an explicit connection to `nextcloud` ('iotstack_nextcloud`).

In other words, the `nextcloud` and `nextcloud_db` containers define a private network because they need to talk to each other but the `nextcloud_db` container doesn't need to talk to the outside world so the 'iotstack_nextcloud` network isn't exposed. All communication with the outside world goes through the `nextcloud` container so it needs to be attached to a network (`iotstack_default`) which **is** exposed to the outside world, while also having a private network to communicate with its database engine.

I hope that makes sense.

larduino, 2022-02-04:
I did your nuke approach and finally it all seems to working. 
Not sure why but initially Influxdb didn't build. I had to run the menu a few times - maybe it was something I didnt' check in one of the boxes ? Anyway it finally made influx and everything I had before seems to be running and I didn't loose my grafana dashboard or anything else except for the past data which is no big deal. 
Thanks again for your help - what a weird problem with those networks - seems like it just came with no warning. 
I didn't realize that the new menu was so different - I was thinking the old menu was from way back . Anyway nice to see how everything works now.

Paraphraser, 2022-02-04:
I think there was a lot more going on than just the networking changes. The error messages about docker and groups were nothing to do with networking.

You should not have lost any data. That's what the "mv" of the volumes directory was supposed to avoid. If the "mv" worked but you still lost data then either the data was lost before we started down this path or something else is going on. Some containers self-repair permissions but it isn't universal. What I do is run up a dummy stack and see how each container sets up ownership and permissions when given a clean slate, then mimic that by hand in the live volumes folder. No idea if it will help you recover any data... 🙁

larduino, 2022-02-04:
Yeah who knows what I did to it - I tried so many things before the nuke approach. And I only half know what I am doing. 
It could be that the data is there - but the telegraf to influx to grafana is working for my tasmota sonoff devices but not for my solar controllers . I will figure that out tomorrow . They go to same grafana through same route so just solar part of it is messed up. 1:30 am bedtime !

larduino, 2022-02-04:
> > Now I am going to change everything to nextcloud in my yaml files.
> 
> Errr - if that means what I think it means (that you intend to use `nextcloud` everywhere), it probably won't work.
> 

> In other words, the `nextcloud` and `nextcloud_db` containers define a private network because they need to talk to each other but the `nextcloud_db` container doesn't need to talk to the outside [...]

**Yes in the light of the new day it makes sense - I changed networks in the solar controllers compose-overrride.yml from iotstack_nw to default  and it now works . 
The old data never filled in but that is okay - it is working for new data now.**

larduino, 2022-02-04:
Is this the new or old menu ? I used to have a different one . Doesn't matter I guess because it works. 
[image]

closed 2022-02-05 by larduino