SensorsIot/IOTstack #490: menu.sh configures the wrong networks

Moddingear, 2022-02-06:
When building for **Grafana, Mosquitto, Node-Red and InfluxDB**, all modules in docker-compose.yml are configured with network **iotstack_nw**, which doesn't exist. Instead there are two networks configured : **nextcloud** and **default**.
Switching the networks in all modules to use default makes it work again.

tester277, 2022-02-06:
Confirm -thats my issue too. Thank You for a workaround.

Paraphraser, 2022-02-06:
Please see [significant change to networking](https://github.com/SensorsIot/IOTstack#significant-change-to-networking). I suspect that what you are seeing is an artefact of existing services as recorded by the menu in `~/IOTstack/services/docker-compose.save.yml` vs what is now defined in the templates. The screen shot shows `iotstack_nw` isn't coming from the templates so the "save" file is the most likely source (the other possibility is an override file).

[image]

The readme instructions do say to re-select all containers but perhaps it wasn't clear enough or firm enough. It really means "run the menu and de-select everything that is selected, then run the menu again and re-select the containers you want." The de-selection causes old service definitions (those with `iotstack_nw`) to be removed from the "save" file. The re-selection causes new service definitions to come in from the template (where `iotstack_nw` no longer exists). In "old menu" parlance, this is the equivalent of "pull full service from template".

Another approach would be to take the stack down, delete the "save" file, run the menu and select the containers you want, then "up" again. Deleting the save file causes the menu to forget all the selections.

Yet another approach would be to hand-edit the save file to conform with revised networking.

I'm really sorry about the chaos that this has created for you but, with all the approaches people use for managing their IOTstacks, it really wasn't prudent to try to automate this migration. The good thing is the networking is now simplified, is common to both old and new menu systems, and returns to letting Docker allocate the internal networks dynamically from the expected range (addresses an obscure problem which would trip people up from time to time).

Plus, with modern docker-compose, those pesky "network defined but not used" will go away as well. Those messages often caused new users to think they had broken something during their first installation.

PR #518 (2022-04-26): docs: Add more warnings about the breaking network change
The recent network change breaks the stack for existing users, requiring
manual fixes.

Add links to doc pages where Discord users have reportedly looked for
solutions, without finding the right place.

Closes #490

[image]
[image]
[image]
closed 2022-04-26 by Slyke