SensorsIot/IOTstack #491: Home assistant native install doesn't install al all.

ralong777, 2022-02-08:
I've done everything in the guide 
https://sensorsiot.github.io/IOTstack/Containers/Home-Assistant/

And at the end I get a mesage:

'''
Preinstallation complete. Your system may run slow for a few hours as Hass.io installs its services.
Press [Up] or [Down] arrow key to show the menu if it has scrolled too far.
Process terminated. Press [Enter] to show menu and continue.
'''

But nothing ever happens. I tried on rpi4 32bit rpi4 64bit and rpi3b+
Always the same

Paraphraser, 2022-02-09:
Here's the basic problem. The sequence `./menu.sh` » "Native Installs" » "Hass.io (Supervisor)" tries to run:

```
~/IOTstack/.native/hassio_supervisor.sh
```

and that, in turn, invokes:

```
curl -sL "https://raw.githubusercontent.com/Kanga-Who/home-assistant/master/supervised-installer.sh" | sudo bash -s
```

It is that "Kanga-Who" script which is broken. As far as I can tell, it is broken because the Home Assistant folks changed how Supervised Home Assistant (aka Home Assistant Core) was installed. If you go to [Kanga-Who/home-assistant](https://github.com/Kanga-Who/home-assistant), you'll see that the script hasn't been updated in a while – but some recently-updated .md files talk about how it is difficult.

It is also true to say that there are some order-of-installation dependencies between HA and IOTstack which always made that "convenience script" installation method slightly sub-optimal. 

The two paragraphs above are also the answer to the question you will likely ask as to why we don't just fix the `hassio_supervisor.sh` script to do the job properly? It's somewhere between *tricky* and *unreliable*, particularly if you consider it from the perspective that the script might be invoked with both new and existing installations, where existing installations can have a wide variety of docker and docker-compose versions. It's just far too messy. It would probably be a good idea to just remove the item from the menu, and change the guide so it doesn't promise what it can't actually deliver.

It seems clear that the Home Assistant folks want you to run their purpose-built image which effectively dedicates a Pi to Home Assistant and which, in turn, excludes the possibility of IOTstack running on the same machine. I don't know about you but my reaction to that idea is … unprintable.

Right now, the most reliable way to install Supervised Home Assistant such that it cohabits properly with IOTstack is to start with a clean Raspbian image and use [PiBuilder](https://github.com/Paraphraser/PiBuilder).

When you visit the PiBuilder link you may well have a reaction like "yikes - far too complicated" but you should try to get past that. PiBuilder has two main use-cases:

1. Getting a Raspberry Pi built for IOTstack (and, optionally, Supervised Home Assistant) with the least fuss.
2. Letting you record all your own customisations so that you can rebuild Pis with your customisations already in place.

It's the second bit that produces most of the apparent complexity.

The first-use process boils down to:

1. Clone the PiBuilder repo onto your support host (Mac, Windows, etc).
2. Customise two files within the PiBuilder scope:

	- `wpa_supplicant.conf`
	- `options.sh` where, among other things, you will enable:

		- `HOME_ASSISTANT_SUPERVISED_INSTALL=true`

3. Choose a Raspbian image and transfer it to your installation media (SD/SSD). That typically ends by ejecting the installation media.
4. Re-mount the installation media on your support host and either:

	- Run the supplied `setup_boot_volume.sh` script (if your support host is macOS or Unix); or
	- Just drag the *contents* of the PiBuilder "boot" folder into the top level of the "/boot" partition on your installation media (if your support host is Windows).

5. Move the installation media to your Raspberry Pi and apply power.
6. Run the scripts in order (left-aligned is support host, indented is RPi):

	```
	$ ssh -4 pi@raspberrypi.local
	--->	$ /boot/scripts/01_setup.sh «newHostName»
	
	$ ssh-keygen -R raspberrypi.local
	
	$ ssh -4 pi@«newHostName».local
	--->	$ /boot/scripts/02_setup.sh
	
	$ ssh pi@«newHostName».local
	--->	$ /boot/scripts/03_setup.sh
	
	$ ssh pi@«newHostName».local
	--->	$ /boot/scripts/04_setup.sh
	
	$ ssh pi@«newHostName».local
	--->	$ /boot/scripts/05_setup.sh
	```

After the 04 script completes, Supervised Home Assistant will be running. Once the 05 script completes, you'll be able to either:

1. Restore a backup (if you aren't taking backups already perhaps see [IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup)); or
2. Run the menu and choose your containers.

Here's my PiBuilder-produced IOTstack with Home Assistant running on a Raspberry Pi 4B/4GB + SSD + 64-bit Bullseye:

```
$ docker ps
CONTAINER ID   IMAGE                                                             COMMAND                  CREATED          STATUS                    PORTS                                                                              NAMES
52570e1c55a0   ghcr.io/home-assistant/raspberrypi4-64-homeassistant:2021.12.10   "/init"                  18 minutes ago   Up 18 minutes                                                                                                homeassistant
d6413e09dfdb   ghcr.io/linuxserver/wireguard                                     "/init"                  18 minutes ago   Up 18 minutes             0.0.0.0:51720->51820/udp                                                           wireguard
26215e626090   iotstack_mosquitto                                                "/docker-entrypoint.…"   18 minutes ago   Up 18 minutes (healthy)   0.0.0.0:1883->1883/tcp                                                             mosquitto
b99aec0b0575   portainer/portainer-ce                                            "/portainer"             18 minutes ago   Up 18 minutes             0.0.0.0:8000->8000/tcp, 0.0.0.0:9000->9000/tcp, 9443/tcp                           portainer-ce
[... cut ...]
3cceb83892b1   ghcr.io/home-assistant/aarch64-hassio-cli:2021.12.0               "/init /bin/bash -c …"   19 minutes ago   Up 19 minutes                                                                                                hassio_cli
014239253da3   homeassistant/aarch64-hassio-supervisor                           "/init"                  19 minutes ago   Up 19 minutes                                                                                                hassio_supervisor
```

Hope this helps.

ukkopahis, 2022-02-23:
@ralong777 This is now fixed by #493, could you please close it?

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke