SensorsIot/IOTstack #495: Homebridge & NodeJS

gjeida, 2022-02-09:
I now have two plugins for homebridge that insist you have Node.js v16.0.0 or later.

`"The homebridge-webos-tv plugin recommends running on Node.js v16.0.0 or later. You are currently running Node.js v14.15.5."`

I have obviously checked if there are any container updates available and it is telling me there are not. Is this a "wait until the container maintainer sorts this out" thing or manually update homebridge in the container or am I just being dumb?

Thanks

Paraphraser, 2022-02-09:
Not "dumb" - just a victim of poor timing. Try a `docker-compose pull`:

[image]

I do this kind of thing a lot myself - lose patience - ask questions - and then it turns out that if I'd just waited a *bit* longer, the problem would have solved itself. 🤦😎

gjeida, 2022-02-09:
haha yep I been waiting months, just got re-inspired to try last night as another plugin was asking for the updated JS, just should have waited a few more hours lol... 

Thanks and thanks for all the hard work on this  :-)

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke