SensorsIot/IOTstack #504: mosquitto password documentation off

the1gofer, 2022-02-21:
the wiki says

> To create a username and password, use the following as a template.
>
>$ docker exec mosquitto mosquitto_passwd -b /mosquitto/pwfile/pwfile «username» «password» 
>Replace «username» and «password» with appropriate values, then execute the command.

when you use that command with the current image, you get 
>-bash: $: command not found

From what I gather whatever image that's based off doens't have bash.  If you do 
`docker exec -it sh`. then
`mosquitto mosquitto_passwd -b /mosquitto/pwfile/pwfile «username» «password» `
it seems to work.

Paraphraser, 2022-02-21:
I suspect a typo. The Mosquitto image has both `ash` and `sh` (as symlinks to busybox) but that really only matters if you need shell services.

[image]

In words:

1. Show the version of Mosquitto running inside the container (2.0.14).
2. Go into the external directory holding the password file.
3. Show the password file is empty.
4. Add a password (note that no `-it` and no `sh` is needed).
5. Show the password made it into the password file.
6. Delete the password using the same approach as step 4.
7. Show the password file is empty again.

If you can't replicate then there must be some explanation other than the mosquitto_password not being there because that command is still in my instance and I didn't do anything special.

Also, once you are running in an interactive shell inside the container, `mosquitto mosquitto_password` will invoke "mosquitto" and pass it "mosquitto_password" etc as arguments. When I try that I get "Error: Unknown option 'mosquitto_passwd'." so, to me, that means mosquitto  (the command) is unable to invoke mosquitto_passwd (another command).

I wonder if you're misunderstanding the command syntax. It's:

```
$ docker exec container command arguments
```

so, in something like:

```
$ docker exec mosquitto mosquitto -h
```

the first "mosquitto" is the container, the second "mosquitto" is the command to run inside the container, and "-h" is an argument to pass to that command running inside the container.

Make sense?

the1gofer, 2022-02-21:
anything is possible, but I thought I was copy pasting.   I went back and tried to add another one, and it worked.  So I don't know.     Thanks for your detailed response.

Paraphraser, 2022-02-21:
You're welcome. I did something similar the other day. The commands were still in my history file and they looked perfectly OK but, somehow, there was this problem in the space between the keyboard and the chair... 🤦‍♂️

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke