SensorsIot/IOTstack #506: mosquito problems

the1gofer, 2022-02-23:
I created a password file, and had clients connected, but at some point the container restarted and now I'm getting the following error.

from  docker container psr:
`b244a6ac0577   iotstack_mosquitto              "/docker-entrypoint.…"   6 minutes ago   Restarting (13) 19 seconds ago                                                                                      mosquitto`

from the logs:
```
2022-02-23T16:13:02.280538764Z 2022-02-23T16:13:02: mosquitto version 2.0.14 starting
2022-02-23T16:13:02.280594059Z 2022-02-23T16:13:02: Config loaded from /mosquitto/config/mosquitto.conf.
2022-02-23T16:13:02.280616355Z 2022-02-23T16:13:02: Error: Unable to open pwfile "enabled".
2022-02-23T16:13:02.280635799Z 2022-02-23T16:13:02: Error opening password file "enabled".
```

I found an article that said this was probably becuase the pwfile has the wrong owner, 
`-rw-r--r-- 1 1883 1883 364 Feb 21 09:00 pwfile`

id -nu 1 says the owner is doemon.  

any suggestions how to fix?

ukkopahis, 2022-02-23:
Try renaming ./volumes/mosquitto/config to config.bak and restart the container. The container will recreate the missing config-folder. Look at what the file permissions and owners are in the container created "clean" configs. Fix your config.bak to match those, stop the container, delete "config" and rename your original "config.bak" back to "config".

Paraphraser, 2022-02-24:
Inside the container, user ID 1883 is "mosquitto". The container expects to launch as root (the RPi's root), then the self-repair code runs which recreates any missing files/folders and unconditionally sets ownership to 1883:1883. Then the Mosquitto process (the broker) is launched which immediateAly downgrades its permissions to user 1883.

If you are able to create any file/folder in the persistent storage area (`~/IOTstack/volumes/Mosquitto`), then launch ("up") or restart the Mosquitto container and that does not result in the file changing ownership to 1883:1883 then it implies something else is wrong. Perhaps there is a "user" statement in the compose fragment for Mosquitto? There should not be.

Otherwise, I'd be suspecting some problem with pwfile itself. If you delete (or rename) that file and restart the container, a new null pwfile should be created. Then you can start adding usernames and passwords to it as documented.

Paraphraser, 2022-02-24:
[image]

config and pwfile folders and contents come from the template. The others are created by Mosquitto as it runs. Log may not be there if you let all logging go through docker.

the1gofer, 2022-02-26:
ok so I've managed to recreate the issue. 

I'm still learning quite bit about linux, containers, etc, so I ended up starting from a fresh image, resetting up IOTstack.  

https://www.dropbox.com/s/8rs4bf2tnmfi9uj/Screen%20Shot%202022-02-25%20at%206.37.16%20PM.png?dl=0

Is it possible it has to do with  the fresh setup? 

https://www.dropbox.com/s/f9m8rdapt10ahp6/Screen%20Shot%202022-02-25%20at%206.41.21%20PM.png?dl=0

Paraphraser, 2022-02-26:
That's true. Mosquitto is **very** sensitive to **any** of its files being futzed with. That's got nothing to do with IOTstack. In historical terms, Mosquitto was always a problem. It is less of a problem with the self-repair code in place but the self-repair code can't fix everything.

If you really want to open a shell into the Mosquitto container, use "ash":

[image]

It's possible that "restart" isn't strong enough and that a recreate is required:

```
$ cd ~/IOTstack
$ docker-compose up -d --force-recreate mosquitto
```

Paraphraser, 2022-02-26:
Actually, a restart is sufficient.
[image]

the1gofer, 2022-02-27:
When you say restart you mean `docker restart mosquito`.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke