SensorsIot/IOTstack #517: postgres documentation incorrect

the1gofer, 2022-03-15:
The wiki says:

> PostgreSQL is an SQL server, for those that need an SQL database. The database credentials can be found in the file ./volumes/postgres/postgres.env. It is highly recommended to change the user, password and default database

I'm not sure if I missed something, but that file doesn't seem to exist. 

[image]

Paraphraser, 2022-03-15:
When IOTstack began with gcgarner/IOTstack, he (Graham Garner) used the approach of a separate environment file in the services directory. In the SensorsIot fork, that scheme is what we sometimes refer to as "old menu". You can go back to that if you do a:

```
$ cd ~/IOTstack
$ git checkout old-menu
```

New menu (on the master branch) made several changes and one of those was to move environment variables inline to docker-compose.yml.

Basically, this is a case of the documentation lagging behind reality. I try to fix these things when I'm "working in the area" but this is one that nobody has gotten to yet. Feel free to [submit a pull request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d). What I generally do with documentation is prepare two parallel PRs. The one for "old-menu" branch reduces the doco on that branch to a stub pointing to the equivalent page on the master branch wiki, and the other PR actually dealing with the substance. I deal with any old/new menu differences in the "authoritative" doco on master branch.

the1gofer, 2022-03-17:
I'll give it a shot.  I've never really contributed to a project like this.

the1gofer, 2022-03-17:
I'm a little confused.

when I found the template in SensorsIot/IOTstack/tree/master/.templates/postgres/service.yml
it says
`- POSTGRES_PASSWORD=%randomPassword%`

However when  running the script, the password is always `IOtSt4ckpostgresDbPw`.    I don't know enough about how the yml is generated, but I'm guessing the randomPassword is broken.

Paraphraser, 2022-03-17:
When you run the menu and put the cursor on Postgres, press the right arrow to select the options. That's where you'll find the ability to set the password.

At least, that's the theory.

I just tried it and the menu crashed so, yup, it's broken.

> and I see you worked that out for yourself as I was typing this. 🤦‍♂️

The Python file producing that error is `~/IOTstack/.templates/python/build.py`. If you're a Python person you could try to figure out why and fix it but what I've been doing is just deleting broken build.py files and changing the service definition to use whatever default password is specified for old menu:

[image]

The above is talking about Pull Request preparation. Now I'll turn to your "what's my password?" question. The thing about usernames and passwords passed via environment variables is that, in 99% of cases, they are only used to initialise the container. So that means if "%randomPassword%" is a valid password string and the container was fired up the first time with that in place then that's the password. Forever. Can't ever change it again via an environment variable.

There are two ways around this. The first is to:

1. change the environment variable to be the password you want
2. terminate the container (or just take down the entire stack)
3. erase the persistent storage area (`sudo rm -rf ~/IOTstack/volumes/postgres`)
4. bring up the container.

The other way is to use the mechanism provided by the process running inside the container. I don't know what it is for Postgres but let's imagine a command like

```
$ postgres -resetAdminPassword "myNewPassword"
```

would get the job done. You just stick that on the end of an instruction to docker to run that inside the container, as in:

```
$ docker exec postgres postgres -resetAdminPassword "myNewPassword"
```

The first "postgres" is the name of the container so that isn't a mistake. If resetting the password had to be done interactively, you'd use something like

```
$ docker exec -it postgres postgres -resetAdminPassword
```

or, sometimes, there's the notion of opening a shell into the container:

```
$ docker exec -it postgres bash
# pgadmin -resetAdminPassword
# exit
$
```

> and, again, I see you worked that out for yourself. I'll shut up now. 🤦🤦😬

the1gofer, 2022-03-18:
Thanks, for the detailed response.   I'm not sure I would say I am a python person, I know enough to hack some stuff together, more of a hobbiest, than anything else.   Might be a good learning opportunity.  I'll see what I can figure out.

Paraphraser, 2022-06-19:
Current. Doco still lagging - needs to explain old menu uses environment file, new uses inline vars.

PR #580 (2022-09-11): postgres: remove broken menu options and fix docs
Add compose variables and indentation as explained in #505

Fixes #383, Fixes #517, Fixes #464
commit f0d53935 (2022-09-11): postgres: remove broken menu options and fix docs

Add compose variables and indentation as defined by #505

Fixes #383, Fixes #517
closed 2022-09-11 by Slyke