SensorsIot/IOTstack #538: zigbee2mqtt exited with code 1 - no log output

janphillipbro, 2022-03-31:
Hi, 
I have the following problem. After rebooting my PI this weekend, the z2m container is in restart loop. 
The docker-compose log is only showing zigbee2mqtt exited with code 1. The Docker log is completely empty.
I already tried pulling the newest image. 
Maybe someone here has an Idea what the problem could be or how to get a more informative log.

Paraphraser, 2022-03-31:
Which version of Raspbian? Bullseye or Buster? ( `cat /etc/os-release`)

If Buster, see [recommended patch)(https://sensorsiot.github.io/IOTstack/Basic_setup/#patch-2-update-libseccomp2) because a restart loop would be a symptom of an up-to-date Zigbee2mqtt container on Buster without that patch. See also #401

janphillipbro, 2022-03-31:
I'm using Bullseye. So this is sadly not the solution. But thanks for helping! 

Off topic: I also set up the PI with your PiBuilder. Loved this tool.

Paraphraser, 2022-03-31:
The next thing to try is to erase the persistent storage area and see if that makes a difference. Something like

```
$ cd ~/IOTstack
$ docker-compose rm --force --stop -v zigbee2mqtt
$ sudo mv ./volumes/zigbee2mqtt ./volumes/zigbee2mqtt.old
$ docker-compose up -d zigbee2mqtt
```

If that doesn't help then maybe also force a rebuild by replacing the third line with

```
$ docker-compose build --no-cache --pull zigbee2mqtt
```

I'all have a bit of a fiddle with a test system to make sure an up-to-date IOTstack + zigbee still works.

Paraphraser, 2022-03-31:
Oh, while I think of it, what does `docker-compose version` say? If not v2.3.4 then maybe run

```
$ sudo /boot/scripts/helpers/upgrade_docker-compose.sh v2.3.4
```

That's assuming PiBuilder is reasonably recent. v.2.3.3 had some oddities after pulls or rebuilds where "up" then didn't work as expected. I think v2.3.4 is a bit more robust.

janphillipbro, 2022-03-31:
> The next thing to try is to erase the persistent storage area and see if that makes a difference. Something like
>
> ```
> $ cd ~/IOTstack
> $ docker-compose rm --force --stop -v zigbee2mqtt
> $ sudo mv [...]

This worked. I will now analyze which file seemed to be faulty. 
Hopefully I don't need to retrain all devices now :)

Paraphraser, 2022-03-31:
Yup - see [doco](https://www.zigbee2mqtt.io/guide/configuration/logging.html)

Paraphraser, 2022-03-31:
Changing the configuration file to set logging to console still doesn't seem to show up in `docker logs zigbee2mqtt` but it does have the effect of stopping logging to disk. I have no idea where the logs are going so I can't tell whether this is an error or not. One thing that does enable docker logs is adding the environment variable:

```
      - DEBUG=zigbee-herdsman*
```

to the service definition. After an "up" docker logs becomes **incredibly** chatty so it's probably not all that useful unless you're chasing something specific.

Anyway, to sum up, "works for me". Are you still having trouble?

Paraphraser, 2022-03-31:
Still haven't figured out the docker logs issue but I did stumble across [this](https://www.zigbee2mqtt.io/guide/configuration/#environment-variables).

The only reason IOTstack uses a Dockerfile for Zigbee2MQTT is to edit the default configuration on the fly to point to the "mosquitto" container (rather than localhost) and to turn on the web front end. If I'm reading it right, it looks like we should be able to do that with:

```
- ZIGBEE2MQTT_CONFIG_MQTT_SERVER="mqtt://mosquitto:1883"
- ZIGBEE2MQTT_CONFIG_FRONTEND_PORT="8080"
#- ZIGBEE2MQTT_CONFIG_FRONTEND_AUTH_TOKEN="PASSWORD"
```

I haven't tested it yet but if that works we can go back to using the image just as it comes from Dockerhub.

janphillipbro, 2022-03-31:
> Changing the configuration file to set logging to console still doesn't seem to show up in `docker logs zigbee2mqtt` but it does have the effect of stopping logging to disk. I have no idea where the [...]

Deleting the storage helped. After copying the database back, I only needed to restore my friendly names, and everything works perfectly fine.

closed 2022-03-31 by janphillipbro

Paraphraser, 2022-03-31:
The environment variables above do work - providing I remove the quote marks from the right hand side. Quite neat. I'll work on a pull request.

Paraphraser, 2022-04-04:
Further to the above, the container writes its logs to

```
~/IOTstack/volumes/zigbee2mqtt/data/log/yyyy-mm-dd.hh-mm-ss/log.txt
```

where `yyyy-mm-dd.hh-mm-ss` is the date and time the container was last started. This means that you have to identify the folder with the latest timestamp before you can inspect the log contained within it.

However, by defining

```
 - ZIGBEE2MQTT_CONFIG_ADVANCED_LOG_SYMLINK_CURRENT=true
```

then the path to the *current* log will be:

```
~/IOTstack/volumes/zigbee2mqtt/data/log/current/log.txt
```

This will also be part of the pull request I'm working on.