SensorsIot/IOTstack #542: Raspbian / RaspberryPiOS no longer allowing user Pi

NossieUK, 2022-04-09:
I know that the script relies on Pi existing as a user ....

Just wondering if any moves have been made to change this to allow other users with the news that user Pi will be changed going forward?

Paraphraser, 2022-04-09:
I am *reasonably* sure that there are no *actual* dependencies on `/home/pi` (rather than `$HOME`) or things like `chown pi:pi` (rather than `chown $USER:$USER`) left. There were a few in gcgarner days but I think they've all been nailed as various eyeballs have passed over them.

There are quite a few "bad examples" ("bad" since 2020-04-04) in the documentation which should probably be turned into "better" examples. But those don't affect anything.

> The assumption of `$HOME/IOTstack` is a bit more deep seated

Quite a few templates have things like `PUID=1000` which implicitly assume that 1000=$USER. Nothing will break if 1000 is not known to `/etc/passwd` or if it's known but doesn't map to the ID of logged-in user. Those are usually about making sure that sub-folders of `~/IOTstack/volumes/<container>/` can be read/written without needing `sudo`. It's possible that something should be done about those but I don't think they break anything.

With IOTstack, the "user pi" at "/home/pi" with "ID 1000" has always been more about testing. Those being the defaults (until now) they got a lot more testing than any "roll your own". Stating those assumptions explicitly has always been directed more towards getting each user to at least ask themselves the question, "if I've rolled my own environment, is it *possible* that that explains my current problem?"

I might turn out to be wrong but we'll only really know if/when issues are opened pointing to specific incompatibilities. Fingers crossed...

Paraphraser, 2022-04-10:
The answer to the ID=1000 question is that if you select a different username (eg "me" instead of "pi") in Raspberry Pi Imager, that user (ie "me") gets ID=1000. So that means all the `PUID=1000` continue to have the same semantics.

simonmcnair, 2022-04-27:
ideally, for me, there wouldn't be any hardcoding.  It's fine to assume that the folder iotstack should exist in the $home folder but there shouldn't be any presumption about the user being 1000, or that you're running on a PI.  It should be detected or enumerated.  I would go so far as to say that, in the case of bluetooth for instance, it should be enumerable and an env flag set for the remainder of the session.

Yes, I know this makes it more work, and more difficult, it's just my opinion.

Paraphraser, 2022-04-27:
Well, any of us saying "shouldn't" is true. But there are just so many **containers** that assume ID=1000 that it becomes a practical question of how to adapt them.

Every individual user is entirely free to declare, "I don't want to use 1000, I want to use X". But it is then up to that person to follow through and make sure that none of the containers they want to run depends on 1000 and, if it does, fix it.

simonmcnair, 2022-04-27:
Possibly the other answer is to create a user and/or group called IOTStack and assign it all to that user/group.  Yes this abstracts everything away from rasppi but the more abstraction there is, the less you're dependant on third party changes ?

Paraphraser, 2022-04-28:
It's probably time for a somewhat longer response to your question.

First, a bit of conceptual background to set the scene.

The web is full of information about processes that have been "containerised" along with examples of how to use those containers in a docker environment. I'm sure you've seen them. In general, they break down into:

1. `docker run ...`
2. sample service definitions to use in a `docker-compose.yml`
3. both of the above.

If you took a random sample of group #2 off the web, you'd find a whole bunch of different approaches. In particular, examples differ in:

1. how persistent storage is handled; and
2. which ports are mapped.

If you concatenated a random sample of group #2 service definitions into a `docker-compose.yml` and told `docker-compose` to bring up that collection of containers (a "stack"), you'd probably wind up with persistent storage scattered all over your Pi's file system, a reasonable chance of port conflicts as containers contended for the same external port numbers, plus a non-zero probability that a few containers simply would not start because they make assumptions that don't hold.
 
IOTstack is not a "system", as such. It's a set of conventions which are designed to give you a better-than-even chance that a group of containers you select from the menu will work out-of-the-box. In particular, IOTstack's conventions:

1. place every container's persistent storage in a sub-folder of `~/IOTstack/volumes/`;
2. minimise the likelihood of port conflicts by not allocating the same external port to two service definitions unless it is unavoidable;

	> An example of an unavoidable port conflict is PiHole and AdGuardHome. Both are going to want external port 53. It makes no sense to run two ad-blockers. The IOTstack menu won't prevent you from selecting both containers but `docker-compose` will grizzle when you try to bring up the stack.

3. create, to the maximum extent possible, the conditions where a container's assumptions hold on first launch.

	> Earlier incarnations of IOTstack used scripts named `directoryfix.sh` to try to preset assumptions. Those scripts were only ever run by the menu and only then under certain conditions. Those scripts have been replaced, progressively, with Dockerfiles which give containers that might misbehave on first lauch the "smarts" to self-repair.
	
A key thing to keep in mind is that, we (IOTstack) still have to work within the rules and constraints imposed by docker and docker-compose.

After a clean clone from GitHub, there is no `~IOTstack/volumes` folder nor any sub-folders. The first time you bring up a stack, docker-compose creates `~IOTstack/volumes` plus the left-hand-side of each volume map. Using the WireGuard service-definition as an example:

```
volumes:
- ./volumes/wireguard:/config
```

docker-compose will do the equivalent of:

```
$ sudo mkdir -p ~/IOTstack/volumes/wireguard
```

On first launch, `/home`, `/home/pi` and `/home/pi/IOTstack` will exist (so they won't be touched) but both `volumes` and `wireguard` will be created and owned by root.

When the WireGuard container starts, it does the equivalent (from **inside** the container) of:

```
# chown -R $PUID:$PGID /config
```

The container is running as root so these changes propagate to the external file system. I assume the logic is to minimise the need for the user to use `sudo` when manipulating files in the container's persistent storage area.

> Some containers also downgrade their privileges. Two examples that I can think of are `mosquitto` and `pihole` but those use user IDs of 1883 and 999, respectively, not 1000.

The obvious next question is, where does the WireGuard container get PUID and PGID? Answer, via environment variables contained in its service definition:

```
$ cat ~/IOTstack/.templates/wireguard/service.yml
wireguard:
  container_name: wireguard
  image: ghcr.io/linuxserver/wireguard
  restart: unless-stopped
  environment:
  - PUID=1000
  - PGID=1000
  - TZ=Etc/UTC
  - SERVERURL=your.dynamic.dns.name
  - SERVERPORT=51820
  - PEERS=laptop,phone,tablet
[... 130 lines cut ...]
```
- PUID=${MYUID:-1000}
- PGID=${MYUID:-1000}
```

Some comments:

1. Changing someone's `.profile` or `.bashrc` is beyond the scope of IOTstack. We could *recommend* it but we can't *enforce* it.
2. A scheme where 1000 was the default would obviously work on systems that didn't define `MYUID` where `UID` was 1000; however, you'd have the same problem as now if another user ID was in effect. In other words, it lacks a certain something as a bullet-proof solution.
3. There are corner cases where `.profile` or `.bashrc` don't run (eg `cron`) so that lacks a certain something too. You could set `MyUID=1000` in `~/IOTstack/.env`. The menu could do this too but it would be a bit of a problem to track down if the ID were to change without the menu being re-run. Don't ask me why Docker decided to make that file invisible. It's a trap for the unwary. 
4. The output from the `find` command above is just the examples that are **exposed** in IOTstack templates. A lot of containers **support** environment variable keys for passing the user and group ID and default to 1000 **internally** if the key is omitted. For IOTstack to be able to claim "we are ID agnostic", we would need to figure out the answer for every container **and** inherit the maintenance problem every time a container's maintainer decided to change something, or whenever a new container was added to IOTstack. [... cut ...]
By the way, I like to explore questions of this kind in a reasonable amount of detail because, for every person who asks something like this, there are bound to be plenty of others who are wondering about it. I reckon it's better to try to cover-off as much of the how, what, when, where and why as possible.

Plus, everything I write is implicitly tied to my own limited knowledge and questionable assumptions. I'm no expert. You, or someone else following this thread may well spot the flies in my logical ointment, find a solution, and we all benefit. Go for it!

simonmcnair, 2022-04-28:
Thank you for generously taking the time to explain the logic.  I guess sometimes things are just the way they are and it would take a larger amount of effort to try and change it, than to go along with the current.  Thanks again for your time.  I appreciate it.

simonmcnair, 2022-04-28:
I've been doing some reading, and investigation, and nothing I'm going to state here detracts from your excellent synopsis above.

Apparently a good way of getting the current userid is `export TEST=$(id -u)`

in a dockerfile (which is not appropriate for most of this) it can be done, I believe, as follows:

RUN export currentuserid=$(id -u);echo $currentuserid;

I think another fair perspective is that a user id is arbitrary (which is harder programmatically) unless it's standard i.e. root or pi (which is worse from a security perspective).

My view would be to create an IOTuser and an IOTgroup and set everything to that (no different to how most daemons act) .  Which would also mean that uninstallation would be easier, and the locations of files could be prescribed definitively and from a security perspective minimum rights, and smaller attack surface.

I would also be inclined to include a random password generator, generate passwords for each service and drop them, and the full urls into a reference file which could then be used to import in to a password manager and/or book mark manager.

At this point the conversation has descended in to conjecture, design and opinion, for which I apologise.

Thanks for tolerating my blathering. 
:-)

ukkopahis, 2022-04-28:
Any problems with a solution to:
* change all templates like this
    ```yaml
    - PUID=${IOTSTACK_UID:-1000}
    - PGID=${IOTSTACK_GID:-1000}
     ```
* when menu is executed: add the the definitions to `~/IOTstack/.env`:
    ```bash
    IOTSTACK_UID=$(id --user)
    IOTSTACK_GID=$(id --group)
    ```

I can implement, update docs and make the PR.

simonmcnair, 2022-04-28:
I, personally, would prompt/test to create an IOTStack user and group with a random pwd (logged) but I'm sure you both know better than me (newb).  Perhaps give the option to run as current user if an additional user is not desired.
I would not assume anything, and would always define to prevent misinterpretation and increase security.  I would never use 1000 for anything, ever, as it would assume that the first user created exists and/or that the current user wants to inherit ownership.
It's easy for me to get on my high horse and not do the work though, so apologies for the presumption.

Thanks for the effort you've all put in, I appreciate it.

ukkopahis, 2022-04-28:
> prompt/test to create an IOTStack user and group with a random pwd (logged)

The new Raspberry OS installer already prompts user/pw.

> Perhaps give the option to run as current user if an additional user is not desired.

This user used in containers is just for convenience, so that your main login can access files it creates without using sudo. Nothing to do with security. Regardless of what you choose, the service will work.

SimonMcN, 2022-04-28:
That will change if you ever implement a samba service or something similar ?  Either way, if you say it's fine, it's fine :-)

Paraphraser, 2022-04-28:
@ukkopahis the only "problem" is the containers that *do* support some form of ID environment variable but where we (IOTstack) haven't yet captured it in our templates. Those will still default to 1000.

I've only really noticed this in passing so don't quote me if I'm mis-remembering but I think it's the LinuxServer builds which do that.

I'm not sure this **matters** in the scheme of things. After all, we get the occasional "WTF?" question about 1883 for Mosquitto which does no real harm so, if the current ID is not 1000 and there are containers assuming it is, the worst that is going to happen is the need to use sudo when mucking about in volumes. Like *that's* an exception to the rule...

On balance, I'd say "go for it". It will capture a reasonable number of cases out-of-the-box and be a standard we can work towards.

ukkopahis, 2022-04-28:
> Regardless of what you choose, the service will work.

This got me thinking. In actuality this statement should hedge "as long as the chosen user-id matches the owner of the existing files in volumes". Its rare (and a bit risky) for a container's entryscript to do the required 'chown' operations for a changed user-id. It's therefore user's responsibility to match the container `PUID` and the file owners of `volumes/<service>/*`.

As such, when we remove the "User has the user ID 1000" assumption from 'Getting started'-page, we can't change the `PUID` of existing containers even if they are 'wrongly' defaulting to PUID=1000. Or if we do such a change later, we'll have to add an entryscript or a post-build migration script to menu.sh that fixes volumes/service/* file ownerships. 

As example: Assumptions: a) an existing service defaults to PUID=1000. b) the logged-in-user `rob` has ID 999:
1. Follows: files in /volumes/service/* have an owner-id of 1000, and aren't easily accessible by `rob`. 
2. Also follows: changing docker service to PUID=999 would make the service have problems accessing/changing /volumes/service/*-files that still have owner-id=1000.

Therefore the pull-request I'm working on, and all future added services, *must* in their service.yml set the user-id ("PUID") correctly from the get-go (if the container supports one). Changing it later is a migration headache not worth it.

Paraphraser, 2022-04-28:
A first time user deploying IOTstack on a Pi where ID is not 1000 is probably not too much of a concern because it's probably safe to assume "running the menu" so fixes can be applied.

I've been wondering more about the migrating user. Running IOTstack now with ID 1000. Takes a backup, rebuilds a Pi with a different ID, then restores the backup. Not sure what the IOTstack restore scripts do but IOTstackBackup passes the "same owner" flag to tar so **everything** will come back with the wrong ID.

I'm leaning towards dealing with this entire scenario via caveat rather than by code. A sentiment along the lines of

> those dumb enough to try it get to solve their own fang-sucking problems

if you get my drift...

ukkopahis, 2022-04-28:
When considering the troubles of accidentally using the wrong UID, I think it's better to not have a default, but use [mandatory variables](https://docs.docker.com/compose/compose-file/compose-file-v3/#variable-substitution) instead:
```yaml
- PUID=${IOTSTACK_UID:?IOTSTACK_UID must be defined in .env}}
- PGID=${IOTSTACK_GID:?IOTSTACK_GID must be defined in .env}}
```

Then, if by some magic, the end-user manages to delete the .env-file, instead of risking a jolly file-ownership-mess we get a clean error:

``` console
$ docker-compose up -d  --force-recreate pihole
ERROR: Missing mandatory value for "environment" option interpolating ['WEBPASSWORD=%randomAdminPassword%', 'INTERFACE=eth0', 'IOTSTACK_UID=${IOTSTACK_UID:?IOTSTACK_UID must be defined in .env}'] in service "pihole": IOTSTACK_UID must be defined in .env
```
which hopefully alerts the user to the .env-file missing due to problem in his backup, restore, cloning, etc...

ukkopahis, 2022-04-28:
> rebuilds a Pi with a different ID

Self inflicted, unforced problem. Redo with better judgment. Practice makes perfect ;)

I'll add instruction about this to docs.

simonmcnair, 2022-04-29:
Newb question if that's okay.  Why not, as part of the script, chown current $uid:$groupid ~/IOTstack/Volumes if the folder exists ?  It won't apply to any externally mounted data but it would be assumed that IOTStack 'owns' the files in that directory ?

This is just me learning, this has been a great help to my understanding so far

simonmcnair, 2022-04-29:
> Newb question if that's okay. Why not, as part of the script, chown current $uid:$groupid ~/IOTstack/Volumes if the folder exists ? It won't apply to any externally mounted data but it would be [...]

Oh, I guess where the docker image doesn't have an option to specify user or group in the dockercompose this will default to 1000 and the chown will possibly break that container.  Got it.

Paraphraser, 2022-04-29:
It's not necessarily safe to chown everything in volumes to the current user. Some containers will barf if you change their ownership on them.

Next point. There is no way (yet) of hooking a user script to docker-compose so you either have to do it in the menu or supply a script that the user just has to know to run.  Not everybody uses the menu (I don't) and the `directoryfix.sh` I mentioned before are an example of something the user just has to know to run. That basically didn't work - we were forever telling people "just run ...". The Dockerfile approach is the solution.

The chown that Dockerfiles do are limited to the container's persistent storage. That's the whole idea of containers - they are contained. But, again, the uid/gid that an internal script uses in a chown can't come from the container. The only real way of getting it the info is what we've been talking about here.

ukkopahis, 2022-04-29:
Though I'm torn on if it'd be better to add to docker-compose.yml:
```yml
env_file: docker-compose.iotstack.env
```
And have all defaults added into the explicitly declared file. I don't like how `.env` is just implicitly found and a hidden file. And 'badly' named, with no clear indication it's for Docker.

Paraphraser, 2022-04-30:
> Did you have a reason to using that flag?

Well, yes. So that the contents of `~/IOTstack/volumes` come back "as is". The general backup (everything non-database) includes the compose file and services so, if the ID was 1000 on the backup, it'll be 1000 on the restore just as a by-product of what is needed for the contents of `~/IOTstack/volumes` to be restored faithfully. But, now that I think about it, it's really no big deal to explicitly do a chown on the compose file and services.

Paraphraser, 2022-04-30:
I wholeheartedly agree with your comments on `.env`. Serious "what on earth was I thinking at the time?" going on there.

Although it isn't hidden, I also dislike `env.yml` in the `.templates` folder. It contains the networking definitions so it should have a name reflecting its purpose.

Personally, I have my compose file structured like this:

```
version: '3.6'

networks:

  default:
    driver: bridge
    ipam:
      driver: default

  nextcloud:
    driver: bridge
    internal: true
    ipam:
      driver: default

services:

  portainer-ce:

…
```

(I do it that way so I can easily `cat` service definitions I want to test onto the end of my compose file.)

Anyway, that `version` line at the top is hard-coded into the menu (old and new). It might be better if:

1. `env.yml` was renamed to something like `docker-compose-header.yml` with content:

	```
	version: '3.6'
	
	networks:
	
	  default:
	    driver: bridge
	    ipam:
	      driver: default
	
	  nextcloud:
	    driver: bridge
	    internal: true
	    ipam:
	      driver: default
	```

2. Both old and new menus were changed to copy that "as is" to the start of the compose file.
3. All knowledge of a compose file version number was removed from both menus.

A wrinkle on the theme might be to first:

```
$ cp -n ~/IOTstack/.templates/docker-compose-header.yml ~/IOTstack/services/.
```

and then prepend `~/IOTstack/services/docker-compose-header.yml` to the compose file. That way, anyone who wants to make permanent changes to either the version number or networking structure has a straightforward mechanism which won't collide with pulls.

In theory, the `version` directive is [deprecated](https://github.com/compose-spec/compose-spec/blob/master/spec.md#version-top-level-element) but I've noticed people complaining that things break if it's omitted (I think it was something to do with cgroup rules).

Paraphraser, 2022-04-30:
> Though I'm torn on if it'd be better to add to docker-compose.yml

So, that would be a candidate for this "header" file too, right?

ukkopahis, 2022-04-30:
> > Did you have a reason to using that (same owner) flag?
> 
> Well, yes. So that the contents of `~/IOTstack/volumes` come back "as is". The general backup (everything non-database) includes the compose file and services so, if the ID was 1000 on the backup, it'll be 1000 on the restore just as a by-product of what is needed for the contents of `~/IOTstack/volumes` to be restored faithfully. But, now that I think about it, it's really no big deal to explicitly do a chown on the compose file and services.

Not telling you how to handle this, but if restoring a backup changes file owner from the original 1883 to 1000, some breakage might occur.

From man tar:
```
       -p, --preserve-permissions, --same-permissions
              extract information about file permissions (default for superuser)

       --no-same-owner
              Extract files as yourself (default for ordinary users).
```

> I also dislike `env.yml` in the `.templates` folder.

Yup, but it might contain other stuff than only network at some point in the future. Could be renamed e.g. 'base-docker-compose.yml'. It's the starting point where all service.yml's are merged into. But I don't at the moment think renaming it is worth the effort. 

> Personally, I have my compose file structured 'networks:' before 'services:'

That's a good idea, might pick it up for #505

ukkopahis, 2022-05-01:
> ```yaml
> env_file: docker-compose.iotstack.env
> ```

Nice idea, but doesn't work. Docker [docs](https://docs.docker.com/compose/compose-file/compose-file-v3/#variable-substitution) state:
> You can set default values for environment variables using a [.env file](https://docs.docker.com/compose/env-file/), which Compose automatically looks for in project directory (parent folder of your Compose file). Values set in the shell environment override those set in the .env file.

And by my tests defining an `env_file` doesn't add variables that can be used for variable replacement in the `environment:`-section, only the default `.env` does. Docker logic :(