SensorsIot/IOTstack #546: Grfana updated and now broken

confuserated, 2022-04-16:
Hi, Not super-techie here but I just updated all of the containers I was using on my Home Assistant Server (the usual suspects) and Grafana has now stopped working, it endlessly crashes and restarts the docker container, no configs have changed at all since last update and everything else is working fine. 

Fairly sure I can't be the only one experiencing this problem on latest version of Grafana, all other systems are latest version and RPi OS is Bullseye latest.

Logs as follows:

pi@RPi4:~/IOTstack $ sudo journalctl --since 01:20 --until "now"
-- Journal begins at Fri 2022-01-28 01:30:20 GMT, ends at Sat 2022-04-16 01:20:49 BST. --
Apr 16 01:20:04 RPi4 systemd[31201]: run-docker-runtime\x2drunc-moby-e5740e62c66def928f7e1b9cf2a76bf7d5d4c38742c781124ec60c960e561e19-runc.YZFTNu.mount: Succeeded.
Apr 16 01:20:04 RPi4 systemd[713]: run-docker-runtime\x2drunc-moby-e5740e62c66def928f7e1b9cf2a76bf7d5d4c38742c781124ec60c960e561e19-runc.YZFTNu.mount: Succeeded.
Apr 16 01:20:04 RPi4 systemd[1]: run-docker-runtime\x2drunc-moby-e5740e62c66def928f7e1b9cf2a76bf7d5d4c38742c781124ec60c960e561e19-runc.YZFTNu.mount: Succeeded.
Apr 16 01:20:07 RPi4 systemd[713]: run-docker-runtime\x2drunc-moby-9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb3a7df-runc.ArH9rf.mount: Succeeded.
Apr 16 01:20:07 RPi4 systemd[1]: run-docker-runtime\x2drunc-moby-9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb3a7df-runc.ArH9rf.mount: Succeeded.
Apr 16 01:20:07 RPi4 systemd[31201]: run-docker-runtime\x2drunc-moby-9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb3a7df-runc.ArH9rf.mount: Succeeded.
Apr 16 01:20:08 RPi4 systemd[1]: docker-fa1f59e7c9f0316b08683ee45259ea61f52f66348176c639f0f5a65826c89554.scope: Succeeded.
Apr 16 01:20:08 RPi4 systemd[1]: docker-fa1f59e7c9f0316b08683ee45259ea61f52f66348176c639f0f5a65826c89554.scope: Consumed 4.200s CPU time.
Apr 16 01:20:08 RPi4 systemd[1]: docker-9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb3a7df.scope: Succeeded.
Apr 16 01:20:08 RPi4 systemd[1]: docker-9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb3a7df.scope: Consumed 3.844s CPU time.
Apr 16 01:20:08 RPi4 dockerd[621]: time="2022-04-16T01:20:08.751651920+01:00" level=info msg="ignoring event" container=fa1f59e7c9f0316b08683ee45259ea61f52f66348176c639f0f5a65826>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.752055214+01:00" level=info msg="shim disconnected" id=fa1f59e7c9f0316b08683ee45259ea61f52f66348176c639f0f5a65826c>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.752245305+01:00" level=warning msg="cleaning up after shim disconnected" id=fa1f59e7c9f0316b08683ee45259ea61f52f66>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.752371045+01:00" level=info msg="cleaning up dead shim"
Apr 16 01:20:08 RPi4 systemd[1]: docker-bb0b8fe6e56513849212c327ad2dc98bcc963728102ba981a2f8c5e00bec8afa.scope: Succeeded.
Apr 16 01:20:08 RPi4 systemd[1]: docker-bb0b8fe6e56513849212c327ad2dc98bcc963728102ba981a2f8c5e00bec8afa.scope: Consumed 19.583s CPU time.
Apr 16 01:20:08 RPi4 dockerd[621]: time="2022-04-16T01:20:08.811927378+01:00" level=info msg="ignoring event" container=9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7b>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.813286573+01:00" level=info msg="shim disconnected" id=9c67352dd77c503907dedd2c06299fa5e7c6a4b8fe3de6e2529db4af7bb>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.814065901+01:00" level=warning msg="cleaning up after shim disconnected" id=9c67352dd77c503907dedd2c06299fa5e7c6a4>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.814666564+01:00" level=info msg="cleaning up dead shim"
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.844432295+01:00" level=warning msg="cleanup warnings time=\"2022-04-16T01:20:08+01:00\" level=info msg=\"starting >
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.869854185+01:00" level=warning msg="cleanup warnings time=\"2022-04-16T01:20:08+01:00\" level=info msg=\"starting >
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.871071195+01:00" level=info msg="shim disconnected" id=bb0b8fe6e56513849212c327ad2dc98bcc963728102ba981a2f8c5e00be>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.871656839+01:00" level=warning msg="cleaning up after shim disconnected" id=bb0b8fe6e56513849212c327ad2dc98bcc9637>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.872084225+01:00" level=info msg="cleaning up dead shim"
Apr 16 01:20:08 RPi4 dockerd[621]: time="2022-04-16T01:20:08.874088805+01:00" level=info msg="ignoring event" container=bb0b8fe6e56513849212c327ad2dc98bcc963728102ba981a2f8c5e00b>
Apr 16 01:20:08 RPi4 containerd[523]: time="2022-04-16T01:20:08.921875252+01:00" level=warning msg="cleanup warnings time=\"2022-04-16T01:20:08+01:00\" level=info msg=\"starting >
Apr 16 01:20:08 RPi4 dhcpcd[444]: veth42a41f5: carrier lost
Apr 16 01:20:08 RPi4 kernel: br-1f4b14d68fa1: port 3(veth42a41f5) entered disabled state
Apr 16 01:20:08 RPi4 kernel: vethbd295d2: renamed from eth0
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Withdrawing address record for 169.254.92.209 on veth42a41f5.
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Leaving mDNS multicast group on interface veth42a41f5.IPv4 with address 169.254.92.209.
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Interface veth42a41f5.IPv4 no longer relevant for mDNS.
Apr 16 01:20:09 RPi4 dhcpcd[444]: veth42a41f5: deleting route to 169.254.0.0/16
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Interface veth42a41f5.IPv6 no longer relevant for mDNS.
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Leaving mDNS multicast group on interface veth42a41f5.IPv6 with address fe80::940f:16ff:fe03:55f.
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 3(veth42a41f5) entered disabled state
Apr 16 01:20:09 RPi4 kernel: device veth42a41f5 left promiscuous mode
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 3(veth42a41f5) entered disabled state
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Withdrawing address record for fe80::940f:16ff:fe03:55f on veth42a41f5.
Apr 16 01:20:09 RPi4 dhcpcd[444]: veth42a41f5: removing interface
Apr 16 01:20:09 RPi4 systemd-udevd[2490]: vethbd295d2: Failed to query device driver: No such device
Apr 16 01:20:09 RPi4 systemd-udevd[2490]: vethbd295d2: Failed to get link config: No such device
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 4(veth51dcc11) entered disabled state
Apr 16 01:20:09 RPi4 kernel: veth54daba9: renamed from eth0
Apr 16 01:20:09 RPi4 systemd[713]: run-docker-netns-5d96685fdf78.mount: Succeeded.
Apr 16 01:20:09 RPi4 systemd[31201]: run-docker-netns-5d96685fdf78.mount: Succeeded.
Apr 16 01:20:09 RPi4 systemd[1]: run-docker-netns-5d96685fdf78.mount: Succeeded.
Apr 16 01:20:09 RPi4 systemd[713]: var-lib-docker-overlay2-bda6ed5a1648374781c9fcf78511603faa23c8459f15cb7475363791c1ddafac-merged.mount: Succeeded.
Apr 16 01:20:09 RPi4 systemd[31201]: var-lib-docker-overlay2-bda6ed5a1648374781c9fcf78511603faa23c8459f15cb7475363791c1ddafac-merged.mount: Succeeded.
Apr 16 01:20:09 RPi4 systemd[1]: var-lib-docker-overlay2-bda6ed5a1648374781c9fcf78511603faa23c8459f15cb7475363791c1ddafac-merged.mount: Succeeded.
Apr 16 01:20:09 RPi4 dhcpcd[444]: veth51dcc11: carrier lost
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 6(veth8e39384) entered disabled state
Apr 16 01:20:09 RPi4 kernel: veth8279769: renamed from eth0
Apr 16 01:20:09 RPi4 dhcpcd[444]: veth51dcc11: deleting address fe80::ee3e:2071:5b5c:e03b
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Withdrawing address record for fe80::ee3e:2071:5b5c:e03b on veth51dcc11.
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Leaving mDNS multicast group on interface veth51dcc11.IPv6 with address fe80::ee3e:2071:5b5c:e03b.
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 4(veth51dcc11) entered disabled state
Apr 16 01:20:09 RPi4 kernel: device veth51dcc11 left promiscuous mode
Apr 16 01:20:09 RPi4 kernel: br-1f4b14d68fa1: port 4(veth51dcc11) entered disabled state
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Interface veth51dcc11.IPv6 no longer relevant for mDNS.
Apr 16 01:20:09 RPi4 avahi-daemon[395]: Interface veth51dcc11.IPv4 no longer relevant for mDNS.
[... 473 lines cut ...]
Apr 16 01:20:31 RPi4 dhcpcd[444]: dhcp_readbpf: vethdfbc788: Network is down
Apr 16 01:20:31 RPi4 systemd-udevd[2853]: vethe1dc48c: Failed to query device driver: No such device
[... cut ...]
Apr 16 01:20:31 RPi4 kernel: br-9b221860107e: port 3(veth938b3f8) entered disabled state
Apr 16 01:20:31 RPi4 dhcpcd[444]: dhcp_readbpf: veth1bacf9e: Network is down
Apr 16 01:20:31 RPi4 dhcpcd[444]: veth5fcf0f4: removing interface
Apr 16 01:20:31 RPi4 dhcpcd[444]: veth5fcf0f4: deleting address fe80::71ab:a7e3:90d2:5db1
Apr 16 01:20:31 RPi4 dhcpcd[444]: veth1bacf9e: removing interface
Apr 16 01:20:31 RPi4 dhcpcd[444]: veth1bacf9e: deleting address fe80::3d8b:cf6a:5811:b67c
Apr 16 01:20:31 RPi4 dhcpcd[444]: veth5e22b4a: waiting for carrier

confuserated, 2022-04-16:
Also:

docker logs grafana
standard_init_linux.go:228: exec user process caused: exec format error

confuserated, 2022-04-16:
does this mean the wrong version of the code is being pulled?

Paraphraser, 2022-04-16:
I hate to say this but I have developed some really deep misgivings about the quality and consistency of Grafana's test regime.

You are correct when you surmise that the wrong version of the code is being pulled. If you take a look on [DockerHub](https://hub.docker.com/r/grafana/grafana/tags) you'll see that "latest" was only built for "amd64" so that's what's being pulled and, of course, won't run.

This is not the first time this has happened. Personally, I gave up about two months ago and pinned to Grafana 8.3.4 but I did that to solve a secondary problem where later versions of Grafana didn't "play nice" with Safari on my iDevices. On Discord, people have recommended pinning to "main". If you look at the same DockerHub page, you'll see "main" a few rows down being build for "amd64", "armv7" and "arm64".

If the concept of "pinning" doesn't make sense, in IOTstack the starting point for Grafana is:

```
grafana:
  container_name: grafana
  image: grafana/grafana
  restart: unless-stopped
  user: "0"
  ports:
    - "3000:3000"
  environment:
    - TZ=Etc/UTC
    - GF_PATHS_DATA=/var/lib/grafana
    - GF_PATHS_LOGS=/var/log/grafana
  volumes:
    - ./volumes/grafana/data:/var/lib/grafana
    - ./volumes/grafana/log:/var/log/grafana
```

That service definition gets copied into your `~/IOTstack/docker-compose.yml` where, depending on what you want Grafana to do for you, you often wind up adding more environment variables, so yours might not quite look like that.

The critical element is:

```
  image: grafana/grafana
```

The absence of a tag implies "latest", as in:

```
  image: grafana/grafana:latest
```

You can try **either** of the following:

* What Discord users are saying:

	```
	  image: grafana/grafana:main
	```

* or what I'm using for my own needs:

	```
	  image: grafana/grafana:8.3.4
	```

If you make the change and then:

```
$ cd ~/IOTstack
$ docker-compose up -d grafana
```

it should improve the situation.

If you want to chase-up this "bug" yourself, the place to do it is [grafana/grafana](https://github.com/grafana/grafana/issues). Currently reporting over 2,000 open issues…

You could also just add a thumbs-up in [Issue 47676](https://github.com/grafana/grafana/issues/47676).

Hope this helps.

confuserated, 2022-04-16:
Thanks very much, that's super helpful and I'm sure that's my problem. I'll have to test that once I had some sleep 🫥

Paraphraser, 2022-04-16:
Having explained about Grafana and (hopefully) gotten you past that problem, there are some other things that struck me in your initial post - which can definitely wait until you've had some sleep.

You said "Home Assistant Server". Can I ask how you built your system? Did you:

1. Start with [Home Assistant's pre-built image](https://www.home-assistant.io/installation/raspberrypi) and add IOTstack to it, or
2. Start with Raspberry Pi OS Bullseye and follow [home-assistant/supervised-installer](https://github.com/home-assistant/supervised-installer), or
3. some other method?

Until very recently, I offered #2 in [PiBuilder](https://github.com/Paraphraser/PiBuilder) but I took it out 10 days ago for the reasons explained [here](https://github.com/Paraphraser/PiBuilder#about-supervised-home-assistant).

That decision began with [PiBuilder issue 4](https://github.com/Paraphraser/PiBuilder/issues/4).

I'm not at all convinced that HA is ever again going to play nice with IOTstack on the same machine. For me, their decision to start claiming port 53 was the killer. That will wrong-foot anyone who wants to run another DNS service like PiHole or AdGuardHome.

But, if you've got another method that solves all these problems and plays nice with IOTstack then I'd really like to hear about it because there's a reasonable demand to be able to run HA and IOTstack on the same Pi.

<hr>

When I look through your `journalctl` output, I see a lot of chatter which makes me think you might not have the first and third [recommended system patches](https://sensorsiot.github.io/IOTstack/Basic_setup/#required-system-patches) in place.

I also see a lot of IPv6. Do you use and rely on IPv6? If not, I've found that my Pis are far more stable and generate far less network chatter if I just turn it off.

> I wore the job title of "network consultant" some years ago so it actually pains me to say that. I always thought the day would come when IPv6 would be the one protocol to rule them all. Decades later and we're still not there.

Paraphraser, 2022-04-16:
Oh, techie OCD - I recognise that one!

confuserated, 2022-04-16:
🤪 you've given me lots to play with there, really must sleep this morning tho'!

Paraphraser, 2022-04-16:
Ah! So that's "Home Assistant Container" (which you can get by selecting Home Assistant in the IOTstack menu, and is fully compatible with IOTstack), not "Supervised Home Assistant". Gotcha.

confuserated, 2022-04-16:
Yep, thanks again for your help, I implemented your suggestions as well and all is good with there world again 🤩, I think I should probably close this issue now, is that the correct etiquette?

closed 2022-04-16 by confuserated

confuserated, 2022-04-16:
I should just add I think, that I get around some of the things supervisor gives me by having ESPHome installed in the HA volume (also as a Docker container) which means I can use it as a file editor for the configuration.yaml as well the ESPHome configs, a little odd I know but it works.