SensorsIot/IOTstack #550: Telegraf depends on mosquitto ?

simonmcnair, 2022-04-27:
I've used telegraf a bit in the past, and I know it has plugins for everything.  but as far as I am aware, none of them are required.
Yes it is required to have an upstream db, but apart from that you can configure anything.  It's a monitoring app, it will monitor what you tell it to, and report it upstream.

For me this is really a question of why is telegraf dependant on Mosquitto ?  It may just be used to monitor docker ?

Paraphraser, 2022-04-27:
Short answer: "no idea".

Longer answer: "history":

1. The IOTstack project began with [Graham Garner](https://github.com/gcgarner).
2. If you look through Graham's original [Telegraf configuration](https://github.com/gcgarner/IOTstack/blob/52c8adbab2065321c390075ddc0ca78c69645d56/.templates/telegraf/telegraf.conf#L167), you will see he set it up that way. Absent another explanation, I assume he thought it was useful, desirable or necessary.
3. Over time, the default configuration that ships with Telegraf grew and Graham's original became more and more out-of-sync.
4. The [current scheme](https://github.com/SensorsIot/IOTstack/tree/master/.templates/telegraf) is an attempt to preserve Graham's ideas in the up-to-date Telegraf context, without breaking any existing installations that rely on Graham's original setup.

How many existing installations rely on Graham's original setup? Absolutely no idea. Could be none. Could be thousands. Other than introducing a breaking change and waiting for complaints to come in, there's no easy way of finding out.

We could ask Graham about his original reasoning and whether it is still relevant but [nobody has heard from him in a long time](https://github.com/gcgarner/IOTstack/issues/194).

At the end of the day, it's "your system, your rules". If you don't want it in there then change it! If I wanted to do this on my own system I would start like this:

```
$ cd ~/IOTstack
$ mkdir -p ./services/telegraf
$ cp -a ./.templates/telegraf/* ./services/telegraf
```

then edit my `docker-compose.yml` to change:

```
  build: ./.templates/telegraf/.
```

to:

```
  build: ./services/telegraf/.
```

Then I could make any changes I wanted inside the `services/telegraf` directory, all independent of anything that might happen on the IOTstack GitHub repo, and keep rebuilding the container until I had it working the way I wanted it:

```
$ docker-compose up --build -d telegraf
$ docker system prune -f
```

Does that answer your question?

ukkopahis, 2022-04-28:
The `mosquitto` dependency seems like remnant misconfiguration. Though there's now an example for reading from mosquitto (`.templates/telegraf/iotstack_defaults/additions/inputs.mqtt_consumer.conf`), it's not enabled by default.

I'd say we remove the dependency:
* It would make the docker compose work even if mosquitto isn't selected in the menu.
  * The other option is to force menu users to select mosquitto as a dependency, which would make no sense.
* nothing will break even if there's a mqtt consumer and startup order is wrong, telegraf will try to reconnect later if a mqtt consumer is configured. At worst a spurious error is logged.

Nicest solution would be to have the dependency only if both services are selected, but menu doesn't currently support such functionality. Maybe after #505 that could be possible.

But what about its `influxdb` dependency? It's needed for telegraf's default configuration, but can be left unselected in the menu! I guess we'll just have to live with this problem, as the menu doesn't currently support dependency handling in any way and changing the current implementation is a bit too challenging.

simonmcnair, 2022-04-28:
I would remove the mosquitto dependency, as you say.  If you have a problem with mosquitto you don't want telegraf to fail.

Influx isn't a dependency, but it is a dependency.  This Telegraf instance only depends on influx because they're on the same machine.  Telegraf would run happily and fail to push metrics , but no-one wants that.

I would thing this would benefit from the same logic as the port duplication check, but as a dependency check.  If one container depends on another highlight it, but don't try to fix it.

In essence:
is telegraph selected ?
*what components have telegraf monitoring enabled (are they selected  ? mosquitto), if so copy in telegraf mosquitto.conf config file in to conf.d
*has user inserted custom influxdb server for telegraf if so proceed without dependancy
*if user has not selected custom influx server then prompt install influxdb and add dependancy

This is probably too much like hard work, but it's an option.  either way I'd remove the dependancy for mosquitto in case it isn't installed
cheers