SensorsIot/IOTstack #552: Wishlist: ioBroker Support

mooseDev, 2022-04-29:
Please add ioBroker to the list. Thx

Paraphraser, 2022-04-30:
Seems straightforward enough. Try this:

```
  iobroker:
    container_name: iobroker
    image: buanet/iobroker
#   hostname: iobroker
    restart: unless-stopped
    environment:
    - TZ=Etc/UTC
    - SETGID=1000
    - SETUID=1000
    ports:
      - "8084:8081"
    volumes:
      - ./volumes/iobroker:/opt/iobroker

```

External ports 8081..8083 already used by other templates. 8084 is first free in that sequence.

Not sure about the purpose of hostname. I know what it does. I just don't see the point in this context. That's why I've commented it out.

Included the uid/gid so you can give it "the treatment" once you settle on what that should be.

Seems to be "well behaved" - self initialises persistent storage. All owned by pi.

Paraphraser, 2022-04-30:
@mooseDev if you don't have a working service definition of your own to compare/contrast with the one I've suggested above, please do a copy/paste of mine into your own `docker-compose.yml` and see if it meets your needs. I don't pretend to understand what iobroker is or does so I'm not really in a position to test or offer critical evaluation. You'll have to decide whether this is fit for purpose or needs tweaking.

Paraphraser, 2022-06-19:
No response. Should probably be considered stale.

closed 2022-09-11 by Slyke