SensorsIot/IOTstack #586: issue with pihole and homer since update

DSew20, 2022-07-01:
Hello - I updated IOTStack via the menu, and since the update neither homer nor pihole start.

I have checked the networks and they seem to have updated per the instructions.  But, neither docker start.

Any ideas how to troubleshoot this?

Homer log:
```
2022-07-01T22:47:21.746132430Z 4147-08-20 07:32:16: (server.c.1568) server started (lighttpd/1.4.64)
2022-07-01T22:47:21.746158837Z 4147-08-20 07:32:16: (server.c.263) warning: clock jumped -68719476736 secs
2022-07-01T22:47:21.746203244Z 4147-08-20 07:32:16: (server.c.271) attempting graceful restart in < ~5 seconds, else hard restart
2022-07-01T22:47:21.746227096Z 1970-01-01 00:00:00: (server.c.1011) [note] graceful shutdown started
2022-07-01T22:47:21.746249651Z 1970-01-01 00:00:00: (server.c.2077) server stopped by UID = 1000 PID = 12

```

PiHole log:
`sudo: unable to get time of day: Operation not permitted
sudo: error initializing audit plugin sudoers_audit
`

Paraphraser, 2022-07-01:
You haven't given me much to go on. I run PiHole normally and I've just spun up homer on both my live and test systems. No problems with either container.

My live system is Raspbian Buster, my test system is Bullseye.

Although I haven't seen the particular time problem you have reported before, **usually** time problems turn out to be the base system being Buster and the libseccomp2 [recommended system patch](echo https://sensorsiot.github.io/IOTstack/Basic_setup/#patch-2-update-libseccomp2) not being installed.

Time problems most commonly show up on containers built on top of Alpine Linux. PiHole is Debian but Homer is Alpine - but I stress that Alpine is not a critical element for this problem to be exposed. There's a bit of background info in [issue 401](https://github.com/SensorsIot/IOTstack/issues/401).

Take all this with a few grains of salt because this libseccomp2 dependency has been around for a long time (10 months) so I'd be a bit surprised that you hadn't hit it before. But, if your base OS is Buster and you're thinking "what libseccomp2 patch?" then it **might** be the answer.

DSew20, 2022-07-02:
Thank you.  Never would have found that, but libseccomp2 seems to have fixed the issue.

.▀█▀.█▄█.█▀█.█▄.█.█▄▀　█▄█.█▀█.█─█
─.█.─█▀█.█▀█.█.▀█.█▀▄　─█.─█▄█.█▄█

Paraphraser, 2022-07-02:
You're welcome. Please remember to close this issue.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-07-17 by DSew20