SensorsIot/IOTstack #601: [menu.sh] virtualenv: error: no such option: --seed

IOIO72, 2022-09-20:
1. I've just updated `menu.sh` to the current version.
2. As I started the new version, it didn't find `virtualenv`.
3. I've installed `virtualenv` manually.
4. Running `menu.sh` again, leeds to the following error:

```
./menu.sh
Checking for project update
Von https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python virtualenv found.
Python Version: 'Python 3.7.3'. Python and virtualenv is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.18 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Existing installation detected.
Creating python virtualenv for menu...
Usage: virtualenv [OPTIONS] DEST_DIR

virtualenv: error: no such option: --seed
```

The virtualenv arguments don't work here:
https://github.com/SensorsIot/IOTstack/blob/5d7c603beca4532a26279aab46245dee506d6408/menu.sh#L396

Paraphraser, 2022-09-20:
Over on the IOTstack Discord group, there was some discussion about this. After some experimentation, I posted the answer below. This solution definitely works on my system and, from the absence of people on Discord saying things like "I tried that fix but mine still isn't working", I'm inferring that it probably works for most people.

Please try it and let me know if it fixes your system.

<hr>

From [Discord](https://discord.com/channels/638610460567928832/638610461109256194/1020475302620704778) ...

After further investigation, I believe this is the correct setup:

```
$ sudo apt install -y python3-pip python3-dev python3-virtualenv
$ sudo pip3 install -U virtualenv
```

When I do that on my own system, I get:

```
$ apt list python3-pip python3-dev python3-virtualenv
Listing... Done
python3-dev/stable,now 3.9.2-3 arm64 [installed]
python3-dev/stable 3.9.2-3 armhf
python3-pip/stable,stable,now 20.3.4-4+rpt1+deb11u1 all [installed]
python3-virtualenv/stable,stable,now 20.4.0+ds-2+deb11u1 all [installed]

$ pip3 list | grep virtualenv
virtualenv                  20.16.5

$ cd ~/IOTstack
$ ./menu.sh 
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python virtualenv found.
Python Version: 'Python 3.9.2'. Python and virtualenv is up to date.
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.18 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Existing installation detected.
Using existing python virtualenv for menu
```

TKTF50, 2022-09-21:
Thanks @Paraphraser,

I just updated my Pi OS and menu.sh and ran into a similar problem. Here's the error message I was seeing in my case (in case it helps others find this fix):

```
Existing installation detected.
Creating python virtualenv for menu...
./menu.sh: line 396: virtualenv: command not found
```

The two commands you listed fixed the problem for me.

Paraphraser, 2022-09-22:
The background to all this is that the "problem" reported in this issue seems to have been triggered by [PR560](https://github.com/SensorsIot/IOTstack/pull/560) which was applied around September 11 2022.

My guess is that a fair number of IOTstack users are going to get tripped up by this as soon as they do a "pull" against the IOTstack repository.

After a bit more experimentation, I've found some combinations which imply that the two commands I mentioned above (`apt install` and `pip3 install`) are probably a bit too simplistic and might not cover all situations.

To try to deal with the problem at its core, I've written a short "repair" script which is documented at:

* [PiBuilder: `reset_menu_enviroment.md`](https://github.com/Paraphraser/PiBuilder/blob/master/reset_menu_enviroment.md)

You do **not** need to have used PiBuilder to build your Raspberry Pi in order take advantage of that script. It **should** work irrespective of how you (a) built your Pi and (b) installed IOTstack.

However, I can only test with what I have so if anyone still has trouble after running the repair script, please let me know and I will try to incorporate an appropriate fix.

jonesrussell, 2022-10-09:
> ```
> $ sudo apt install -y python3-pip python3-dev python3-virtualenv
> $ sudo pip3 install -U virtualenv
> ```

This worked for me too thanks.